pvvx/ATC_MiThermometer #260: [feature request] humidity and temperature alarm as per original ATC1441 firmware

mpartington, 2022-11-01:
Apologies if I am missing this,, but I cannot se an equivalent to this in your documents: (below taken from atc1441 )

Temp or Humi instant advertising
When the temp or Humidity changes too fast between the main loop (5 seconds interval), the Advertising will be instant for that one.

byte0 0xFC = temp_alarm_point // value divided by 10 for temp in °C

byte0 0xFD = humi_alarm_point

byte1 as int8_t

Temp alarm from 0,1°C to 25,5°C Range <- 0,5°C Default

Humi alarm from 1% to 50% Range <- 5% Default

pvvx, 2022-11-05:
All data is transmitted with each measurement from the sensor. 
Measurement step and transmission interval are set.
The transfer takes place with duplication.
The transmission and duplication step cannot be more than 2.5 seconds for a reliable connection.
Those. at default settings, data is transferred every 2.5 seconds.
Measurement at default settings occurs every 10 seconds and is transmitted 4 times.
With Atc1441, if "alarm" is set, the transmitted data does not change until the threshold is passed. But the measurement occurs at a given interval.

mpartington, 2022-11-06:
Thanks for the reply, so I think with ATC, if it measures a user defined change in humidity (and triggers alarm), then it will immediately transmit the data rather than say waiting the 10 min advertising interval . Can I replicate this?

pvvx, 2022-11-06:
Sorry, I pressed the wrong button and deleted your comment.

> with ATC, if it measures a user defined change in humidity (and triggers alarm), then it will immediately transmit the data rather than say waiting the 10 min advertising interval .

No. 
Such possibilities are not implemented and are not available for the sensor.
In ATC, temperature and humidity measurements are performed only at the intervals set in the settings. The default is 50 seconds. And if the temperature and humidity do not change by a step of 0.5, then the data transmitted every 1.6 seconds will not change.

closed 2023-02-18 by pvvx