pvvx/ATC_MiThermometer #261: Daylight saving time

tomaszduda23, 2022-11-02:
It would be nice if clock adjust automatically for https://en.wikipedia.org/wiki/Daylight_saving_time_by_country

pvvx, 2023-02-18:
The thermometer does not have such resources and Internet access.

tomaszduda23, 2023-03-22:
It can be done offline.

https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv
https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html

As alternative I use esp32 which updates time once a day.

bvermes17, 2023-05-19:
@tomaszduda23 Could you share the script or config for the ESP that upgrades the time on the device once a day? I am currently trying to do the same, but did not find a solution yet. 

Thanks!

tomaszduda23, 2023-05-19:
```
esp32_ble:
  io_capability: keyboard_only

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: true

bluetooth_proxy:

external_components:
  - source: github://pr#4258
    components: [ble_client, esp32_ble, esp32_ble_client, esp32_ble_tracker]
    refresh: always

ble_client:
- mac_address: "XX:XX:XX:XX:XX:XX" # <- mac address
  id: pvvx_ble_display
  on_passkey_request:
    then:
      - logger.log: "Authenticating with passkey"
      - ble_client.passkey_reply:
          id: pvvx_ble_display
          passkey: xxxxx #<- paring key if you have one

display:
- platform: pvvx_mithermometer
  ble_client_id: pvvx_ble_display
  update_interval: 24h
  validity_period: 0s
  disconnect_delay: 30s
  time_id: homeassistant_time

```

Ircama, 2023-05-20:
> It can be done offline.

You might also check [setting the device date to the host system's date"](https://github.com/Ircama/ATC_MiThermometer/blob/atc-mi-intercace/python-interface/README.md#examples-of-usage) in PR https://github.com/pvvx/ATC_MiThermometer/pull/337:

```
atc_mi_config -m A4:C1:38:AA:BB:CC -D
```

tomaszduda23, 2024-07-11:
just for reference https://github.com/josmithua/utz it seems that 3KB is needed for timezone db

mkaluza2, 2026-10-26:
@pvvx @tomaszduda23 I've just hit the issue due to DST change on Sunday... There is another solution to this - let the sensor always run in UTC and do any TZ conversions client side - then there is no resource problem and functions that convertUTCtoLocal() are either readily available in most languages or very easy to write.

only then the TelinkMiFlasher.html page and python cli would need an extra "use UTC" switch