pvvx/ATC_MiThermometer #263: iBeacon advertisement

Smilebags, 2022-11-08:
Now that the Home Assistant companion app now supports a 'Beacon Monitor' (where it can detect advertisements in the iBeacon format), and considering I already have the Xiaomi Mijia LYWSD03MMC in every room of my house, I would like the ability to detect the strength of the advertisements from the thermometers on my phone, and therefore determine the rough location of my phone within the house.

Is it possible to send iBeacon format BLE advertisement packets from the Xiaomi Mijia LYWSD03MMC, or, would it be possible to add support for this function, even if it reduces the battery life of the device?

chatziko, 2022-12-29:
+1

It would be quite interesting to turn LYWSD03MMC into an ibeacon for a variety of purposes.

jamestutton, 2023-01-04:
Was about to raise the same request as was looking to add some beacons to my setup but have these all over the place so if they able to send a iBeacon compatible message then that would be great as could be used along side the companion app to do some presence detection.

biemster, 2024-09-30:
A bit late to the game, but for people interested in this it is probably quite easy to add the ibeacon frame as detailed here: https://github.com/sandeepmistry/node-bleacon#ibeacon-advertisement-format
to the advertiser the same way as was done for airtag style frames here: https://github.com/biemster/tlsr825x_FindMy/blob/d79d41d0c2c5c72221be1698faf5bccc83160297/app.c#L57