pvvx/ATC_MiThermometer #267: Issues with Reading Measurements from Flash on Android/Chrome

navlisilvan, 2022-11-17:
Hi,

i have some issues with Reading Measurements from Flash on Android Chrome. With "Get the last 50 records" in "Flash"-Menu i get at the moment 48 velues, but in "Memo Graph" I get unreliable less values. One time 12, one time 17, one time 22, one time 19.
One time is says "Read count: 48" but afterwards "Disconnected, Load 2 samples"

On Windows Chrome it loades 49 samples. 

Hardware Version: LYWSD03MMC B1.7, Software Version: 3.8, Sensor: SHT4x
Custom config: [173, 96, 0, 0, 40, 4, 169, 124, 49, 5, 60]

Any Ideas?

pvvx, 2022-11-18:
1) For a stable BLE connection with Xiaomi LYWSD03MMC, it is desirable that the CR2032 battery has a capacity of more than 40%. The reasons lie in the cheapening of the circuit by the manufacturer - the lack of recommended capacitors for power supply.
2) Some smartphones have bad driver software for the BLE chip. And very often manufacturers do not correct these errors, but copy them to the next series.

trevi020, 2022-12-06:
> Hi,
>
> i have some issues with Reading Measurements from Flash on Android Chrome. With "Get the last 50 records" in "Flash"-Menu i get at the moment 48 velues, but in "Memo Graph" I get unreliable [...]

I'm having the same issue both from PC and from Android Chrome. Get last 50 values works great in the Flashing page, but when using the Memo Graph page it is both very slow and after getting the values as "Read Count" it only displays much less that the readed count. The displayed values varies. This is such a pity considering the Memo Graph page for the original firmware worked great and was able to read 4 months of values in just a few minutes.

strelok15, 2022-12-09:
I have the same issue: "Get the last 50 records" works fine, but "Memo Graph" misses some starting records.

I supposed the listener is not ready yet to listen Advertisments when the LYWSD03MMC starts sending data. The first "Count:" in console log depends on client: starts from 9 on my Windows PC and from 50-100 on Android.

The following workaround helps for me: Delay on sending get-memo command
```
        ...
	log('Send command get-memo (0..19632 samples)...');
	//return characteristicVTH.writeValue(new Uint8Array([0x35,0xB0,0x4C,0,0]));
        return setTimeout(() => characteristicVTH.writeValue(new Uint8Array([0x350xB0,0x4C,0,0])), 3000);
```
The necessary delay 3000 may be different for your devices.
I'm very far from JS skills, probably that's an incollrect solution, but it works. So I decided to share it.

Sorry for my bad English, hope you could understand me

pvvx, 2022-12-09:
> This is such a pity considering the Memo Graph page for the original firmware worked great and was able to read 4 months of values in just a few minutes.

Adapter: CSR8510 A10, Windows, log Chrome : 
_Read count: 19624
Read time: 19.276 sec_

Old Android smartphone - about 30 seconds.

trevi020, 2022-12-09:
> > This is such a pity considering the Memo Graph page for the original firmware worked great and was able to read 4 months of values in just a few minutes.
> 
> Adapter: CSR8510 A10, Windows, log Chrome : _Read count: 19624 Read time: 19.276 sec_
> 
> Old Android smartphone - about 30 seconds.

Sorry my bad, reading is extremely fast, no issue there, it's the displaying that I'm having problem with. The Load value is inconsistent. Shouldn't the displaying phase or "Loading" phase to generate the graph just depend on the displaying machine, not the ATC device?  Or its still syncing the readed values during the Load phase? I currently have 2700mv, could this be the issue? I've seen that reading the flash memory is one of the most current drawing operations from the test perfomed.

pvvx, 2022-12-09:
The receiver acknowledges the transmission of each packet. If there is no confirmation, then the sender duplicates the packet and waits again (puts the battery down).

pvvx, 2022-12-10:
> I currently have 2700mv, could this be the issue?

Slow and old adapters greatly discharge the thermometer battery - they do not have time to confirm transmissions and there are constant retransmissions.

pvvx, 2022-12-10:
Test: Read "Memo", ATC_v39, pin-code enable.  
Chrome log:
_Read count: 19632
Read time: 18.542 sec_

Real read cycle: 17320 ms 

**Average per package**: 17320/19632 = **0.8822 ms**  -> 1133.5 package per second
[image]

mfur, 2023-02-17:
I have similar issue and also a practical workaround.

If I use Memo Graph, it is hard to establish connection. When connection is finally established in multiple attempts, datapoints are read but not all of them are loaded. Currently I miss last 20 datapoints from both of my thermometers quite consistently.

If I open two Chrome Android tabs and connect with Flasher in one tab, then connect with Memo Graph, readout is done much faster, reliable and in full - all datapoints are read and loaded in graph. When using this method, initial icon for device in pairing window is different - BT icon instead of usual signal strength icon.

Is there much bigger battery drain when using this approach? It is slowly becoming default for me as it works so well.

Hope that helps as a practical workaround and perhaps as idea on how to do reliable readout in code if there is no big battery drain as a downside (perhaps option of connection method could be added?). Let me know if there is additional info that I could provide.

Thank you for great firmware, I'm really enjoying it.

pvvx, 2023-02-17:
Use software version: 4.1 or read the Warnings in the README.md repository:
_[On devices with missing power capacities](https://github.com/pvvx/ATC_MiThermometer/issues/265#issuecomment-1431495494), do not set 'Connect latency' to more than 1000 ms!_

mfur, 2023-02-17:
Thanks for quick reply. Flashed 4.1 - unfortunately it didn't help. Battery is new (got first two new thermometers yesterday).

Attempts in Memo Graph, FW 4.1:
1. Worked in full
2. Stuck at Send command get-memo
3. Stuck at GATT server
4. Loaded 28 out of 113 sampley
5. Loaded 93/113
6. Loaded 93/113 (most common scenario in 4.0 FW as well)

If connecting with Flasher first as described in previous posts, it works 100% of the time and incredibly fast (you can't even see the counter increasing, it just completes right away). Flasher also does auto reconnect so no messing around with that.

pvvx, 2023-02-17:
I changed the code many times, but never managed to get connection stability in "Memo Graph". The main reason is that it works stably on my devices and I don't know what else to change there.
And I am not a programmer in High-level programming language and have not thoroughly studied Java.
Can you point or give direction on how to do this in js?

closed 2023-12-23 by pvvx