pvvx/ATC_MiThermometer #268: phy6222 controller support.

Jarled, 2022-11-20:
I bought the newer version of this sensor and it seems to be using a PHY6222 controller, could get it working on this please?
[image]
[image]
[image]

pvvx, 2022-11-20:
The PHY6222 has a higher consumption (13 uA) than the TLSR825x (1.8 uA) in sleep mode with active SRAM. Also, the PHY6222 software and hardware has an increased SoC wake-up time - the result is an even higher average current.
Thermometers with TLSR825x in combination with an LCD/E-Ink driver and sensor have a sleep current of up to 9 µA.
With PHY6222, battery life will be significantly reduced...  Unsuccessful option.

Jarled, 2022-11-20:
i dont think so, its meant to be a Bluetooth 5.0 BLE version
https://www.aliexpress.com/item/1005004865399211.html?spm=a2g0o.order_list.0.0.54e218024YIBU1

pvvx, 2022-11-20:
Oh, it's Tuya :)
For Tuya, any hacky manufacturers are produced ...

pvvx, 2022-11-20:
Another "miracle" BLE thermometer CX0726B:
[image]
It has CR2032, but the average current in ad mode is 80 uA. And when connected to a regular Chinese program, the average current is 0.8 mA.
Temperature and humidity transfers only when connected. (This is for all Tuya)
[image]
[image]

[image]

Jarled, 2022-11-20:
i have no experience in this space, if someone can help me remove the bluetooth bind key bit and i can add it to HA.

pvvx, 2022-11-20:
Temperature and humidity transfers only when connected. (This is for all Tuya)

Jarled, 2022-11-20:
Thanks, i got that info now, is there a browser page i can test with the local key ?

pvvx, 2022-11-20:
https://www.aliexpress.com/item/1005004865399211.html
I ordered it, I'll see when it arrives.

Jarled, 2022-12-02:
did it arrive ?

pvvx, 2022-12-03:
Last info in tracking:

Day 5 11/24/2022 18:11:12
Left the country of transit
Day 6 11/25/2022 00:00:00
Loading on the plane
Day 7 11/26/2022 13:15:00
[Hong Kong] Sent
day 13 02.12.2022 00:30:00
Arrived at destination airport

pvvx, 2022-12-09:
Got a few pieces half an hour ago. Boxes like that - I took a picture of a couple:
[image]
Haven't opened the box yet :)

pvvx, 2022-12-09:
I will post data on TX-05 in https://pvvx.github.io/TH-05/

Jarled, 2022-12-09:
What do you want me to write, have you managed to get it working ?

pvvx, 2022-12-09:
TX-05 consumes a lot [in operating mode](https://pvvx.github.io/TH-05/): Average 33.4 uA

Jarled, 2022-12-09:
Can these be added to Home assistant via BTHome or will it require creating a new program ?

pvvx, 2022-12-09:
https://github.com/rospogrigio/localtuya/issues/1166 -- :`(

https://www.home-assistant.io/integrations/tuya/ +++ :)

Jarled, 2022-12-09:
ok but what about doing it via ESPHome instead ?
https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc

pvvx, 2022-12-09:
No way. It requires creating a new program for PHY6222 and (wired) flashing the thermometer, which is not worth doing for this chipset version. It is not suitable for long work from CR2032.

pvvx, 2022-12-09:
When temperature or humidity changes, TH-05 transmits data to BLE advertising encrypted using the Tuya protocol.
HA requires integration that supports BLE Tuya.

pvvx, 2022-12-10:
https://developer.tuya.com/en/docs/iot-device-dev/Porting-Guide-BLE?id=Kam0xjtz4n6e0#title-4-Advertising%20data%20format

pvvx, 2022-12-23:
##Advertising packages:

UUID 0xFD50 - TUYA BLE PROTOCOL VERSION 4

| type |  Flag  | UUID 02  | UUID 16   | UUID 16 Data                       |
| ---- | ------ | -------- | --------- | ---------------------------------- |
| msg  | 020106 | 030250fd | 141650fd  | 488040000001b36eb30d64e80a4d9cbc38 |
| adv  | 020106 | 030250fd | 141650fd  | 490000080c9c105bc01d9f87           |

adv - always transmitted.
msg - transmitted when temperature and/or humidity changes.

##Adv contains:

before bond - device pid, after bond - encrypted device_id with login_key and device_id

##Packet bytes, except for standard identifiers, are formed as follows:

[7] pid_len+7
[11,12] Frame Control: bound_flag |0x08, DEVICE_SHARED |0x04, connect request bit |0x02
[13] pid_type
[14] pid_len
[15...] code

Clarification https://github.com/tuya/tuya-ble-sdk/blob/main/sdk/src/tuya_ble_main.c#L453

cconde, 2023-06-05:
We shouldn't expect this great firmware on this device soon, isn't it?

pvvx, 2023-06-07:
In this repository and TelinkMiFlasher.html Telink SoC devices are supported.
There is no gcc on the PHY chipset. Only proprietary paid Keil.

serpent213, 2023-06-30:
> TX-05 consumes a lot [in operating mode](https://pvvx.github.io/TH-05/): Average 33.4 uA After registration in "Smart Life": Average 43.2 uA (!)
> 
> Minimum chipset power consumption (sleep mode, datasheet): PHY6222QC-W04I - 13uA @ Sleep Mode with 32KHz RTC and all SRAM retention BL55028 - 9 uA Sensor? - 1 uA? Total: above 23 uA (A lot for CR2032. Half a year maximum?) Writing a program for such equipment does not build time. Such products (Tuya) go to the trash.

According to the datasheet the RAM is split up into three blocks (32kB + 2 x 16kB) that can be switched off individually. So with only 16kB enabled I would expect a power consumption of 1uA + 1/4 * (13uA - 1uA) = 4uA. So for the whole device 4uA + 9 uA + 1uA = 14uA. Sounds reasonable?

pvvx, 2023-07-01:
> According to the datasheet the RAM is split up into three blocks (32kB + 2 x 16kB) that can be switched off individually. So with only 16kB enabled I would expect a power consumption of 1uA + 1/4 * (13uA - 1uA) = 4uA. So for the whole device 4uA + 9 uA + 1uA = 14uA. Sounds reasonable?

Cycle 2.5 sec:
14 uA sleep mode - 2595 ms
6 mA active mode - 5 ms (аctually more)
( 6 * 5 + 0.014 * 2595 ) / 2500 = 0.026532 mA

Monacoslo, 2023-12-01:
So conclusion is?

almirus, 2023-12-15:
@pvvx  How long did the device ultimately last on battery?

NeoMatrixJR, 2024-01-07:
> In this repository and TelinkMiFlasher.html Telink SoC devices are supported. There is no gcc on the PHY chipset. Only proprietary paid Keil. That is, there can only be an internal non-public or paid version. No one is doing [Taya BLE (OTA)](https://github.com/pvvx/ATC_MiThermometer/issues/339#issuecomment-1567624516), and there is no interest in doing it alone.

Looks like there's Community Edition of Keil - can you not release code using community edition as open source? https://www.keil.arm.com/community/

pvvx, 2024-01-07:
Everything has already been decided. For PHY62x2, GCC sources are collected with RF library sources.
The implementation will be available soon at https://github.com/pvvx/THB2.

pvvx, 2024-01-07:
https://github.com/pvvx/THB2 обновлен для сборки в GNU Arm Embedded Toolchain.

almirus, 2024-01-07:
> HTTP ERROR 451
нужен впн

pvvx, 2024-01-07:
> нужен впн

Обойдется. Мне Keil не нужен

pvvx, 2024-01-24:
Первая тестовая версия уже на https://github.com/pvvx/THB2
Все обсуждения туда.

mr-aleks, 2024-02-06:
Возможно ли что то сделать вот с таким датчиком https://github.com/pvvx/ZigbeeTLc/issues/53 ?  Чип TLSR8250 и SH40. Если нужна еще какая то информация, предоставлю, все что потребуется. :)

pvvx, 2024-02-06:
> Возможно ли что то сделать вот с таким датчиком https://github.com/pvvx/ZigbeeTLc/issues/53 ?

У меня  такого нет. Ради одного датчика никто шевелиться не будет, только если сами возьмете исходные коды и поправите под свой вариант.