pvvx/ATC_MiThermometer #270: CGDK2  Unable to activate

rjh169, 2022-11-23:
Using V4.4 Telink Flasher    I loaded firmware on to 4  LYWSD03MMC sensors with no problems.
When I tried to load firmware on to some CGDK2 sensors I could not get them to activate.
The Device id appeared but then it just sat with the _Activating now, please wait_ message

[image]
The sensors work correctly with the Xiaomi BLE integration.
Deleting them from the integration and trying again made no difference.
Any ideas?

pvvx, 2022-11-23:
To activate, read the description in the instructions for CGDK2 "Restoring factory settings".

rjh169, 2022-11-25:
I tried that with a PC and BT adaptor and then an android tablet with no success.
After many attempts I finally found a method that worked for me and may do so for others.

Connect, select “Do activation” The log will say _Activating now, please wait_
Hold the button at the rear of the device until the Bluetooth symbol flashes. The device will disconnect.
Reconnect, select “Do activation” then _momentarily_ press the button.
The Device Id, MI Token and Binding key values will be retrieved.
The firmware file CGDK2_v38.bin will not automatically appear so you need to choose the file manually.

I was able to do this without any preparation. My devices were running and displaying in Home Assistant when I selected them in Telink.  I did 8 devices this way.

7 devices finished up being identified as CGD_xxFF  1 identified as CGD_xx32.    Any idea why?
After flashing the Bluetooth icon was no longer displayed.     Correct?
After flashing 5 devices I noticed that the battery’s capacity had dropped by 15% to 18% for each.
I noted the before and after for the last 3.   (100% 3.1V down to 84% 2.956V)  (98% 3.082V down to 81% 2.929V) and  (97% 3.073V down to 81% 2.929V).  Flashing has a price!

Am I understanding Advertising interval and Measure interval correctly??
Advertising interval is the time between BT transmissions sent by the device. So at 2500ms Home Assistant receives a data update every 2.5 seconds.
Measure interval is the time between readings by the device of its internal sensor. Default 10 seconds.

Lastly, what is Tx measures used for?

pvvx, 2022-11-26:
I flashed the firmware to the new CGDK2 yesterday.
I pressed and held the button as in the instructions from the box.

> The firmware file CGDK2_v38.bin will not automatically appear so you need to choose the file manually.

The firmware file CGDK2_v38.bin will automatically appear.

pvvx, 2022-11-26:
> 7 devices finished up being identified as CGD_xxFF 1 identified as CGD_xx32. Any idea why? After flashing the Bluetooth icon was no longer displayed. Correct? After flashing 5 devices I noticed that [...]

CR2032
[image]
Nominal output voltage - 2.9..2.95 V at load.

Convert voltage to percentage:
100* (battery_mv - 2200)/(3100-2200) = 100*(2950-2200)/(3100-2200) = 83.33%

rjh169, 2022-11-27:
You flashed the firmware to a _new_ CGDK2.  My devices were purchased early in 2021.
From my notes the firmware was 1.1_0102.  I have now overwritten the custom firmware on 2 of them
with Original firmware 1.1.1_0217  and then reflashed the custom firmware in the way 
you had described and the firmware file CGDK2_v38.bin appeared automatically.

So it would be advisable to update original firmware to 1.1.1_0217 before flashing the custom firmware.

The Bluetooth symbol remains displayed before and after a device is connected via Telink.
However when you finish flashing and disconnect, the symbol disappears never to be seen again.

The voltages and battery % were measured, before and after flashing, through the Xiaomi BLE integration in Home Assistant.  I have no idea of the validity of the % calculation it makes.  The CGDK2 uses a CR2430 battery.

pvvx, 2022-11-27:
CR2430 battery - analogically:
[image]

CGDK2 with original firmware consumes more than 5 mA during flashing. Flashing time > minute.

CGDK2 with custom firmware consumes 1.8 mA during flashing. Flashing time ~ 35 sec. -> 0.0175 mA/h

rjh169, 2022-12-01:
Using Telink Flasher V4.6, BT icon appears immediately a device is connected.
Flashed CGDK2_v39.bin,    BT icon disappears approximately 13 to 18 seconds after disconnection.
The timing is affected by LCD sample rate.

Then flashed original firmware 1.1.1_0217    BT icon remains after disconnection.

Flashed CGDK2_v38.bin,    BT icon disappears approximately 13 to 18 seconds after disconnection.

I have 12 CGDK2 flashed and working with no BT icon.
Changing configuration settings made no difference to the BT icon disappearing.

rjh169, 2022-12-02:
> _In all devices with BLE and display, the BT icon only appears during the connection._

[image]
With original firmware 1.1.1_0217, BT icon remains displayed after flashing is complete.

[image]
With firmware CGDK2_v38 / CGDK2_v39,  BT icon remains absent after flashing is complete.

Is this correct?

twaymouth, 2023-04-12:
I have a similar issue with a MHO-C401N device, the device is detected by the flasher web app however I have not been able to complete the activation phase. 

07:51:36: Reconnect 1 from 5
07:51:45: Hardware Revision String: 0000
07:51:45: Software Revision String: 0016
07:51:45: Firmware Revision String: 1.0.0_0016
07:51:45: Detected Mi device
07:51:45: Connected
07:51:45: Found new version of SW:16 MHO-C401N
07:51:50: Activating now, please wait...
07:51:52: DataError
07:51:58: Activating now, please wait...
07:52:20: Activating now, please wait...

I pressed and held the button as per directions, after a period the device reboots, not sure if it needs to be held for longer than this? I also note on the box it has a revision date of 2023.02, not sure if this means anything or if that is just when it was produced.

I attempted a manual flash of the device using usb to serial with MHO_C401N_v43.bin however it seemed to brick the device, there was no response on the screen and Bluetooth did not start transmitting.

I attempted to restore the device using Original_OTA_Xiaomi_MHO_C401_v1.0.0_0016 bin file however this also did not do anything, the screen did not activate and Bluetooth did not start transmitting. 

The only way I was able to restore the device was reading the original firmware off another thermometer of the same revision 2023.02 ( I purchased a couple of them at the same time).

Not sure if there has been a hardware change with revision 2023.02 or if I am doing something incorrect? let me know if there is any way I can help investigate this.

pvvx, 2023-04-13:
> The only way I was able to restore the device was reading the original firmware off another thermometer of the same revision 2023.02 ( I purchased a couple of them at the same time).

Is it possible to attach the firmware from these thermometers to the message?

twaymouth, 2023-04-13:
Attached, thanks. 
[MHO_C401_2023.02.zip](https://github.com/pvvx/ATC_MiThermometer/files/11217149/MHO_C401_2023.02.zip)

pvvx, 2023-04-13:
This firmware contains [Xiaomi and Azarton MHO_C401 Original Firmware v1.0.0_0016](https://github.com/pvvx/ATC_MiThermometer/raw/master/Original_OTA_Azarton_MHO_C401_v1.0.0_0016.bin)
Full equality.

twaymouth, 2023-04-13:
Thanks for the confirmation, seems like my devices may be unsupported.

twaymouth, 2023-05-20:
If you are interested in adding support for the Xiaomi and Azarton MHO_C401 variant (or as general information for anyone else who wants to work with this thermometer) I have done some work on the display with a logic analyser and found its functionality is very similar to that already implemented for the MHO_C401N.

The pinout, segments, buffer etc are the same, the key difference is the LUTs are slightly different as is the sequence by which they work. 

The LUTs are as follows:

`const uint8_t T_LUT_ping[5] =  {0x7B, 0x81, 0xE4,0xE7,0x0E};`
`const uint8_t chargePump[2] = {0xAC,0x2B};`
`const uint8_t T_LUT_init[10] = {0x2B,0xA7,0xE0,0x82,0x68,0x50,0xE8,0xD0,0xA8,0x65};`
`const uint8_t T_LUT_work[7] = {0x082, 0x080, 0x000, 0x0C0, 0x080, 0x080, 0x062};`
`const uint8_t initData[16] = {0};`

Depending whether a full refresh / init is occurring LUT_WORK or LUT_INIT is transmitted, this is followed by  LUT_PING in both instances, a slight delay occurs (5ms) the charge pump is then turned on, another slight delay occurs and the data is then transmitted with a header and footer the same as the MHO_C401N. During a refresh the data transmitted is all 0x0.

The display then goes busy and once the busy line goes high a power off LUT is transmitted same as the MHO_C401N.

the other difference is that between the full refresh and regular update the display is powered down and a full cycle starts again.

After 35 partial refreshes a full refresh occurs

I am having trouble getting your full firmware to compile using the Telelink IDE / SDK however put together a quick test firmware using the function below to test the display, as far as my basic testing goes everything appears to work as expected, the display refreshes and updates correctly etc.

```
attribute_ram_code_ int task_lcd(void) {
	if (gpio_read(EPD_BUSY)) {
		switch (stage_lcd) {
		case 1: // Update/Init, stage 1

			if (epd_updated) {
				transmit_blk(0, T_LUT_work, sizeof(T_LUT_work));
			} else {
				transmit_blk(0, T_LUT_init, sizeof(T_LUT_init));
			}
			stage_lcd = 2;
			break;
		case 2: // Update/Init, stage 2

			transmit_blk(0, T_LUT_ping, sizeof(T_LUT_ping));

			pm_wait_ms(5);
			transmit_blk(0,chargePump,sizeof(chargePump));
			pm_wait_ms(5);
			stage_lcd = 3;
			break;
		case 3: // Update/Init, stage 3

			transmit(0, 0x040);
			transmit(0, 0x0A9);
			transmit(0, 0x0A8);
			if (epd_updated) {
				transmit_blk(1, display_buff, sizeof(display_buff));
			}else{
				transmit_blk(1, initData, sizeof(initData));
			}
			transmit(0, 0x0AB);
			transmit(0, 0x0A4);
			transmit(0, 0x0AF);
			stage_lcd = 4;
			break;
		case 4: // Update, stage 4

			transmit(0, 0x0AE);
			transmit(0, 0x028);
			transmit(0, 0x0AD);
		if (!epd_updated){
			epd_updated = 1;
			stage_lcd = 1;
			break;
		}
		default:
			stage_lcd = 0;
		}
	}
	return stage_lcd;
}
```
The only caveat is that for whatever reason despite the LUTs being identical to the factory firmware the displays power consumption is slightly higher with the custom firmware vs the factory firmware, it almost seems like there is some extra parasitic draw or the controller is not able to go to sleep fully while a refresh is occurring?  Possibly this is related to timing of the LUTs? I am continuing to investigate this when I have time.

I am also trying to find time to try and get the SDK working and compile your full firmware with these changes however if you felt like doing so yourself I would be happy to carry out testing.

pvvx, 2023-05-22:
> If you are interested in adding support for the Xiaomi and Azarton MHO_C401 variant

https://github.com/pvvx/ATC_MiThermometer/issues/276#issuecomment-1441849596

> The only caveat is that for whatever reason despite the LUTs being identical to the factory firmware the displays power consumption is slightly higher with the custom firmware vs the factory [...]

For the test with the cheapest battery (less than 0.1 USD) is the CR2025.

[image]

[image]

The thermometer operates in the 'LE Long Range' mode.
Consumption during transmission in the "Long Range" mode is more than 8 times as standard.
The connection with TelinkMiFlasher is possible after pressing the button on the back of the device.

_Hardware Version: MHO-C401N 2022, Software Version: 4.2, Sensor: SHTC3 (SHTV3)
Custom config HEX string: 55077000005003a931638878_

[image]

twaymouth, 2023-05-22:
Perfect, thanks for your feedback, I misunderstood that MHO_C401N_v42.bin supported the  Xiaomi and Azarton MHO_C401 variant, everything seems to be working as expected and power consumption for the display is inline with the factory firmware.