pvvx/ATC_MiThermometer #271: Wrong temperature advertised

1cko, 2022-11-24:
After flashing with the latest pvvx firmware I have strange temp readings. Connected to Raspberry Pi3 B running Home Assistant using Passive BLE monitor integration. The second sensor (ble temperature Kitchen) is with stock firmware. I get this behavior once every few days.
[image]

pvvx, 2022-11-25:
https://github.com/pvvx/ATC_MiThermometer/issues/246#issuecomment-1323280418 ?

1cko, 2022-11-25:
> [#246 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/246#issuecomment-1323280418) ?

I somehow missed that. I will try with encrypted Adv on the sensor workaround mentioned there. Thank you!

rubencm, 2022-11-26:
I had no problems when using https://github.com/custom-components/ble_monitor (now I use the bthome integration), it started to happen me too when I upgraded the version of the pvvx firmware

pvvx, 2022-11-26:
BTHome uses bluez and active scanning.
[Due to the active mode, the batteries in the devices run out faster and there is a lot of writing to the disk. ](https://github.com/Bluetooth-Devices/bthome-ble/issues/26)
Switch to passive mode.

sordfish, 2023-01-01:
I've started to see the same data glitches after upgrading 7 LYWSD03MMC sensors to 3.8 

I'm using https://github.com/JsBergbau/MiTemperature2 and had a quick look at the code and noticed some of the packets are causing processing errors as it seems that some packets aren't getting flagged as encrypted and getting decoded as lywsdcgq 22byte packets instead. I've commented some code out to try and work round the issue.

I've ruled out hardware for now as this is happening on multiple devices - pi zero w 2, pi 2 with ebay dongle and my computer with intel bluetooth

pvvx, 2023-01-02:
> flagged as encrypted and getting decoded as lywsdcgq 

In the Mijia format, in addition to the encryption bit, there is also a device identifier number. lywsdcgq has a different one.
Also, if the packet type indicator is not transmitted, then this indicates a complete malfunction of the device.

And this is not to mention the fact that lywsdcgq works with Mijia UUID, and LYWSD03MMC with multiple formats set in the configuration.
And since version 3.7, there have been no changes in the formats.

sordfish, 2023-01-02:
Hey pvvx, thanks for the in depth reply!

I've manged to "fix" my issue by commenting out some of the lines in [LYWSD03MMC.py](https://github.com/JsBergbau/MiTemperature2/blob/master/LYWSD03MMC.py) and as you can see things are looking better. When I have more time I'll try debug further.

[image]

closed 2023-02-18 by pvvx