pvvx/ATC_MiThermometer #272: Bug in Telink Flasher v4.5

miXza81, 2022-11-28:
At least in Chrome on Android, I have to select different firmware for each device when I'm uploading firmware using 'Select firmware' field. This because 'Start Flashing' button is hided after each upload, even when 'Select firmware' field contains selected file.

pvvx, 2022-11-29:
How to reproduce the described problem?
Has the problem gone in the latest version?

pvvx, 2022-11-29:
There are uncertainties in such a decision.
For example, if you connect to another thermometer - let's say CGG1, and the last downloaded firmware from CGDK2... And the user will be able to flash it into an incompatible device.

miXza81, 2022-11-29:
Sure, but I own four ATC sensors which all are same types - So I have no that risk.
I have much greater risk to select wrong firmware file because I must re-select it after each upload or upload atempt... Maybe it would be possible to reset value of selected file IF different type of sensor is connected?

pvvx, 2022-11-30:
Later I will try to add the full solution to TelinkMiFlasher.html...

closed 2023-02-18 by pvvx