pvvx/ATC_MiThermometer #273: There is a problem with CGDK2

1286840533, 2022-11-29:
After flashing the Custom firmware.  Temperature and humidity both show 0.  E2 is displayed after restoring the original firmware

1286840533, 2022-11-29:
[image]
[image]
[image]

bogi-bo, 2022-12-05:
I have the same Problem Problem, with custom FW 0 and with original E2, or EZ

1286840533, 2022-12-06:
> I have the same Problem Problem, with custom FW 0 and with original E2, or EZ I have one which I haven't flashed, there is the Firmware v1.1.1_0223.bin as stock

I've tried all the firmware to no avail.

pvvx, 2022-12-12:
Has the sensor type changed?

bogi-bo, 2022-12-12:
> Has the sensor type changed?

What do you mean with this?

pvvx, 2022-12-12:
> What do you mean with this?

It is described that "After flashing Custom firmware. Temperature and humidity show 0."

> Is it possible to download firmware with OTA BLE from a working one and put it to a broken one?

Yes.

bogi-bo, 2022-12-12:
> 
> > Is it possible to download firmware with OTA BLE from a working one and put it to a broken one?
> 
> Yes.
do you have maybe a short instruction or a link? Would be great!
If I can help you with providing Data from the exact sensor inside, just let me know! I'd love to support your great work!

pvvx, 2022-12-13:
https://github.com/pvvx/ATC_MiThermometer/issues/114#issuecomment-979458611

bogi-bo, 2022-12-13:
> [#114 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/114#issuecomment-979458611)
> 
> > there is the Firmware v1.1.1_0223.bin as stock
> 
> I don't have Firmware v1.1.1_0223

Hope this can help. I've done the instruction and here is the file. 
Thank you again for your support! 

[ff.zip](https://github.com/pvvx/ATC_MiThermometer/files/10221793/ff.zip)

pvvx, 2022-12-13:
Add [Qingping CGDK2 Original Firmware v1.1.1_0223](https://pvvx.github.io/CGPR1/img/Original_OTA_CGDK2_v1.1.1_0223.bin)

bogi-bo, 2022-12-13:
> Сut and added file [Qingping CGDK2 Original Firmware v1.1.1_0223](https://pvvx.github.io/CGPR1/img/Original_OTA_CGDK2_v1.1.1_0223.bin) (The checksum of the original firmware is normal.)
> 
> You can delete your file if you have any concerns about the data from it.

It is working, thank you! 
For me it looks "solved" 

@1286840533 can you try?

pvvx, 2022-12-13:
It remains to find out which sensor is used in the new version.

bogi-bo, 2022-12-13:
> It remains to find out which sensor is used in the new version.

If I can help, let me know…

pvvx, 2022-12-13:
A good [photo of the board](https://pvvx.github.io/CGDK2/) or reading the marking from the sensor is required.
To do this, you need to disassemble the case. The action is complicated by the possibility of spoiling the plastic latches and the rear view of the case...

pvvx, 2022-12-13:
Without sensors, bare TLSR8253, firmware v1.1.1_0223 (fullflash), start log with I2C.
[i2c_log.zip](https://github.com/pvvx/ATC_MiThermometer/files/10222506/i2c_log.zip)
Now guess who is at 0x88 (0x44), and 0xE0(0x70), and 0x00:
[image]
[image]
[image]
[image]
[image]

pvvx, 2022-12-14:
I don't see any changes.
And what doesn't work for you?
What does it mean "Problem, with custom FW **0** and with original **E2**, or **EZ**"?

bogi-bo, 2022-12-14:
> I don't see any changes.
> And what doesn't work for you?
> What does it mean "Problem, with custom FW **0** and with original **E2**, or **EZ**"?

I can try to open it and take a picture for you.

a better description would be:
With the custom FW the temperature and humidity is showing 0.0 on both.

when flashing with „original FW 217“ on the display is showing „E2“

according what i‘ve found in a documentation from qingping it says the „sensor“ is broken, so it looks like what you thought, that there is an other sensor.

[image]

opcode02, 2022-12-18:
I have the same CGDK2 issue. I took mine apart and took a close up picture of the sensor. It appears its a SHTV3 maybe?

[image]

pvvx, 2022-12-18:
SHTV3 = SHTC3

1286840533, 2022-12-22:
> > 删减文件[清平CGDK2 Original Firmware v1.1.1_0223](https://pvvx.github.io/CGPR1/img/Original_OTA_CGDK2_v1.1.1_0223.bin) (原固件校验正常)
> > 如果您对其中的数据有任何疑虑，可以删除您的文件。
> 
> 它正在工作，谢谢！ 对我来说它看起来“解决了”
> 
> @1286840533你能试一下吗？

Sorry for the late reply, this firmware works fine

yuchangyuan, 2022-12-24:
after flash v39, same issue happen(both temp & humi display 0). 

flash https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_CGDK2_v1.1.1_0223.bin back make device work again.

pvvx, 2022-12-24:
There is no [PCB photo to support the new version of CGDK2](https://github.com/pvvx/ATC_MiThermometer/issues/273#issuecomment-1349720699). Perhaps the circuit related to the sensor has been changed.

yuchangyuan, 2022-12-24:
I use this sensor in bathroom, so I have do hardening, so it may not usefull.

[image]
[image]

joe-sydney, 2022-12-30:
I recently bought a few of these and I am experiencing this issue. Happy to open it up to take a picture of the PCB but it wasn't obvious how to do so. If that would be useful, please let me know how to dismantle it.

pvvx, 2022-12-31:
@yuchangyuan 
The capacitor in the power circuit is excluded. This affects the operation of the device = incomplete use of the battery life!
[image]
[image]

MatMaul, 2023-01-02:
@pvvx nice catch, any idea if/why it would affect sensor reading ? I am having the same trouble.

I can compile the firmware so let me know if you want me to try out a fix.

pvvx, 2023-01-02:
I don't think it affects the sensor.
Last fortune-telling on the version of CGDK2 that I do not have indicate the possibility of tricky behavior of the i2c bus. :)
Perhaps it is running a modified LCD controller occupying the i2c bus signals.

MatMaul, 2023-01-02:
> And reality says that no one needs it, because. anyone can remove all logs from any bus or just take a photo of a printed circuit board.

I did take the photos yesterday actually. I am not sure the quality is enough however, let me know if you need a clearer view of some parts.
https://photos.app.goo.gl/uaS9eQzTZX1BKiMx7

The sensor itself is definitively a SH3V.

Absolutely please stop buying more :) I have one and can help pretty easily.

I am not sure what you are asking regarding logs, I can easily change code and report. I however don't have any USB-UART around for now, but we can probably dump some values in the EEP to read them afterwards ? Or just dump some values in the ZigBee network frame and read them live ?

I can tinker with the code, but mostly new to this so need some guidance to what/where to look for.

MatMaul, 2023-01-02:
I did take the photos yesterday actually. I am not sure the quality is enough however, let me know if you need a clearer view of some parts.
https://photos.app.goo.gl/ccjWPhPaBh7bHwLh6

The sensor itself is definitively a SH3V.

Absolutely please stop buying more :) I have one and can help pretty easily.

I am not sure what you are asking regarding logs, I can easily change code and report. I however don't have any USB-UART around for now, but we can probably dump some values in the EEP to read them afterwards ? Or just dump some values in the ZigBee network frame and read them live ?

I can tinker with the code, but mostly new to this so need some guidance to what/where to look for.

pvvx, 2023-01-02:
So far the only thing you can find:
Marking = identity:
[image]
Differences:
[image]
[image]

pvvx, 2023-01-02:
[CGDK2_test.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGDK2_test.bin) - Firmware for SHTC3 sensor only. No auto-detect.

jerome-73, 2023-01-02:
> [CGDK2_test.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/CGDK2_test.bin) - Firmware for SHTC3 sensor only. No auto-detect.

Hi,

I've got the same issue and tested the CGDK2_test.bin.
Nothing changed. The display seems to work but displays 0 for temp and humidity.

MatMaul, 2023-01-02:
Same here.

I began adding some "tracing" in `read_sensor_cb` by setting `measured_data.temp`.
First observation, I can see my fake temperature updates in bluetooth, but not on the display, so here is another problem, or a related one.

 It seems like we never receive an ACK: I put 2 different fake temp value in https://github.com/pvvx/ATC_MiThermometer/blob/master/src/sensors.c#L105 and https://github.com/pvvx/ATC_MiThermometer/blob/master/src/sensors.c#L108, and it seems the second is never reached.

Not sure where to go next.

pvvx, 2023-01-04:
This means that the sensor has a different crystal and protocol than what is indicated on the label.
Those. it is necessary to remove the logs from the working sensor from the of.firmware.

joe-sydney, 2023-01-09:
Is there any chance that custom firmware will work on this version of the CGDK2? I just bought several of these and if it's never going to work, perhaps the documentation should be updated with a warning of the risk of being sold the new version.

pvvx, 2023-01-19:
It is possible that a problem with the sensor is detected if the board has protective diodes on the I2C bus. 
A similar situation arose with the new [MJWSD05MMC](https://pvvx.github.io/MJWSD05MMC/), but it was immediately resolved, since the devices are available, unlike the new CGDK2.
The test will be in a new beta version with the number 4.0 later, since no one has helped :)

1286840533, 2023-01-20:
> It is possible that a problem with the sensor is detected if the board has protective diodes on the I2C bus. A similar situation arose with the new [MJWSD05MMC](https://pvvx.github.io/MJWSD05MMC/), [...]

The latest firmware doesn't show the temperature
Status: Hardware Version: CGDK2 2.1.0, Software Version: 4.0
Vbat: 2833 mV , Temp: 0.00°C, Humi: 0.00%, ID: 0, Reed switch counter: 0, flg: 0x01:r1/t0

Sawadee2u, 2023-02-09:
I can confirm that beta 4.1 does not correct the temperature and humidity issue. Still displays "0" for both.
Hardware Version: CGDK2 2.1.0, Software Version: 4.1

pvvx, 2023-02-11:
[image]

Requires documentation or log taken from a working SHTV3 AAC29.

Sawadee2u, 2023-02-11:
I have the skills to do this but not the knowledge of this device interface.
What device do I need to interface with this device?
Can you point me to documentation?
Also. If you would like me to send you one (unopened) I will be happy to. Whatever I can do to help.

Giom-fm, 2023-02-16:
> after flash v39, same issue happen(both temp & humi display 0).
> 
> flash https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_CGDK2_v1.1.1_0223.bin back make device work again.

@yuchangyuan How did you manage to flash the device back. After the "E2" error im not able to connect to the device again.

```
19:08:36: Searching for devices
19:08:38: Connecting to: Qingping Temp RH Lite
19:08:42: Not found Telink OTA service!
19:08:42: Disconnected.
```

Sawadee2u, 2023-02-16:
I got the E2 notice also. I was able to reflash with the 223 bin after pulling the battery and replacing.

Giom-fm, 2023-02-16:
Unfortunately, I have had no success so far.

Sawadee2u, 2023-02-16:
Sorry to hear that. For what it’s worth, I was able to reprogram with 4.0 then 4.1 then back to233 bin. I only did one unit so may have been dumb luck.. I also held down the reset button for 10 seconds or do. This is the button that changes C to F. I believe the display flashed. Best of luck. Hoping for an update.I do have this working with Home Assistant using 233 bin and bind key.* * * * * * * * * * * * * * * * * * * * * * *“Beer is Proof that God Loves Us and Wants Us to Be Happy"  Benjamin Franklin* * * * * * * * * * * * * * * * * * * * * * *

mafrmt00, 2023-02-17:
The Sensor used is a SHTV3
[image]
The Datasheet is:
[HT_DS_SHTC3_D1.pdf](https://github.com/pvvx/ATC_MiThermometer/files/10764705/HT_DS_SHTC3_D1.pdf)

I took a quick look into I2c with the stock firmware.
[image]
[image]

As you can see, the address is 0x70 and the sleep/wakeup commands are according to the datasheet.

I hope this helps.

yuchangyuan, 2023-02-21:
> 

> @yuchangyuan How did you manage to flash the device back. After the "E2" error im not able to connect to the device again.

I can still connect to the sensor, maybe I replace battery as @Sawadee2u do, I forget the details. Replace with a new battery maybe necessary because flashing will consume a lot of power.

> ```
> 19:08:36: Searching for devices
> 19:08:38: Connecting to: Qingping Temp RH Lite
> 19:08:42: Not found Telink OTA service!
> 19:08:42: Disconnected.
> ```

mafrmt00, 2023-02-21:
> I can still connect to the sensor, maybe I replace battery as @Sawadee2u do, I forget the details. Replace with a new battery maybe necessary because flashing will consume a lot of power.

In my case, withe SW displaying E2 I only was able flash it via USB.

pvvx, 2023-02-21:
> As you can see, the address is 0x70 and the sleep/wakeup commands are according to the datasheet.

This is the protocol used in custom fw for CGDK2.
All versions for all types of thermometers use the same driver. The driver supports 2 types: SHTV3(SHTC3) and SHT4x.

mafrmt00, 2023-02-21:
> @mafrmt00 - What version of CGDK2 do you have?

The Version with the problem. It shows only zeros when flashed with alternative firmware and E2 with the older stock firmware. (also with the missing capacitor as described in this thread)

pvvx, 2023-02-21:
Logs of commands to measure and read data from the sensor are required.
The rest matches so far.

mafrmt00, 2023-02-21:
Ok, maybe this helps:
[image]

pvvx, 2023-02-21:
What it is: _Transfer at Address = 0x00?_

mafrmt00, 2023-02-21:
[image]
I have no idea, but it happens quite often:

pvvx, 2023-02-21:
That is, some kind of crooked sensor that does not work according to the description in the documentation.

mafrmt00, 2023-02-21:
I think we are getting somewhere.
General call is mentioned in the datasheet. But only for reset (0x0006). 
But the data shown in my picture (0x7ca2) is a valid command (Read T first)

So maybe instead of it's real address the sensor somehow only works with the general call address.

pvvx, 2023-02-21:
When creating a "General Calling Address", the next byte must indicate the address of the device.
But according to the oscillogram, everything is not so.

mafrmt00, 2023-02-21:
Sure, it violates every spec. It seems, just replace 0x70 with 0x00 for read commands.

pvvx, 2023-02-21:
How do I know which commands didn't have responses for sensors I don't have?

pvvx, 2023-02-21:
Test of operation on defective SHTv3 sensors.
CGDK2_v42.bin replaced.

mafrmt00, 2023-02-21:
It still displays only zeros.

_20:49:59: Received device infos are correct
20:49:59: Login successfull
20:50:06: File: CGDK2_v42_test.bin
20:50:06: File size: 78372 bytes
20:50:06: Count: 4899
20:50:08: Start DFU
20:50:29: Update done after 20.555 seconds
20:50:30: Disconnected.
20:51:17: Searching for devices
20:51:38: Connecting to: Qingping Temp RH Lite
20:51:50: Hardware Revision String: 2.1.0
20:51:50: Detected custom Firmware
20:51:50: Hardware Version: CGDK2 2.1.0, Software Version: 4.2, Sensor: SHTC3 (SHTV3)
20:51:50: Custom config HEX string: 55012000002804a9313186b4000000
20:52:07: File: Original_OTA_CGDK2_v1.1.1_0223.bin
20:52:07: File size: 116100 bytes
20:52:07: Count: 7257
20:52:11: Start DFU
20:53:10: Update done after 58.157 seconds
20:53:14: Disconnected._

[29 more replies]

closed 2023-05-07 by pvvx