pvvx/ATC_MiThermometer #276: MHO-C401 new revision - display frozen

evlo, 2022-12-08:
Hi, there seems to be new MHO-C401 with some dot or particle on the display and after flashing v3.8 and v3.9 beta display is frozen

[image]
[image]
[image]

https://pvvx.github.io/MHO_C401/

related https://github.com/pvvx/ATC_MiThermometer/issues/114

cgernert, 2022-12-08:
I also ordered four new Mijia MMC, but inside the box are the Azarton ones, with Azarton display. Box comes with Chinese text, owl and all  😁
Seems they switched some things up.
[image]

evlo, 2022-12-09:
Altrough even with this https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Azarton_MHO_C401_v1.0.0_0016.bin firmware screen is stuck.

pvvx, 2022-12-09:
> Seems they switched some things up.

[image]
[image]

cgernert, 2022-12-09:
Maybe there will be a future display version with the top icon, I don't know.

But I did not want to distract from evlo's issue.
Perhaps it is another controller for the display, but why the original firmware is working then? Very strange.
The screwdriver marks on the PCB can't be unseen. Luckily, there seems to be nothing under it. You really wanted that battery out of there, hm? 😂

evlo, 2022-12-09:
https://github.com/pvvx/TlsrComSwireWriter/issues/6

I don't get it, I think one is just bricked

pvvx, 2022-12-09:
> https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html says Hardware Version: LYWSD03MMC B1.6, Software Version: 3.9, Sensor: SHTC3 so now I'm a bit confussed, isn't LYWSD03MMC the small lcd one?

Why did you write firmware from Xiaomi LYWSD03MMC into it?

evlo, 2022-12-09:
I dunno, this is what the https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html put on the device, i noticed it too late, but whenever i tried after that it showed correct firmware as the original, maybe because before i was flashing LYWSD03MMC or maybe some stupid oversight on my side like with the read instead of write script

pvvx, 2022-12-09:
Firmware from Xiaomi LYWSD03MMC will work in MHO-C401. But will not display on the display. OTA will work and you can flash any other firmware. 
It is impossible to flash firmware from MHO-C401 or CGG1 to other options. Nothing will work.

evlo, 2022-12-09:
I have tried building https://github.com/kloemi/ATC_MiThermometer and flashing it to the device, that also results in frozen display, so probably really even different than [Azarton](https://github.com/kloemi/ATC_MiThermometerAzarton)

kloemi, 2022-12-12:
That's how it should work. Strange that display controller was updated again ...

cgernert, 2022-12-12:
> > Seems they switched some things up.
> 
> [image] [image]

Boom, there it is. I also built and flashed kloemi's firmware.
I don't know if this symbol is always present on the Azartons
[image]

evlo, 2022-12-12:
So for you display updates with kloemi's fork?

cgernert, 2022-12-12:
> So for you display updates with kloemi's fork?

Yes, it does. But the hardware of the devices' I got seems to be Azarton or related. (X-battery on photos)
https://github.com/pvvx/ATC_MiThermometer/issues/276#issuecomment-1343186448

pvvx, 2022-12-13:
To support Azatron, you need a [read original full flash](https://github.com/pvvx/ATC_MiThermometer/issues/114#issuecomment-979458611) and a description of all E-Ink segments.
[Samples description of all E-Ink segments.](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/epd_cgg1_2022.c#L16)
The command to output to the E-ink screen buffer is 
**60**b0b1b2...b13
where b0 is byte0, b1 is byte1, ...

evlo, 2022-12-13:
> > So for you display updates with kloemi's fork?
> 
> Yes, it does. But the hardware of the devices' I got seems to be Azarton or related. (X-battery on photos) [#276 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/276#issuecomment-1343186448)

Oh right i finally get the X thing, I was super confused by it but now I see that mine has flash or something instead of X.

BTW it does not even activate in Mi Home with region CN - it connects, but does not "verify" - it would be fun if somebody was doing fake version of these.

If display does work with the https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin firmware is it really necesary to do the dump? I'm very bad with soldering, that is why I'm asking.

kloemi, 2022-12-13:
I'm wondering that it works although the FPC is the same as in the old MH0_C401. 
Regarding the bluetooth symbols:
Azarton has both as shown in the documents, but I decided to use only the first at connected state.

And as mentioned before the display buffer seems not to contain a bit for the battery sign and the degree sign. No idea how to set them.

pvvx, 2022-12-13:
> This is with ble_hacker.bin still flashed on the device

Be careful with ble_hacker - it is designed to read and write Flash, as well as a one-time delete function with a old firmware startup restore.

pvvx, 2022-12-13:
> And as mentioned before the display buffer seems not to contain a bit for the battery sign and the degree sign. No idea how to set them.

Is the buffer size correct?

> The command to output to the E-ink screen buffer is
60b0b1b2...b13
where b0 is byte0, b1 is byte1, ...
https://github.com/pvvx/ATC_MiThermometer/issues/276#issuecomment-1347620089
60FFFFFFFFFFFFFFFFFFFFFFFFFFFF All On
600000000000000000000000000000 All Off

evlo, 2022-12-13:
> [I have such models](https://pvvx.github.io/MHO_C401/) and works with MHO_vXX.
I have tried flashing `MHO_C401_v38.bin` onto the one I extracted that FW from and it works flawlessly, i thought i received both same units, but clearly not, would be helpful to extract firmware from the one where the display freezes after flashing it with `MHO_C401_v38.bin`, but is working after flashing it with `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` using BLE_hacker or it would just basically get  `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` again?

pvvx, 2022-12-13:
I don't really understand what you are describing. BLE_hacker has the option to restore the launch of the previous firmware. But this button should be selected only once and disconnect the connection - disconnect...

kloemi, 2022-12-13:
> Is the buffer size correct?

Hmm you're right. Don't understand why I reduced it to 14 byte but in my recordings it seems to have 16 byte. Going to test.

evlo, 2022-12-14:
Hmmm, i thought the files would be identical, but there are some differences between two FF exports
https://1drv.ms/u/s!Ard77GaIVWRslMh15PZUTwMyzcdNkQ?e=FHuISp

***
_I have bought two MHO-C401

one is the one giving me issues, that i can flash using wired connection (if it feels like getting flashed) - it is very particular and even likes one ch340 flasher better then other (maybe i could try lowering baud rate) - I have fixed all my soldering. But I have tried soldering the RST pin, impossible for me, mayb ei have messed up something in the process.

from the non problematic one i uploaded extracted firmware from it in the post https://github.com/pvvx/ATC_MiThermometer/issues/276#issuecomment-1348813227

problematic one works with `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` firmware, but i can't connect it to mi app even - again, I could have messed up some soldering etc.

With `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` firmware it connects with [[Telink Flasher v4.7](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) , i have uploaded `BLE_hacker` to it, then connected using [[Telink OTA Hacker](https://pvvx.github.io/CGPR1/TLSR825xOTA_Hacker.html)](https://pvvx.github.io/CGPR1/TLSR825xOTA_Hacker.html) - that I think would only restore `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` I flashed to it that before, but the two files are not exactly the same - maybe just different tokens etc.

Honestly I now think that it is just MHO C401 with HW defect, no new version or anything, now I even have issues flashing it when I soldered wires nicely and using different usb uart adapters.

I'm curious if the extract is the same or not, especially since i flashed `Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin` onto it before. Maybe i do not understand how the flashing works in this case

Weirdly flashing work whether whichever way i connect rx and tx - rx to rx or rx to tx with 1k ohm resistor in the way_

pvvx, 2022-12-14:
The firmware azarton_mho-c410_ff.bin has only 87332 bytes from the beginning of Flash. And in the posted azarton_mho-c410_ff.bin, the firmware is intact and not changed.

pvvx, 2022-12-14:
I made a mistake: compared one, wrote another:
```
> fc /b "ff (problematic).bin" Original_OTA_Xiaomi_MHO_C401_v1.0.0_0010.bin 
Comparison files ff (problematic).bin and ORIGINAL_OTA_XIAOMI_MHO_C401_V1.0.0_0010.BIN
FC: ff (problematic).bin is longer than ORIGINAL_OTA_XIAOMI_MHO_C401_V1.0.0_0010.BIN
```

kloemi, 2022-12-15:
Increased buffer size to 16 did not fix it :-(

cgernert, 2022-12-15:
> Increased buffer size to 16 did not fix it :-(

Non-Azarton has 18, right? But Azarton has additional symbols compared to Non-Azarton.
Could you also please enable issues on your repo? Or should we create a new Issue on this repo, because this issue is originally based on evlo's Non-Azarzon freezing unit.

evlo, 2022-12-15:
Please dom, or continue this one https://github.com/pvvx/ATC_MiThermometer/issues/114 ?

If i understand correctly that one MHO-C401 does indeed have different firmware even when flash ORIGINAL_OTA_XIAOMI_MHO_C401_V1.0.0_0010.BIN onto it, so at this point mixing Azarton issues with it is really confusing.

But for that device I have that is having these issues, it is most likely not really worth investigating, until more turn out ...

MattXcz, 2023-01-17:
Today I flashed 3x MHO_C401 , all are still frozen =(

MattXcz, 2023-01-18:
Hi, I received 3 from Aliexpress today - all with new display version.

juskalalie, 2023-01-23:
> I also ordered four new Mijia Miaomiaoce MMC, but inside the box are the Azarton ones, with Azarton display. Box comes with Chinese text, owl and all 😁 Seems they switched some things up. [image] [image]

Got 4 copies exactly like yours after ordering from the Mijia SmartHome Store on AliExpress
https://www.aliexpress.com/store/2185008

MattXcz, 2023-01-23:
Mine was from "M-i M-ijia 18 Store"

pvvx, 2023-01-23:
> Got 4 copies exactly like yours after ordering from the Mijia SmartHome Store on AliExpress
> https://www.aliexpress.com/store/2185008

I ordered, waiting for delivery (weeks 3) ...

juskalalie, 2023-01-24:
Just to check, I flashed them with custom firmware -> screen got stuck. Was able to unstuck screen when flashing back to the original Azarton firmware available here.

pvvx, 2023-02-22:
[image]

Several [MHO-C401s (2022.11)](https://pvvx.github.io/MHO_C401N/) have arrived.

cgernert, 2023-02-22:
> 

nice!

pvvx, 2023-02-23:
Added support ver MHO-C401 (2022.11). 
Released beta version 4.2. Tested - all ok.

closed 2023-02-23 by pvvx

juskalalie, 2023-02-23:
Wow that was blazing fast!!
Many thanks for your efforts! 👍

pvvx, 2023-02-23:
The opportunity to make it faster was created thanks to @kloemi. He found how the signals from the SoC are distributed to the E-Ink driver chip. I did not even have to open the device, and I put the code from the CGG1 driver. The only thing I had to do was create a [map of the E-Ink segments](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/epd_mho_c401n.c#L17).

MattXcz, 2023-02-24:
Hello @pvvx  - I updated it to 4.2 beta, but sadly it still not update numbers on the display

MattXcz, 2023-02-24:
> > Hello @pvvx - I updated it to 4.2 beta, but sadly it still not update numbers on the display
> 
> Try the MHO_C401**N**_v42.bin

Yes. That did it. Now it works :) thank you !

homepersonalprojects, 2025-12-12:
> [pvvx/TlsrComSwireWriter#6](https://github.com/pvvx/TlsrComSwireWriter/issues/6)
> 
> I don't get it, I think one is just bricked, others seem to have succeeded with that tool
> 
> I think the marks on pcb are from the factory sticker under battery
> 
> [image] [image]
> 
> I think there is no chance I can solder the reset with my skills anyway

How do you reset at the hardware level? It's a shame that both the units I have got bricked in 2 different ways. One doesn't simply turn on and another connects to ZMA but display is frozen.