pvvx/ATC_MiThermometer #277: Telink Flasher not showing "Custom firmware" button

jsahoka, 2022-12-20:
Hi, 
I just received a batch of the Xiaomi temperature sensors. 
When I connect them in the Telink flasher I get the following output :

> 19:33:49: Searching for devices
> 19:33:53: Get Advertising MAC (Web Experimental Features!)
> 19:33:53: Connecting to: LYWSD03MMC
> 19:34:02: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 1, CtrID: 0x5830 Not bound, Standard certification, MAC: A4C1383A9EBC, Capability: 0x28 - front binding, IO: 0x0001
> 19:34:11: Hardware Revision String: B1.5
> 19:34:11: Detected Mi device
> 19:34:11: Connected
> 19:34:16: Activating now, please wait...
> 19:34:20: Registration successfull
> 19:34:20: Send Login, please wait...
> 19:34:20: Received device infos are correct
> 19:34:20: Login successfull

Device known id:
blt.3.129vSdhPZ7ATC

It does not show the "Custom Firmware" button, only the "Upload File" option.
Is this version even supported for custom firmware? If so, what file should I upload?

Thanks so much in advance!

ni9997, 2022-12-21:
Hi,
I have the same problem with two devices.
On the packaging it states the manufactoring date is 11.2022. And the online Tool also reads the hardware revision as B1.5.
Maybe this is a new version. 
I tried to flash it manually and it still showed the right temperature and humidity. I even could flash the "original" fimeware, but now I cannot connect to it using the webinterface anymore.
I dumped the firmware using the BLE-Hacker firmware. The dumb is attached. Maybe someone can do something usefull with that.
[B1_5.zip](https://github.com/pvvx/ATC_MiThermometer/files/10279985/B1_5.zip)

pvvx, 2022-12-24:
> I dumped the firmware using the BLE-Hacker firmware. The dumb is attached. Maybe someone can do something usefull with that. B1_5.zip

This is B1.5  "F2.0-CFMK-LB-JHBD---"  with FW [Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin)

pvvx, 2022-12-24:
> Hardware Revision String: B1.5

I don't have version B1.5 and it's not in the detection list to check the firmware version.

ni9997, 2022-12-24:
Thank you for your reply. I will try to write a patch, after I submitted my master thesis mid january, to get it working with B1.5.

pvvx, 2022-12-24:
The current version of Telink Flasher for version B1.5 and B1.8 offers to download universal custom firmware for LYWSD03MMC.

NerdyProjects, 2023-01-03:
I also recently received a batch of B1.5 hardware. The firmware works fine, and in the most recent flasher it shows me automatically either 3.8 or 3.9, so this is resolved at least for me.

HenrikNyberg, 2023-02-13:
I've got the same problem, a batch called 1.6 mfd 2022.09. Custom firmware button is not showing.

pvvx, 2023-02-15:
Was this thermometer flashed before using TelinkMiFlasher.html? Does his HW version not match the characters "B1.4" or is it similar to "B1.4   "?
[Trim the HW string in the new version.](https://github.com/pvvx/pvvx.github.io/commit/8cba2307b8e5df79dbe31f472e412129e0880a2b)

closed 2023-02-18 by pvvx