pvvx/ATC_MiThermometer #279: Support for XIAOMI Mijia Bluetooth Thermometer 3?

PetrMa, 2022-12-24:
Hello,

do you plan to support new XIAOMI Mijia Bluetooth Thermometer 3?

Thanks

pvvx, 2022-12-24:
Mijia Smart Thermometer 3 - id: MJWSD05MMC - The chipset is still unknown.
Alternative firmware is only supported on thermometers with Telink SoC (TLSR825x).
The rest of Xiaomi works in Mijia format, at https://github.com/custom-components/ble_monitor.

pvvx, 2023-01-14:
https://pvvx.github.io/MJWSD05MMC/

Rushmed, 2023-02-22:
Hi,

I am able to connct my MJWSD05MMC but I get stck on `Activating now, please wait...`
Any hint?

pvvx, 2023-02-23:
Use the side button to switch the sensor to pairing mode, the Bluetooth icon will flash on the screen.

Rushmed, 2023-02-24:
Thans it worked.

pvvx, 2023-03-05:
Full support of MJWSD05MMMC in version 4.2.

closed 2023-03-05 by pvvx

maiandrei, 2023-11-27:
> Hi,
> 
> I am able to connct my MJWSD05MMC but I get stck on `Activating now, please wait...` Any hint?

Hi, and many thanks for your great work!
I am also stuck with `Activating now, please wait...`, despite the fact I am connected to the device and it shows
Status: 'MJWSD05MMC HW:V2.3' connected, you can now Do the Activation to either get the Token or flash a new Firmware
Temp: 20.2°C, Humi: 45%, 3179 mV

Initially the BT icon flashes, but as soon as I connect to device it is steady on.

Could be the HW:V2.3 an issue?

pvvx, 2023-11-27:
The button was pressed at the wrong time or in the wrong sequence...