pvvx/ATC_MiThermometer #283: [feature request] dewpoint calculation

b3-4r, 2023-01-14:
Hi, I was wondering- would it be possible to add a dewpoint display (and perhaps logging) based on the drybulb and RH measurements? It's a pretty useful number to know to predict fogging and surface condensation.

Thanks a lot!

b3-4r, 2023-03-05:
Ah right, low power silicon. I'll figure out a way to do it offboard. Perhaps the bangle.js app could do the calculation and optionally beam it back to be displayed but that's a topic for another forum.

cultur98, 2023-04-05:
Hi there, 
thanks for bringing this up!
It actually refers to a question [I already asked a while ago](https://github.com/pvvx/ATC_MiThermometer/issues/29).
Victor provided a off device [download solution](https://pvvx.github.io/ATC_MiThermometer/DevPoint.html) which works fine.

But I was interested in live values.
So I wrote myself an ATC Thermometer to MQTT Gateway based on ESP32 and Arduino.
I also used this [code](https://github.com/pvvx/ATC_MiThermometer/tree/master/esp32) from Victor.
If you are interested in the code I can share it. 
(But be aware that is terrible .. spaghetti .. hardly commented .. but it works ... somehow.)

However I still like the idea the same as you @b3-4r to show the dew point **on** the device.
This would be a very cheap, usable  and portable solution!
A anti-mould-ometer in the pocket. So to say.

To calculate the dew point on the thingy internally I can think of three possible solutions:
1. Use a [lookup table](https://www.google.com/url?sa=i&url=https%3A%2F%2Fwaterleak.co.uk%2Fhelp-tips%2Fdew-point-chart%2F&psig=AOvVaw2DtbCQLmeCni5LaSKy6Qs4&ust=1680812951659000&source=images&cd=vfe&ved=0CA0QjRxqFwoTCID3gqvKk_4CFQAAAAAdAAAAABAD). 
This needs flash memory but does not need a lot of computational power. 
One probably would need to sacrifice a few values for the logger. (Which is also a great feature of this project!!)
2. Run the calculation on the device with [fixed precision math](https://github.com/PetteriAimonen/libfixmath). 
Does need some computational power and battery energy. 
But the dewpoint must not be calculated with every new value from the sensor. 
3. Use the [approximation](https://iridl.ldeo.columbia.edu/dochelp/QA/Basic/dewpoint.html) which is fine for relative humidity values higher than 50%.

My favorite is option 2. although (and maybe probably) I don't know whether it works.

cultur98, 2023-04-07:
Holiday and mediocre weather:
I managed to test the dew point calculation 
on one of my DEVICE_LYWSD03MMC
The implementation is quit a hack hijacking the show_clock() function.
But it works!

I use functions from [libfixmath](https://github.com/PetteriAimonen/libfixmath) and the [first formula](https://wikimedia.org/api/rest_v1/media/math/render/svg/e172e1a233ea113da9f7a276cdbc1df2a8b5a1f8) mentioned in the [Wikipedia article for the dew point](https://en.wikipedia.org/wiki/Dew_point).

It shows temperature/humidity and the dew point (and "-1" below) alternating.

[image]
[image]

cultur98, 2023-04-07:
Code for the dew point calculation the log() eats the time
```
// the fixed point number functions have been copied from
// the library https://github.com/PetteriAimonen/libfixmath
// Great thanks !!

typedef int32_t fix16_t;

static const fix16_t fix16_one = 0x00010000; /*!< fix16_t value of 1 */
static const fix16_t fix16_e   = 178145;     /*!< fix16_t value of e */

static const fix16_t fix16_maximum  = 0x7FFFFFFF; /*!< the maximum value of fix16_t */
static const fix16_t fix16_minimum  = 0x80000000; /*!< the minimum value of fix16_t */
static const fix16_t fix16_overflow = 0x80000000; /*!< the value used to indicate overflows when FIXMATH_NO_OVERFLOW is not specified */

static inline fix16_t fix16_from_int(int a)     { return a * fix16_one; }
static inline float   fix16_to_float(fix16_t a) { return (float)a / fix16_one; }

static inline int fix16_to_int(fix16_t a)
{
#ifdef FIXMATH_NO_ROUNDING
    return (a >> 16);
#else
    if (a >= 0)
        return (a + (fix16_one >> 1)) / fix16_one;
    return (a - (fix16_one >> 1)) / fix16_one;
#endif
}

static inline fix16_t fix16_from_float(float a)
{
    float temp = a * fix16_one;
#ifndef FIXMATH_NO_ROUNDING
    temp += (temp >= 0) ? 0.5f : -0.5f;
#endif
    return (fix16_t)temp;
}

static inline uint32_t fix_abs(fix16_t in)
{
    if(in == fix16_minimum)
    {
        // minimum negative number has same representation as
        // its absolute value in unsigned
        return 0x80000000;
    }
    else
    {
        return ((in >= 0)?(in):(-in));
    }
}

fix16_t fix16_add(fix16_t a, fix16_t b)
{
    // Use unsigned integers because overflow with signed integers is
    // an undefined operation (http://www.airs.com/blog/archives/120).
    uint32_t _a = a;
    uint32_t _b = b;
    uint32_t sum = _a + _b;

    // Overflow can only happen if sign of a == sign of b, and then
    // it causes sign of sum != sign of a.
    if (!((_a ^ _b) & 0x80000000) && ((_a ^ sum) & 0x80000000))
        return fix16_overflow;
    
    return sum;
}

fix16_t fix16_sub(fix16_t a, fix16_t b)
{
[... 241 lines cut ...]
    return guess + fix16_from_int(scaling);
}

int16_t calc_dew_f1(int16_t t_e2, int16_t rh_e2) {
	// the next four initializations have to be done only once 
	// could run in an init function to safe time
  fix16_t f_100 = fix16_from_int(100);
  fix16_t f_10000 = fix16_from_int(10000);
  fix16_t f_b = fix16_from_float(18.678);
  fix16_t f_c = fix16_from_float(257.14);

	// convert input values to fix16_t
  fix16_t T  = fix16_from_int(t_e2);
  T = fix16_div(T, f_100);
  fix16_t RH = fix16_from_int(rh_e2);
  fix16_t RH_100 = fix16_div(RH, f_10000);

	// the actual calculation of the dewpoint
  fix16_t gamma = fix16_add(fix16_log(RH_100), fix16_div(fix16_mul(f_b, T), fix16_add(f_c, T)));
  fix16_t Tdp = fix16_div(fix16_mul(f_c, gamma), fix16_sub(f_b, gamma));
  fix16_t i_Tdp = fix16_mul(Tdp, f_100);
  return (fix16_to_int(i_Tdp)); // in Celsius
}

```

CraigHutchinson, 2023-06-12:
@cultur98 Great work. You'd greatly optimize code size & performance this by using a power-of-two scaling factor as these become simple shift operators etc. https://en.wikipedia.org/wiki/Fixed-point_arithmetic#Choice_of_scaling_factors