pvvx/ATC_MiThermometer #285: Discrepancy between displlayed and advertised values at low temperatures

nilspak, 2023-01-17:
I have a LYWSD03MMC sensor flashed with the 3.9 custom firmware. There seems to be a divergence between displayed values and actual values at colder temperatures (below freezing). It seems to happen both when displaying in C or in F. The advertised temperature will remain correct, and different from the displayed temp (up to 10 degrees different). I tried returning it to the stock firmware and that seemed to fix the display issue. Is this a known bug/Has anyone else experienced this?

pvvx, 2023-01-17:
[image]

nilspak, 2023-01-18:
That makes a lot of sense, though when I look back at what data I have, the battery voltage doesn't seem to drop below 2.95V (at ~ -9C). I have only a couple of days of data and am seeing a similar looking pattern to the graph above, though the low end voltage is still pretty high at the lowest temp.  
I don't have another LYWSD03MMC to experiment with at the moment but I will try to pick some more up. 
And, thanks for all your work on this xiaomi sensor platform!

closed 2023-02-18 by pvvx