pvvx/ATC_MiThermometer #286: Battery reporting issue after updating to 3.9

gminadak, 2023-01-17:
Hello.
First of all thank you for all the effort you have put in this project.
After upgrading to firmware 3.9 I see that the battery level is at 100% for both of my thermometers.
Before the one of them was at approximately 80%.
I'm using the BThome integration with an on board BTdevice(Laptop).
[image]
[image]

pvvx, 2023-01-17:
The correction is made taking into account the characteristics of CR2032 and similar elements.
A new battery without load puts out more than 3 volts. At light load, during the first days, the voltage drops to about 2.9 V and this continues for 80-90% of the battery life. Two years of experience in the operation of thermometers led to the fact that
readings need to be corrected.

gminadak, 2023-01-17:
@pvvx
> Your battery is draining too much. This indicates that you are using a BTHome with "active scanning" or poor quality batteries.
The battery's is what came with the device.
I see under Bluetooth integration on HA that "Passive scanning" is not selected.
Do you suggest to enable it?
[image]

pvvx, 2023-01-17:
Active scanning requires the device to respond. This is additional consumption. In addition, there is a constant overwriting of files on the disk, which renders the SSD unusable.
In the current implementation of BTHome, not all devices work in the "passive scan" mode.
To work with BTHome in passive mode, you need to enable "AdFlags" in TelinkMiFlasher:
[image]

gminadak, 2023-01-17:
I have enable "passive scan" and AdFlags, the only BT device's I'm using is this two thermometers for now.
It looks like the BThome wasn't the best choice after all, I will look for other options.
Thank you for the help.

closed 2023-01-17 by gminadak