pvvx/ATC_MiThermometer #287: SJWS01LM support?

ilgrank, 2023-01-23:
Hi
I see that only thermomethers are supported, but then in telink flashed there' also a water sensor, SJWS01LM between the device ids.
It is supported for Mi-unlocking only, right?
Do you know if it will be possible to flash it in the future or it does have a different chipset?

alessocci, 2023-01-24:
I too would be very interested in the future implementation of the SJWS01LM flood detector.
Thanks

pvvx, 2023-01-25:
This requires a hardware revision of the thermometer - external contacts and sealing of the case. Other than that, no problem...

closed 2023-01-25 by pvvx

ilgrank, 2023-01-30:
@pvvx : I can offer support for users, I do speak english and I'm willing to help

pvvx, 2023-01-30:
@ilgrank - Для создания прошивки и программного обеспечения к нему необходимо иметь сам SJWS01LM. У меня его пока нет.
И в стадии разработки альтернативной прошивки сейчас находится [MJWSD05MMC](https://pvvx.github.io/MJWSD05MMC/). Он тоже имеет отличия с текущими термометрами. Сделать всё универсальное одному сложно - это требует много времени. 
И, если создавать совместную поддержку пользователей, то потребуется не только знание разных языков, но и опыт эксплуатации, тестов и доработки программного обеспечения для поддерживаемых датчиков. Я не против объединиться, но ещё не имею достаточного опыта совместной работы в github.

ilgrank, 2023-01-31:
@pvvx Привет
Что касается одного, я могу купить его и отправить на ваш адрес, если вы предоставите его мне в частном порядке.
Я не программист, но могу помочь вам ответить на некоторые вопросы. Конечно, у меня нет вашего представления о коде, поэтому, возможно, я могу выступать только в качестве первого уровня поддержки.
Я спросил вас, собираетесь ли вы его поддерживать, так как с первого взгляда мне кажется, что это просто термометр МИ без датчика температуры/влажности и показания реле (я имею в виду, что вы уже внедрили 90% код уже для этого)
Дайте мне знать, если я могу быть чем-то полезен

pvvx, 2023-01-31:
Я уже заказал SJWS01LM на https://aliexpress.com/item/1005001814960091.html
Обычно доставка составляет 3..4 недели.

ilgrank, 2023-01-31:
Это отличная новость, я буду следить здесь, если я могу чем-то помочь, просто @упомяните меня, и я свяжусь с вами
спасибо

pvvx, 2023-03-07:
The case is not disassembled - on glue.
https://pvvx.github.io/SJWS01LM/

ilgrank, 2023-03-09:
Thanks for dissecting it!
I also seen you added support for it in Telink Flasher.. so it is very similar to other ATC devices after all?

ilgrank, 2023-10-16:
> The case is not disassembled - on glue. https://pvvx.github.io/SJWS01LM/ [image]

Hi @pvvx , has there been any progress on SJWS01LM?
I seen you merged #310 some time ago, if I understand correcty, battery level can be read, but alarm can not, right?
Thanks!