pvvx/ATC_MiThermometer #289: Округление значений на кастоме PVVX LYWSD03MMC

KaportsevIA, 2023-01-25:
Доброго времени суток. Можно сделать стандартным округление значений температуры и влажности на кастомной прошивке для термометра LYWSD03MMC. До 1 и 0 (соот-но) знака после запятой, т.е. Температура 26,1, влажность 47 например. Смена формата в флешере почему-то не работает. А писать под каждый датчик конвертер с системах УД немного геморройно, тем более что датчиков может быть много. До 10-15 штук.

pvvx, 2023-01-29:
Стандартным с кем или чем?
У каждого свой стандарт.
При округлении до указанных значений выходят очень кривые и скачкообразные графики в Home Assistan. Невозможно на это смотреть:
[image]
Хотя интеграция пытается сгладить и усреднить.
А без усреднения всё ещё хуже:
[image]

KaportsevIA, 2023-01-30:
Честно говоря, не вижу смысла в применении сотых и тысячных долей в градусах/влажности в сценариях умного дома. Те кому нужны датчики, например влажности почвы, скорее всего используют что-то более специализированное, умеющее считать еще и фертильность почвы. А большинство климатической техники (сплиты/обогреватели/осушители/увлажнители) дай бог реагируют на десятые доли. Т.е. такой сценарий когда показатели температуры будут плавать в пределах сотых долей, например от 26.499 д до 27.002 вполне реальны, а если сплит (который реагирует на , 0,5 )настроить на выключение при 27??? И у них нет параметров гистерзиса. Ну и кроме того смена показателей с 26,5, на 26,51 - меняется размер шрифта.  
[image]

pvvx, 2023-01-30:
> Честно говоря, не вижу смысла в применении сотых и тысячных долей в градусах/влажности в сценариях умного дома. 

Вам же показаны графики управления обогревом. И передачи значений в 0.01 C уже мало для многих применений.
Требуется не абсолютная точность, а дифференциальная и как можно больше значений в секунду. Особенно для ПИД  регулирования.

> Ну и кроме того смена показателей с 26,5, на 26,51 - меняется размер шрифта. 

Это не термометр меняет, а ваша программа. И большинство программ выводят усредненное значение, полученное путем усреднения нескольких принятых значений.
И если приято 22.7 девять раз и 22.8 то индикация будет 22.71.
Как это исправить термометру?

closed 2023-01-30 by pvvx