pvvx/ATC_MiThermometer #290: Can't connect after flash

Nahanna, 2023-01-26:
I want connect after lfash to configure options and doesn't works.

8:53:11: Searching for devices
8:53:21: Connecting to: LYWSD03MMC
8:53:24: Detected Mi device
8:53:24: Hardware Revision String: B1.4
8:53:24: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 1 from 5
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 2 from 5
8:53:25: Detected Mi device
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 3 from 5
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 4 from 5
8:53:25: Detected Mi device
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 5 from 5
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Something went wrong, too many reconnects
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 1 from 5
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 2 from 5
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 3 from 5
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 4 from 5
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:25: NotSupportedError: GATT operation failed for unknown reason.
8:53:25: Reconnect 5 from 5
8:53:25: Hardware Revision String: B1.4
8:53:25: Detected Mi device
8:53:26: NotSupportedError: GATT operation failed for unknown reason.
8:53:26: Something went wrong, too many reconnects

pvvx, 2023-01-29:
Try changing the battery.

Nahanna, 2023-01-31:
Thnx fot answer but not works. Changed battery, hard reset... Maybe corrupted.

pvvx, 2023-01-31:
Flashing via BLE does not damage the thermometer. Interrupting the update does not damage the thermometer and the checksum is checked during firmware. Incorrectly written code will not be perceived as an update.
Try using another smartphone or BT adapter to connect.

pvvx, 2023-01-31:
Incorrect settings can be reset in "nrfConnect":

[image]

[image]

Nahanna, 2023-01-31:
Done but same.

13:08:10: Searching for devices
13:08:18: Connecting to: ATC_CA6F68
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Reconnect 1 from 5
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Reconnect 2 from 5
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Reconnect 3 from 5
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Reconnect 4 from 5
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Reconnect 5 from 5
13:08:40: Hardware Revision String: B1.4
13:08:40: Detected Mi device
13:08:40: NotSupportedError: GATT operation failed for unknown reason.
13:08:40: Something went wrong, too many reconnects
13:08:51: Disconnected.

 I will check with other Blt device.

BunpGhost, 2023-02-15:
Hi
I was having this issue recently and I've change my batteries to new ones and now it works fine.

closed 2023-02-18 by pvvx