pvvx/ATC_MiThermometer #294: Problem with battery calculation and resulting broadcast

AleXSR700, 2023-02-04:
Hello,

I just updated to the latest firmware a week ago and now one of my Xiaomi Mijias is no longer sending all the information.

I went to check it out and saw that the display is showing 0% battery. Obviously impossible if the device is still running ^^

This then leads the firmware to not send a battery value (imho it should then send 0% as value instead of no value).

```
11:27:38.438 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:38","ATC76598c":{"alias":"xiaomi_master_bedroom","mac":"a4c13876598c","Temperature":15.7,"Humidity":82.7,"DewPoint":12.8,"Btn":1,"Battery":83,"RSSI":-92},"TempUnit":"C"}
11:27:39.197 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:39","ATC801437":{"alias":"xiaomi_dining_room","mac":"a4c138801437","Temperature":19.3,"Humidity":64.2,"DewPoint":12.4,"Btn":1,"Battery":86,"RSSI":-78},"TempUnit":"C"}
11:27:40.767 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:40","ATC0d5c0c":{"alias":"xiaomi_office","mac":"a4c1380d5c0c","Temperature":16.1,"Humidity":69.8,"DewPoint":10.6,"Btn":1,"Battery":79,"RSSI":-62},"TempUnit":"C"}
11:27:42.819 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:42","ATC8f1b4f":{"alias":"xiaomi_kitchen","mac":"a4c1388f1b4f","Temperature":16.8,"Humidity":72.7,"DewPoint":11.9,"Btn":1,"Battery":91,"RSSI":-80},"TempUnit":"C"}
11:27:49.170 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:49","ATC801437":{"alias":"xiaomi_dining_room","mac":"a4c138801437","Temperature":19.3,"Humidity":64.2,"DewPoint":12.4,"Btn":1,"Battery":86,"RSSI":-78},"TempUnit":"C"}
11:27:49.871 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:49","ATCf28078":{"alias":"xiaomi_living_room","mac":"a4c138f28078","Temperature":18.4,"Humidity":74.1,"DewPoint":13.6,"Btn":1,"Battery":95,"RSSI":-83},"TempUnit":"C"}
11:27:50.323 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:50","ATC80f246":{"alias":"xiaomi_walk_in","mac":"a4c13880f246","Temperature":16.0,"Humidity":76.8,"DewPoint":11.9,"Btn":1,"Battery":70,"RSSI":-82},"TempUnit":"C"}
11:27:51.674 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:51","ATC801437":{"alias":"xiaomi_dining_room","mac":"a4c138801437","Temperature":19.3,"Humidity":64.2,"DewPoint":12.4,"Btn":1,"Battery":86,"RSSI":-78},"TempUnit":"C"}
11:27:52.420 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:52","ATCf28078":{"alias":"xiaomi_living_room","mac":"a4c138f28078","Temperature":18.4,"Humidity":74.1,"DewPoint":13.6,"Btn":1,"Battery":95,"RSSI":-85},"TempUnit":"C"}
11:27:52.925 MQT: stat/tasmota_D2EE70/SENSOR = {"Time":"2023-02-04T11:27:52","ATCe28b41":{"alias":"xiaomi_master_bathroom","mac":"a4c138e28b41","Temperature":16.6,"Humidity":74.5,"DewPoint":12.0,"Btn":1,"RSSI":-81},"TempUnit":"C"}
11:27:52.936 A4C138E28B41 = xiaomi_master_bathroom  does not exist and will be created
11:27:52.938 BRY: Exception> 'key_error' - Battery
11:27:52.940 stack traceback:
11:27:52.941 	<native>: in native function
11:27:52.943 	<unknown source>: in function `_anonymous_`
11:27:52.944 	<unknown source>: in function `try_rule`
11:27:52.945 	<unknown source>: in function `exec_rules`
11:27:52.947 	<unknown source>: in function `event`
```
As you can see, the others are sending the "Battery" whereas the bathroom one is not.

Is the first issue caused by the new calculation method?

The second looks like a bug in the firmware :)

Thank you
Alex

pvvx, 2023-02-13:
Looks like a bug in tasmota.

---

Reduce the voltage to 2.06V on the go:
```
22:31:44: Requesting Bluetooth Scan...
22:31:46: > Requested ATC_0B5EED
22:31:46: Watching advertisements from "ATC_0B5EED"...
22:31:54: ServiceData Data: 0000181c-0000-1000-8000-00805f9b34fb, 0200062302D5080303840D020108
22:31:54: HA-BLE, PacketId: 6, temperature: 22.610°C, humidity: 34.600%, battery: 8%
22:32:09: ServiceData Data: 0000181c-0000-1000-8000-00805f9b34fb, 02000A2302D50803038A0D020100
22:32:09: HA-BLE, PacketId: 10, temperature: 22.610°C, humidity: 34.660%, battery: 0%
22:32:22: ServiceData Data: 0000181c-0000-1000-8000-00805f9b34fb, 02000C2302D5080303860D020100
22:32:22: HA-BLE, PacketId: 12, temperature: 22.610°C, humidity: 34.620%, battery: 0%
22:32:34: ServiceData Data: 0000181c-0000-1000-8000-00805f9b34fb, 02000E2302D1080303820D020100
22:32:34: HA-BLE, PacketId: 14, temperature: 22.570°C, humidity: 34.580%, battery: 0%
```
[image]

All ok.

PS: Cutoff will be at 2.0V with recovery pending at 2.2V

pvvx, 2023-02-15:
> As you can see, the others are sending the "Battery" whereas the **bathroom** one is not.

Maybe just a dead battery or a humidity problem?

AleXSR700, 2023-02-15:
> > As you can see, the others are sending the "Battery" whereas the **bathroom** one is not.
> 
> Maybe just a dead battery or a humidity problem?
> 
> Flashing requires additional energy from the battery, and if it is already discharged, then it is completely over.

What are you referring to?
The device is still running (display is on) but I can no longer see it in tasmota (no the flasher).
So either the hardware is now undersupplied or the firmware is "protecting" the device by disabling ble now.

pvvx, 2023-02-15:
The thermometer has several advertised formats.
Third-party programs may not accept all options. Check the compatibility of the selected thermometer format in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).
https://github.com/pvvx/ATC_MiThermometer/issues/295#issuecomment-1428486547

AleXSR700, 2023-02-15:
I have not experienced any connection issues between that device and the esp32. I think I can rule that out as a cause.

I also cannot see the device using the flasher, which led me to the conclusion that the device is no longer sending BLE packages. 
The display now only shows the temperature value and the 0% battery. It is no longer switching displays.

Of course I could change the battery, but I am trying to find out what is causing low battery (but sufficient for operstion)  to no longer work a) with tasmota properly and b) at some point stop ble operation.

pvvx, 2023-02-18:
> Of course I could change the battery, but I am trying to find out what is causing low battery (but sufficient for operstion) to no longer work a) with tasmota properly and b) at some point stop ble operation.

This does not apply to the general firmware version of thermometers.
I can't help you with these experiments.

AleXSR700, 2023-02-18:
Well, the cool firmware you wrote stops ble when battery is low. So I think you can help with that.
There would be no problem with temperature and humidity if the firmware did not stop ble just because it wrongly thinks the battery is empty.

I would say that the battery calculation you implemented is not correct and causes wrong behavior.

pvvx, 2023-02-18:
These are just your speculations.
The limit on the lower supply voltage level is the same for all versions. On the official firmware from Xiaomi, the limitation is much larger.
If you limit even lower, then in 90% of cases this leads to a sharp reboot with the chip freezing with constant consumption and a sudden battery landing.
The current algorithm, built many versions ago, makes it possible for the thermometer to work from a solar panel as well.

Writing to Flash is limited to 2 volts.
The CR2032 has a very high internal resistance and a voltage drop occurs when the chip is running.
Most of the new Xiaomi LYWSD03MMC and CCGD2 have a display control chip that requires a supply voltage of 2.6 V.
On older versions of LYWSD03MMC, at a voltage of less than 2.6V, the LCD screen is already gray and nothing is visible on it.

If you have the equipment, you can check all conditions.

closed 2023-02-18 by pvvx

AleXSR700, 2023-02-19:
I am not inventing anything. It is called observation.
Feel free to reproduce using an adjustable power source.

pvvx, 2023-02-20:
> Feel free to reproduce using an adjustable power source.

Thermometers do not have built-in laboratory power supplies.
All calculations are made to work with the existing characteristics of the CR batteries.

AleXSR700, 2023-02-20:
I am not sure how the temperature factors in.
For measurement I of course took out the battery and was measuring in a warmer room (~ +3 °C) and measured 2.96 V there.
After reinserting the device ran over night in that room with working BLE. I now returned it to the colder room and it is again/still working.

So for me the question remains if somewhere in the code there is a saftey (?) function that limits functionality based on calculated battery voltage.
If you say that that is not the case and the BLE unit is simply underpowered and stops working, then okay. Then that is simply tough luck.
I am just checking because the display behavior changes with low battery, so the code checks and changes behavior. If at no point BLE is impacted, then there is nothing that can be done.

pvvx, 2023-02-20:
> For measurement I of course took out the battery and was measuring in a warmer room (~ +3 °C) and measured 2.96 V there.

This is not the correct measurement. CR batteries always put out about 3V with no load (+25C).

AleXSR700, 2023-02-20:
No restarts or crashes observed. Only a display that shows low battery and a lack of ble communication.

pvvx, 2023-02-20:
And what should a thermometer do if the battery is no longer capable of delivering voltage and current for transmission? Show blank screen?

AleXSR700, 2023-03-01:
Bluetooth should transmit until battery is too low to be able to transmit. At that point it would stop transmitting.
If it stops sooner, then something is disabling BLE.

The issue has nothing to do with tasmota as the Mijia is not broadcasting anymore. No device can pick up anything if there is not transmission to pick up.

Tasmota could handle 0% values, but unfortunately that is not wanted. 
Question is if it would not make more sense to actually roundup all values so that "0" is never transmitted as it is a false/pointless value.

pvvx, 2023-03-01:
Once again - 0% does not mean that the LCD screen cannot work. 0% is the value, then there is no energy for transmission and full operation of the device.

AleXSR700, 2023-03-01:
And once again: I am aware of what it is supposed to be but in reality transmissions do not end on calculated 0%.

So broadcasting 0% is not sensible. It would be better to use roundup for 0.x% to ensure there is always a valid value being transmitted. If true 0 is reached, then as you say, transmission will end anyway.

pvvx, 2023-03-01:
Telink Semiconductor SDK Development Handbook

8.3.1 Low voltage detection protection 
Combine with the introduction of low power protection chapter, it is necessary to consider doing voltage detection before all Flash write and erase operations to avoid the situation of operating Flash at too low  voltage. In addition, in order to ensure that the system is always working at a safe voltage, it is also recommended to do low voltage detection in main_loop at regular intervals to ensure the normal operation of the system.
Note:
• About flash low-voltage protection, the following many places appear 2.0V, 2.2V and other thresholds, emphasize that these values are only examples, reference values. Customers have to assess the actual situation to modify these thresholds, such as single-layer boards, power supply fluctuations and other factors, are to improve the safety threshold as appropriate.
Take the low voltage detection in the SDK demo as an example:

Step 1 First, when powering on or waking up from deepsleep, before calling the Flash function, a low voltage test must be performed to prevent flash problems caused by low voltage:
Considering the working voltage of MCU and the working voltage of flash, if the Demo is set below 1.8V, the chip will directly enter suspend, and the chip below 1.8~2.0V will directly enter deepsleep, and once the chip is detected to be lower than 2.0V, it needs to wait until the voltage rises to 2.2V, the chip will resume normal operation. Consider the following points in this design:
• At 1.8V, there is a risk that the voltage is lower than the flash operating voltage. When you wake up after entering deepsleep, it may cause the flash to be abnormal and crash, so enter the suspend below
1.8V to ensure the safety of the chip;
• At 2.0V, when other modules are operated, the voltage may be pulled down and the flash will not work normally. Therefore, it is necessary to enter deepsleep below 2.0V to ensure that the chip no longer runs related modules;
• When there is a low voltage situation, need to restore to 2.2V in order to make other functions normal, this is to ensure that the power supply voltage is confirmed in the charge and has a certain amount of
power, then start to restore the function can be safer.
The above is the timing detection voltage and management method in SDK Demo, users can refer to it for design.
Note:
• About flash low-voltage protection, the threshold values that appear above are only reference values.
Customers have to assess the actual situation to modify these thresholds, such as single-layer boards, power supply fluctuations and other factors, are to improve the safety threshold as appropriate.

pvvx, 2023-03-01:
CRxxx batteries do not have a complete zero. Battery voltage depends on load, temperature and many other factors. No one in the world has been able to get the values of the residual energy in CRxxx batteries. I invite you to do scientific work after you know how the battery works. :)

AleXSR700, 2023-03-01:
> CRxxx batteries do not have a complete zero. Battery voltage depends on load, temperature and many other factors. No one in the world has been able to get the values of the residual energy in CRxxx batteries. I invite you to do scientific work after you know how the battery works. :)

Exactly the reason why the firmware should never report 0% ;)

pvvx, 2023-03-01:
At your request, I will remove any indication and transmission of% battery in the new thermometer firmware.
It will be better this way?

AleXSR700, 2023-03-01:
Please don't remove battery :)
Just use the roundup function (if supported) rather than accurate number rounding :)
That way the lowest value will always be 1%.

pvvx, 2023-03-01:
You have not provided technical data. For example, in voltage thresholds, when all users of various thermometers consider that the battery is 0% discharged. Until you do this, further conversation is useless.

AleXSR700, 2023-03-01:
I think you are misunderstanding.

The technical data is irrelevant. The technical data is used for the calculation. The calculation is fine. But the reporting is the issue.
So once you calculated 0.x % (with x<5) you broadcast 0%.
Instead you should do roundup(0.x) resulting in "1".
That way the value of the next bigger integer is broadcasted.

This has nothing to do with the calculation of the value. Only the broadcasting/displayal is altered due to the roundup of the calculated value.

pvvx, 2023-03-02:
Please tell me. We take CR2032 and measure the voltage with a tester - readings are 2.9V. Is it 0% or 100%?

AleXSR700, 2023-03-02:
I give up. You clearly are not getting it.

pvvx, 2023-03-02:
4. If you take two rulers and shift one by 0.5 divisions, which one do you think rounds off correctly?

AleXSR700, 2023-03-03:
Accuracy has nothing to do with the issue. You are still arguing things that have nothing to do with the issue abs the oh so simple fix.

pvvx, 2023-03-03:
I understand that you want to enter your own rules for recording the percentage of battery charge.
But this is only your desire, which does not meet any standards.
And if the operation of the thermometer is accompanied by your rules, then severe failures are guaranteed.