pvvx/ATC_MiThermometer #295: Flashing and connecting is stable, advertising not working

rjvtww, 2023-02-12:
I got a brand new LYWSD03MMC B1.7, and flashed it first with ATC1441, after which I realised that pvvx was newer and flashed it with that one.

Process went fine (OTA), and connecting to the device is stable and I can change settings. The only setting I changed was the advertising format to ATC1441. Sadly I cannot pick up any advertising, either using my RPI setup or using the HTML tool (using the same pc and bt adapter that I use to connect and flash the device).

Any thoughts on what I can check?

pvvx, 2023-02-13:
[image]

rjvtww, 2023-02-13:
Can you explain what does AdFlags do?

Thanks!

pvvx, 2023-02-13:
Enables additional flags in the ad pack and battery consumption increases.
This option is made to support older third party programs...

rjvtww, 2023-02-17:
I was able to get it working, but on my setup (RPI4, node red), I get a fixed value for a long time, as if the signal was not changing at all. Here is a screenshot of the last 24 hours as an example.

[image]

pvvx, 2023-02-18:
I don't have an RPi4 and the RPi4 BT adapter is not the best.

closed 2023-02-18 by pvvx