pvvx/ATC_MiThermometer #297: Работа термометров в Home Assistant c Bluetooth LE Long Range.

pvvx, 2023-02-17:
**Предварительное описание решения.**

Для примера можно взять: 
Low-cost Realtek RTL8761 BT adapter (current version of Linux fw on RTL8761 supports BT5.1)
[image]

Для работы HA c LE Long Range в интеграции “Bluetooth” желательно установить последнюю версию bluez и добавить в файле /lib/systemd/system/bluetooth.service
_ExecStart=/usr/local/libexec/bluetooth/bluetoothd_ **--experimental**

Перед запуском Home Assistant необходимо переключить USB-BT5.0+ адаптер на работу в CodedPHY:
_hcitool -i hci1 cmd 08 31 03 04 04_
Номер hciN - свой, желательно от второго адаптера, а первый может принимать стандартные рекламные пакеты BT4.2.
Если адаптер будет работать неустойчиво, тогда HA может его переинициализировать и настройка на работу в Coded PHY (Long Range) собьется. Ожидать официальной поддержки в интеграции “Bluetooth” нет никакой возможности. Выбранный автором API “Bleak” не позволяет управлять PHY адаптера и не предвидится никакой поддержки BLUETOOTH SPECIFICATION Version 5.0 от 2016 года.
```
BLUETOOTH SPECIFICATION Version 5.0 | Vol 1, Part C page 291
Core Specification Change History
9 CHANGES FROM v4.2 TO 5.0
9.1 NEW FEATURES
Several new features are introduced in the Bluetooth Core Specification 5.0
Release. The major areas of improvement are:
• Slot Availability Mask (SAM)
• 2 Msym/s PHY for LE
• LE Long Range
• High Duty Cycle Non-Connectable Advertising
• LE Advertising Extensions
• LE Channel Selection Algorithm #2 
```
Проверить адаптер на совместимость возможно с помощью команд:
```
# btmgmt phy
Supported phys: BR1M1SLOT BR1M3SLOT BR1M5SLOT EDR2M1SLOT EDR2M3SLOT EDR2M5SLOT EDR3M1SLOT EDR3M3SLOT EDR3M5SLOT LE1MTX LE1MRX LE2MTX LE2MRX LECODEDTX LECODEDRX
Configurable phys: BR1M3SLOT BR1M5SLOT EDR2M1SLOT EDR2M3SLOT EDR2M5SLOT EDR3M1SLOT EDR3M3SLOT EDR3M5SLOT LE2MTX LE2MRX LECODEDTX LECODEDRX
Selected phys: BR1M1SLOT BR1M3SLOT BR1M5SLOT EDR2M1SLOT EDR2M3SLOT EDR2M5SLOT EDR3M1SLOT EDR3M3SLOT EDR3M5SLOT LECODEDTX LECODEDRX

# hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 8C:88:2B:20:8B:42  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING
        RX bytes:2020 acl:0 sco:0 events:178 errors:0
        TX bytes:23965 acl:0 sco:0 commands:178 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'nanopineoplus2'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 5.1 (0xa)  Revision: 0x999
        LMP Version: 5.1 (0xa)  Subversion: 0x646b
        Manufacturer: Realtek Semiconductor Corporation (93)
```

vartom, 2023-02-24:
НА в докере, не получилось заставить работать с LR по предложенной инструкции. Может быть нужно ещё что то предпринять?

pvvx, 2023-02-24:
На ходу, когда уже загружен HA, переключение адаптера не работает.
Требуется остановить HA.
Интеграция "bluetooth" работает через прокладку "Bleak". "Bleak" в свою очередь работает через DBUS, Далее следующая "прокладка" - "Bluez", далее kernel. Это упрощенно, т.к. там десятки так называемых API прокладок.
Писатели "Bleak" вообще не в курсе что в 2016 году вышел стандарт BT5.0 и есть какие-то переключения PHY у адаптеров.

По идее, в докере, всё должно быть аналогично, если произвести переключение в той части, которая управляет адаптером.

pvvx, 2023-02-24:
Интеграция "[ble_monitor](https://github.com/custom-components/ble_monitor)" в HA включает сканирование только на 1M PHY. Автор ещё не умеет переключать адаптер на работу в BT5.0.

vartom, 2023-02-27:
Спасибо за описание. Удалось таки перевести датчики lywsd03mmc на использование BLE LR  в HA в докере. Пришлось перейти с ble_monitor на xiaomi_ble и заново их определить.
Через btmon проверил подключение действительно в LE Coded.
Посмотрю как это будет работать.

pvvx, 2023-02-27:
> Посмотрю как это будет работать.

Потребление от батарейки сильно возрастает в режиме LE Long Range. Более чем в 3 раза из-за увеличения времени передачи в 8 раз на пониженной модуляции.
Желательно изменить период трансляции рекламных сообщений на более чем 3.5 секунды.
Включение LE Long Range актуально если термометр расположен на удалении и адаптер его плохо видит.
Тестовые проверки при включении LE Long Range в Xiaomi LYWSD03MMC показывают, что связь с ним сохраняется на расстояниях более 500 метров и до 1 км по прямой видимости. В основном это требуется на отдельных частных участков с большими размерами. В городских "человейниках" эта опция бесполезна, т.к. 2.4ГГц плохо распространяется через препятствия в виде железо-бетонных и каменных стен.

pvvx, 2023-05-13:
[image]

[aliexpress.com](https://aliexpress.com/item/1005004929786108.html)

```
# lsusb
Bus 007 Device 002: ID 0a12:0001  USB2.0-BT
```

```
# hciconfig -a
hci0:   Type: Primary  Bus: USB
        BD Address: 04:7F:0E:4A:0C:C9  ACL MTU: 1021:9  SCO MTU: 255:4
        UP RUNNING INQUIRY
        RX bytes:6445 acl:0 sco:0 events:463 errors:0
        TX bytes:3579 acl:0 sco:0 commands:347 errors:0
        Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
        Name: 'BlueZ 5.64'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 5.1 (0xa)  Revision: 0x89
        LMP Version: 5.1 (0xa)  Subversion: 0x89
        Manufacturer: Barrot Technology Limited (2279)
```

`hcitool -i hci0 cmd 08 31 03 05 05`

Long Range - ok.

almirus, 2023-05-22:
@pvvx 
Добрый день! Xiaomi LYWSD03MMC прошит [LYWSD03MMC Custom Firmware Version 4.2](https://github.com/pvvx/ATC_MiThermometer/raw/master/ATC_v42.bin) на компе веб скрипты (сканер) видит устройство, дает прошить, иногда не может получить температуру влажность, иногда получает :)
`19:48:54: Watching advertisements from "ATC_0B27B9"...
19:48:54: Argh! TypeError: device.watchAdvertisements is not a function`
НО в HA со свистком [Bluetooth 5.0 Realtek](https://www.ozon.ru/product/adapter-bluetooth-5-0-realtek-dlya-pk-noutbuka-kompyutera-na-windows-linux-radius-do-100-585270678/) один раз добавился через Xiaomi BLE. Потом я недолго поигрался с параметрами через веб конфигуратор, и потом, чтобы я не делал, устройство более не добавляется (не видится)в HA (при этом CGPR1 продолжает нормально работать). Я и прошивал и на оригинальную и на модифицированную и батарею вынимал, и команду DD слал и 56, ничего не помогает.
Термометр в 3 метрах от Bluetooth свистка

pvvx, 2023-05-22:
> на компе веб скрипты (сканер) видит устройство, дает прошить, иногда не может получить температуру влажность, иногда получает :)
> 19:48:54: Watching advertisements from "ATC_0B27B9"... 19:48:54: Argh! TypeError: device.watchAdvertisements is not a function

На Windows "Advertising.html" практически не работает. Это проблема в Windows c `Web bluetooth API`.
Работает только на Android.

>НО в HA со свистком [Bluetooth 5.0 Realtek](https://www.ozon.ru/product/adapter-bluetooth-5-0-realtek-dlya-pk-noutbuka-kompyutera-na-windows-linux-radius-do-100-585270678/) один раз добавился через Xiaomi BLE. 

В HA две интеграции для работы с термометрами - "Bluetooth" и "Bluetooth Low Energy Monitor - Passive BLE monitor".  
"Bluetooth Low Energy Monitor" работает со всеми выбранными в [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) форматами.
"Bluetooth" имеет расширения "BTHome" и "Xiaomi BLE". Для них необходимо выбрать в [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) формат BTHome или Mijia. А так-же поставить галочку "AdFlag":
[image]

almirus, 2023-05-22:
@pvvx 
Может стоит в шапке веб конфигуратора указывать что нужно с мобильного заходить или я не нашел. с телефона конфигуратор заработал. Спасибо
Попробовал два формата BTHome, Mijia - не находит. Настройки же сразу применяются, батарею вытаскивать не нужно?

pvvx, 2023-05-22:
Настройки применяются сразу, но BLE реклама не работает во время соединения.
Подключаться для настроек в [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) можно в любой ОС.
А вот прием BLE рекламы в Windows зависит многих факторов. И пока в Chrome и других это опция - экспериментальная и предупреждение описано.

almirus, 2023-05-22:
> По USB VID/PID можно узнать какой чип.

[image]

pvvx, 2023-05-22:
Google: _USB\VID_0BDA&PID_8771_ а это и есть 8761B

almirus, 2023-05-22:
то есть просто ждать и выгадать верное время для соединения ?

pvvx, 2023-05-22:
Может отсоединяется по причине посаженной батарейки. Для соединения желательно чтобы было более 40% (+20С). 
Далее будет работать c HA и до 0%.

almirus, 2023-05-22:
99%

pvvx, 2023-05-22:
Во время подключения BLE реклама, ту которую принимает HA, не передается.
При подключении потребление больше...

almirus, 2023-05-22:
спасибо за уделенное время. пока безуспешно. на тлф в браузере все ок
на HA - тишина, хотя с таким же конфигом неделю назад термометр соединился, а теперь ни в какую

pvvx, 2023-05-22:
Отключите соединение с термометром в Android и в других. Закройте эксплорер.
В HA термометр появляется с задержкой... Лучше перезапустить всю систему с HA и подождать пару минут.

almirus, 2023-05-22:
@pvvx еще маленький вопрос, значок Bluetooth на своем устройстве что обозначает и должен ли он отображаться в момент рекламы (или соединения)?

pvvx, 2023-05-22:
Значок Bluetooth активен только во время соединения и мигает если нажата впаянная кнопка или геркон с выбором в меню функции геркона/кнопки "соединение" (обычно для Long Range).

almirus, 2023-05-22:
@pvvx вместо Xiaomi BLE, установил аддон [Passive BLE Monitor](https://github.com/custom-components/ble_monitor)
[image]
И он нашел оба устройства ¯\_(ツ)_/¯ (на том же железе)

fanoush, 2023-06-28:
> [image]
> 
> [aliexpress.com](https://aliexpress.com/item/1005004929786108.html)
> 
> 
> `hcitool -i hci0 cmd 08 31 03 05 05`
> 
> Long Range - ok.

Hello, I got this dongle and tried to enable long range on my thermometer. Since none of my phones can do 5.0/coded phy and nrfconnect don't see the device I tried this dongle in Ubuntu 22.04 and I have partial success :-)
hcitool switch works and then when using `bluetoothctl` for scanning like described here https://stackoverflow.com/questions/60318829/bluez-bluetoothctl-scan-vs-hcitool-scan I can see the thermometer - its name and mac address. Then I tried gattool to connect to the mac address but unfortunately it does not work when trying to enumerate services/characteristic and for everything it fails with
```
Command Failed: Disconnected
```
So most probably it does not even connect properly. Here is output from btmon
advertisement
```
> HCI Event: LE Meta Event (0x3e) plen 57                                                                   #247 [hci0] 1065.463599
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0001
            Props: 0x0001
              Connectable
            Data status: Complete
          Address type: Public (0x00)
          Address: A4:C1:38:C9:52:1E (Telink Semiconductor (Taipei) Co. Ltd.)
          Primary PHY: LE Coded
          Secondary PHY: LE Coded
          SID: 0x00
          TX power: 127 dBm
          RSSI: -96 dBm (0xa0)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x1f
        12 16 1a 18 1e 52 c9 38 c1 a4 db 09 14 12 c1 0a  .....R.8........
        4f 0e 04 0b 09 4c 59 57 53 44 30 33 2d 31 45     O....LYWSD03-1E 
        Service Data (UUID 0x181a): 1e52c938c1a4db091412c10a4f0e04
        Name (complete): LYWSD03-1E
```
and here is connection attempt
```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 42                                          #250 [hci0] 1065.465694
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: A4:C1:38:C9:52:1E (Telink Semiconductor (Taipei) Co. Ltd.)
        Initiating PHYs: 0x05
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE Coded
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                   #251 [hci0] 1065.466611
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0                                             #252 [hci0] 1069.513740
> HCI Event: Command Complete (0x0e) plen 4                                                                 #253 [hci0] 1069.515681
      LE Create Connection Cancel (0x08|0x000e) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                   #254 [hci0] 1069.521684
      LE Enhanced Connection Complete (0x0a)
        Status: Unknown Connection Identifier (0x02)
        Handle: 0
        Role: Central (0x00)
        Peer address type: Public (0x00)
        Peer address: 00:00:00:00:00:00 (OUI 00-00-00)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 0.00 msec (0x0000)
        Connection latency: 0 (0x0000)
        Supervision timeout: 0 msec (0x0000)
        Central clock accuracy: 0x00

```
somehow it does not handle the connection - gives error " Unknown Connection Identifier (0x02)" what is interesting is that there are those two connection entries
        Entry 0: LE 1M
        Entry 1: LE Coded

Is this something that can be changed in the firmware? Maybe it does not make sense to offer LE 1M connection if the advertisement is only over Coded PHY? Maybe with only one "LE Coded" entry gattool would not be confused and would possibly connect?

fanoush, 2023-06-28:
but when thinking about it maybe those two entries are on PC side(?) so not related to firmware - maybe it says which phy versions the PC supports/can try.

Just tried `hcitool -i hci0 cmd 08 31 03 04 04` and indeed, it enables only Coded PHY and there is only 1 entry when connecting and it does not work too, same error.
With this 04 setting bluetoothctl scan shows only thermometer and nothing else, which may be sometimes good.

fanoush, 2023-06-28:
Oh, is is a bug, and quite old one https://www.spinics.net/lists/linux-bluetooth/msg65444.html
> The unknown connection identifier is actually the success case for LE Create Connection Cancel command.

And the cancel is sent becasue of the 2s connection timeout.

pvvx, 2023-06-29:
Linux does not support BT5.0.
All utilities included in Bluez have long been declared obsolete and cannot be used to work with BLE devices.
There are only small pieces of Bluez and kernel v6.4+ code that work with some adapters.

fanoush, 2023-06-29:
> Now the solution is as follows: https://gist.github.com/mironovdm/cb7f47e8d898e9a3977fc888d990e8a9

Thanks! That worked and now it waits much longer however then after those 22 seconds  it is cancelled anyway (maybe it does not get any data from thermometer?). Can this be related to some connection parameters like the scan window or max connection interval so that the dongle is not really listening when the data comes?
```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                       #1336 [hci0] 78319.407277
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Public (0x00)
        Peer address: A4:C1:38:C9:52:1E (Telink Semiconductor (Taipei) Co. Ltd.)
        Initiating PHYs: 0x04
        Entry 0: LE Coded
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                #1337 [hci0] 78319.408178
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0                                          #1338 [hci0] 78341.829151
> HCI Event: Command Complete (0x0e) plen 4                                                              #1339 [hci0] 78341.830377
      LE Create Connection Cancel (0x08|0x000e) ncmd 1
        Status: Success (0x00)

```

pvvx, 2023-06-29:
> That worked and now it waits much longer however then after those 22 seconds it is cancelled anyway (maybe it does not get any data from thermometer?). 

Linux does not support the Bluetooth 5.0 specification released back in 2016.

pvvx, 2023-06-29:
> now it waits much longer however then after those 22 seconds 

This is the reason for the use of incorrect algorithms in BT Linux drivers.
Earlier it was announced that the implementation of scanning and connecting via the BLE protocol in Linux blocks the driver. This is unacceptable for implementation. It's as if the whole system or program is standing still until the driver figures it out. 
To connect, it is enough to get one advertising event. But not in Linux - it requires at least two. And if the second one is lost, then a timeout will come out, and 22 seconds will not be enough. Although for other operating systems, 10 seconds is enough, without any blocking.

fanoush, 2023-06-29:
Ok, then maybe it is worth  to try  some other bluetooth stack like https://github.com/bluekitchen/btstack and forget about bluez. btstack it is opensource and free for non-commercial usage and is supposed to work in linux directly with usb dongles over libusb. BTW, this is the stack which also used in Raspberry Pico W.

pvvx, 2023-06-29:
The scanning and connection functions available in the Linux API for BLE do not have a primary and secondary PHY installation. This is required by the BLE specification starting from version 5. 

The current Bluez has a separate function for installing primary PHY, but most programs don't know about it.

Ircama, 2023-06-30:
Perhaps, rather than relying on BLE USB sticks with dubious O.S. support of the BLE 5.x version, an alternative method could be exploiting an I2C-BLE 5 module, like TB-03F (or similar), to acquire Coded PHY advertisements sent by BLE thermometers in range, so that they are saved to RAM and cataloged by each MAC with related storage timestamp. Then, an I2C slave interface implemented in the device can read the last values saved in RAM each time an I2C master (host) polls them. But this requires developing this gateway from scratch.

pvvx, 2023-06-30:
@Ircama - The main purpose of the project is precisely to raise the level of external software related to BLE. Only in this way is it possible to ensure the availability of these technologies for the majority of users.
I am not a programmer, but a developer, and I can only provide such an option - the gradual inclusion of new standards in the device so that users increase the requirements for external software.

fanoush, 2023-08-17:
BTW I recently got Motorola Edge 30 phone and can confirm it finds my thermometer advertising over Coded PHY just fine in nRF Connect. I went a bit farther away so all other BLE devices from same location could not be found and I still saw my LYWSD03-1E in the scan and could connect to it reliably and repeatedly and read various characteristics when RSSI was reported as dBm -106.

So I can confirm it works just fine when BLuez stack is not involved :-)

splitice, 2023-11-15:
Has anyone ever found a single USB BT device that is capable of scanning both 1M and CODED at the same time?

I've got a `0bda:8771 Realtek Bluetooth Radio` and quite a few other chips. Including one USB module that I flashed Zephyr too (which unfortunately lacks LE Coded support in hardware)

Ignoring Linux support, I'm yet to find a single hardware product that can complete this element of the specification.

pvvx, 2023-11-15:
Realtek RTL8761 BT-USB scans all PHYs.

PHY 1M/2M/Coded:
`hcitool - i hci1 cmd 08 31 03 07 07`

splitice, 2023-11-15:
@pvvx When you have time I would love a btmon dump from start to finish for comparison.

I wonder if mine has an older firmware or similar. Or perhaps its the commands I'm sending.

Whenever I try and connect over either LE_1M or LE_CODED if CODED scanning is enabled the module refuses to connect. It's fine if scanning in LE_1M mode. I just assumed the dongle had incomplete CODED support.

I'm developing a custom bt stack (using raw HCI socket) currently. Not currently open source, but hoping the stakeholder will allow release to open source in the future.

Tend to agree re; linux. HCI support is incredibly buggy.

pvvx, 2023-11-15:
Connection to Coded PHY in Linux does not work! BLE scanning only.
(Also, SSD “fast parking” is not called up in Linux. If there is an unexpected power outage, the SSD may become damaged! Micron also failed with kernel patches...)

The firmware in the RTL chip in Linux is loaded at startup.

splitice, 2023-11-15:
> The firmware in the RTL chip in Linux is loaded at startup.

Its likely loaded by the Realtek drivers, no guaruntee we are on the same versions - or even that is the cause.

> Connection to Coded PHY in Linux does not work! BLE scanning only.

In raw (HCI_CHANNEL_USER) mode this might be bypassable if the cause is understood and the issue not the fault of firmware or hardware.

splitice, 2023-11-15:
Anyway we will never know if there could be support because I don't have my USB dongle even getting to the stage you have yours at. Nor any way to tell what the difference is.

pvvx, 2023-11-15:
Android connection with encoded PHY works well.  Even in Explorer.
However, the Web Blutooth API cannot scan the encoded PHY. But if the device (MAC) is stored in Explorer, then Explorer is connected to Coded PHY.

For Realtek RTL8761 BT-USB (0bda:8771) use these fw:
[bt_adpter.zip](https://github.com/pvvx/ATC_MiThermometer/files/13362442/bt_adpter.zip)

fanoush, 2023-11-15:
> However it appears the nRF52840 can only scan one PHY at a time.

what do you mean one at a time?
The S140 stack do accept multiple PHY in a bitfield for scanning https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v6.1.0/structble__gap__scan__params__t.html#a369859ca03e34c8220452ae95d1aa02f and it should work

As for hardware features I guess with single radio you can scan only one channel and one PHY at the same time so the solution is to alternate between them and that would be true for most/all of the BLE hardware? True that for nrf52 there is MODE register https://infocenter.nordicsemi.com/topic/ps_nrf52840/radio.html?cp=5_0_0_5_19_0#register.MODE where you select data rate  but I guess most/all other hardware would use same solution?

pvvx, 2023-11-15:
And who will upgrade Bluez and the Linux kernel to support HCI from Nordic?

pvvx, 2023-11-15:
> what do you mean one at a time? 

The **LE Coded PHY** uses **1Mbit PHY** but payload is coded at 125kbps or 500kbps. It also adds Forward Error Correction and Pattern Mapper. RF reception occurs in the same buffer and is then decoded. Purely software implementation.

splitice, 2023-11-15:
> And who will upgrade Bluez and the Linux kernel to support HCI from Nordic?

Actually very little is needed you get a mostly complete HCI stack from nordic, Zephyr wraps this up (and the HCI stack is "mostly to spec"). hciattach can handle the uart attachment already, no special support required.

We have our hardware wired up this way. Zephyr firmware with Nordic Radio exposing HCI over USB. udev rules to call hciattach, custom bt stack to speak hci (and gatt etc)

> And add support for Bluetooth 5.0?

Our custom stack has theoretical support. However we do not have any known working bluetooth 5.0 dongles, hence the request for a btmon. Its easy enough to replay those command parameters outside the linux stack from a custom stack and verify that the hardware does indeed support BT 5.0. We would love that to be the RTL dongle, we have that in hand and got pretty close with it in the past.

With that we can begin to investigate what issues are blocking and if they are software or hardware (and share those results back to the community).

Regardless of who builds support and which stack knowledge on working hardware is incredibly useful to any further development efforts (and this issue is easily findable on Google). If theres no hardware capable of BT Coded connection for example you should not expect BlueZ or Linux BT maintainers to prioritise support.

> what do you mean one at a time?

We have not (yet) been successful in getting it to receive coded announcements when both PHYs are active (only when CODED is the only one active). Or to begin join over coded (this may be our test device). This has been a side project however (which is getting consideration for some more engineer time) so it might be something simple.

Its easy for us to blame the hardware at this point, however I'm certainly willing to entertain that we might not be not exercising HCI correctly etc. Or a HCI command need hooking and handling differently within Zephyr (as we have done a couple times already).

At some point we will update our sniffer dongle to one that supports coded sniffing (any suggestions?) and it will become more obvious probably. Also planning on getting a Xiaomi Mijia or one of the other devices to act as a known working BT 5 coded device incase the issue is with our android phone we are using as the announcing & slave device.

We are using a special order nRF52840 devkit with a LNA&PA however any devkit should work. There exists plenty on marketplaces like aliexpress.

fanoush, 2023-11-15:
>  HCI from Nordic

AFAIK Nordic does not support  HCI over USB or serial (or SPI) with nrf5x chips at all, typically it is standalone solution where the application runs on same CPU as the BLE stack (or in case of nrf5340 on separate application core) . Also  it offers so called [serialization](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v15.0.0%2Flib_serialization.html) for multi MCU solution which is similar to HCI but non standard. It is also used by their own nrfConnect software for linux/windows via their own pc-ble driver with so called connectivity firmware https://github.com/NordicSemiconductor/pc-ble-driver/tree/v4.1.4/hex so it is completely separate from any OS BLE support (which may actually help with solving some use cases 
 where the OS support is poor?) I'll check whether it supports CODED PHY at all.

nrf52 chips  work with linux as HCI over usb or serial only in Zephyr with the hci sample applications https://docs.zephyrproject.org/latest/samples/bluetooth/hci_usb/README.html but that is opensource solution not supported by Nordic (?) Also it can only do BLE not classic bluetooth.

splitice, 2023-11-15:
> nrf52 chips work with linux as HCI over usb or serial only in Zephyr with the hci sample applications https://docs.zephyrproject.org/latest/samples/bluetooth/hci_usb/README.html but that is opensource solution not supported by Nordic (?) Also it can only do BLE not classic bluetooth.

Correct, BLE only. Mostly correct on the lack of direct Nordic support (not that you should ever rely on a hardware manufacturer for software support). The Zephyr HCI implementation is very small, its mostly just framing, flow control etc of the Nordic commands for the Radio. It still requires the Nordic Radio components which provide the stack (and therefore define many of the potential limitations).

You can get it to provide a decent HCI stack. We have done so. I do not know if BlueZ, noble or similar would find it compatible, its not something I've looked into.

fanoush, 2023-11-15:
> The **LE Coded PHY** uses **1Mbit PHY** but payload is coded at 125kbps or 500kbps.

Interesting, I am not that familiar with it. If it is only about payload then it should work indeed, however the Nordic docs says even the packet preamble is different https://infocenter.nordicsemi.com/topic/ps_nrf52840/radio.html?cp=5_0_0_5_19_0#concept_hz4_cjl_4r "For MODE = Ble_LR125Kbit and MODE = Ble_LR500Kbit the PREAMBLE is 10 repetitions of 0x3C." so I am not sure the hardware could match it together with Ble_1Mbit mode which uses different preamble. All  just my guesses, haven't tried it on such low level yet.

pvvx, 2023-11-15:
> You can get it to provide a decent HCI stack. We have done so. I do not know if BlueZ, noble or similar would find it compatible, its not something I've looked into.

But mostly people use Home Assistant, which is built on Bluez... 

> however the Nordic docs says even the packet preamble is different 

The hardware features on the chips may vary, including hardware CRC packet checking...

fanoush, 2023-11-15:
> The hardware features on the chips may vary, including hardware CRC packet checking...

yeah, but the preamble marks start of the packet, if it is not matched the packet is not received - the whole receiving flow does not start no matter how many other hardware features it has. If preamble is different for coded PHY than regular 1MBit you can't receive both types of packets at the same time. Well unless the hardware can match more of them at the same time to start receiving. Nordic hardware can match multiple addresses (which follows after preamble) but not multiple preamble patterns
https://infocenter.nordicsemi.com/topic/ps_nrf52840/radio.html?cp=5_0_0_5_19_6#concept_v4s_gkl_4r
BTW the preamble for coded phy is longer for redundancy
https://devzone.nordicsemi.com/f/nordic-q-a/70156/ble-coded-phy-preamble-and-range/287855
So do you know hardware that can match both preamble patterns (and then possibly non coded or coded addresses) at the same time? 

Was checking zephyr source and for nrf there is radio_phy_set
which sets mode according to phy
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio.c#L288
and also packet configure which sets the preamble in PCNF0
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/radio/radio.c#L435

Sadly I see  no other BLE radio hardware supported in zephyr source to match similar code there. However I guess even from software point of view this would be complicated to do, you first set PHY mode and then do something with the radio - send packet receive packet, setting multiple PHYs at once would complicate it even from software POV.

pvvx, 2023-11-15:
In principle, periodically switching on different PHYs is quite enough.
The main current problem is that in Linux there is no connection to a device running in Coded PHY. But LE advertising packages are accepted.

pvvx, 2023-11-16:
> Has anyone ever found a single USB BT device that is capable of scanning both 1M and CODED at the same time? And connecting at minimum over LE 1M.

Install several USB-BT adapters. Each to their own PHY.

fanoush, 2023-11-16:
> pc-ble driver with so called connectivity firmware https://github.com/NordicSemiconductor/pc-ble-driver/ so it is completely separate from any OS BLE support (which may actually help with solving some use cases
> where the OS support is poor?) I'll check whether it supports CODED PHY at all.

While the nrfconnect gui app for windows/linux does not support coded phy for scanning/advertising the pc-ble driver with connectivity firmware does
https://devzone.nordicsemi.com/f/nordic-q-a/55337/long-range-and-coded-phy-support-for-a-pc-application-on-nrf52840-dongle/224472

I downloaded latest release from https://github.com/NordicSemiconductor/pc-ble-driver/releases and there is prebuilt `connectivity_4.1.4_usb_with_s140_6.1.1.hex` inside that works with my E104-BT5040U dongle from aliexpress (was US$6 thing few years ago, now it is about $10, it is dongle compatible with Nordic PCA10059 - has basically same schematics just different form factor )
So with the dongle and this pc-ble-driver one can write C code for linux/windows that can use Nordic s140 6.1.1 softdevice api which should allow using coded PHY.  I'll try some basic example.

fanoush, 2023-11-17:
> What is the problem with using Telink TLSR825x chips? They work on all PHYs, there is an SDK with BT5.2 and HCI

How that will help with long range/coded phy working in code running in linux/windows?

HCI is too low level for using in your own code, this gives bluetooth API layer (which is even compatible with NRF SDK bluetooth APIs) on top of usb serial. 

Alternative can be noble on top of HCI but you'll probably still get linux kernel messing things for you if you run hciattach?

pvvx, 2023-11-17:
> How that will help with long range/coded phy working in code running in linux/windows?

Windows and Linux do not fully support Bluetooth version 4.2. 
And doing anything there is useless, because there is no API for Bluetooth v5+ and there never will be.

olivluca, 2024-01-01:
> Scans and connects to Coded PHY and other BLEs: https://apps.rustore.ru/app/ru.kompius.thermo_plung Supports thermometers with firmware from this repository. Works also for thermometers switched to LE Long Range (Coded PHY, BT5+).

I tried this app but I'm having a couple of problems:

1. I cannot find a way to publish to an mqtt broker.
2. if I enable long range I can no longer find the thermometer with miflasher (I have to use thermo_plung to switch long range off then I can connect again with miflasher).

pvvx, 2024-01-01:
> if I enable long range I can no longer find the thermometer with miflasher (I have to use thermo_plung to switch long range off then I can connect again with miflasher).

Web bluetooth API (Chrome) cannot scan BT5.0 devices. The reason is there is no support in Linux/Bluez. Say thank you to Linux, because... it hasn't given this to users since 2014.

olivluca, 2024-01-02:
I'm using chrome on the same android tablet where I use thermo_plung :disappointed:

Haldyz, 2024-01-03:
Hello,
I have bout BT dongle from [Aliexpress.com](https://aliexpress.com/item/1005004929786108.html).
I want to use it in BLE mode on my Raspberry Pi 4, where is running my HA in Supervisor mode.
Once i plugged in the dongle to USB 2.0 port on my RPi, I can see this:

```batchfile
#lsusb
Bus 001 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
``` 

```batchfile
# hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 04:7F:0E:5A:C6:9E  ACL MTU: 1021:9  SCO MTU: 255:4
	UP RUNNING 
	RX bytes:12589 acl:0 sco:0 events:656 errors:0
	TX bytes:7093 acl:0 sco:0 commands:462 errors:0
	Features: 0xbf 0xee 0xcd 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: SLAVE ACCEPT 
	Name: 'raspberrypi #1'
	Class: 0x4c0000
	Service Classes: Rendering, Capturing, Telephony
	Device Class: Miscellaneous, 
	HCI Version:  (0xa)  Revision: 0x89
	LMP Version:  (0xa)  Subversion: 0x89
	Manufacturer: not assigned (2279)
``` 

I have executed also following command:
`hcitool -i hci0 cmd 08 31 03 05 05
`

I can see the Adapter in HA:
[image]

How can I know that its working in BLE mode?

Thank you

pvvx, 2024-01-03:
This adapter (ID 0a12:0001) works very poorly.

lbarjak, 2024-01-04:
ASUS BT500

***@***.***:~$ bluetoothctl devices
Device F2:FE:77:F8:2B:5C F2-FE-77-F8-2B-5C
Device 54:EF:44:48:5D:E2 54-EF-44-48-5D-E2
Device A4:C1:38:94:08:E2 LYWSD03MMC
Device A4:C1:38:93:D5:64 LYWSD03MMC
Device A4:C1:38:D1:0C:8C LYWSD03MMC
Device C5:83:F5:63:9F:A6 SumUp 100
Device C4:30:18:EA:B9:C7 JBL PartyBox 100
Device A5:73:B3:0F:94:CE Wuzhi Audio

Haldyz ***@***.***> ezt írta (időpont: 2024. jan. 4., Cs,
9:56):

> I would buy a new one, but i would like to have some recommendation. This
> one i have bough this one from the same Ali link you have shared
> previously. It seems to be the same like on your picture marked as V5.1 (*Barrot
> BR865101*).
> image.png (view on web)
>
> I have also one more dongle, which is marked as V5.0, where is *Barrot
> 8041A02* chip:
> image.png (view on web)
>
> This one is recognized on my RPi like bellow, but its not connecting to Mi
> Thermometers at all:
>
> #lsusb
> Bus 001 Device 003: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
>
> #hciconfig -a
> hci2:	Type: Primary  Bus: USB
> 	BD Address: 00:1A:7D:DA:71:11  ACL MTU: 679:9  SCO MTU: 48:16
> 	UP RUNNING
> 	RX bytes:684 acl:0 sco:0 events:48 errors:0
> 	TX bytes:3362 acl:0 sco:0 commands:48 errors:0
> 	Features: 0xff 0xff 0x87 0xfa 0xdb 0xbf 0x7b 0x83
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
> 	Link policy: RSWITCH HOLD SNIFF PARK
> 	Link mode: SLAVE ACCEPT
> 	Name: 'raspberrypi #3'
> 	Class: 0x4c0000
> 	Service Classes: Rendering, Capturing, Telephony
> 	Device Class: Miscellaneous,
> 	HCI Version: 5.0 (0x9)  Revision: 0x810
> 	LMP Version: 5.0 (0x9)  Subversion: 0x2512
> 	Manufacturer: Cambridge Silicon Radio (10)
>
> You have previously recommended to buy a Realtek RTL8761 BT adapter, but
> its not easy to find it on Ali according to the chipset name. Can you
> recommend some dongle which is tested and working well please?

[75 more replies]