pvvx/ATC_MiThermometer #298: Имеющиеся форматы BLE рекламы.

pvvx, 2023-02-18:
Наименьшее потребление имеет формат рекламы ‘ATC1441’ и ‘PVVX’, включая шифрованные варианты.
Но данные форматы не содержат возможности передачи событий. К примеру, нажатия кнопки или открытия-закрытия если установлен геркон.

‘Mijia (MiHome)’  использует большую длину рекламного сообщения и за один раз не может передавать разные данные. В итоге данные передаются по очереди, используя несколько рекламных передач. В шифрованном варианте это требует ещё дополнительного потребления для перепаковки и шифрации данных такой BLE рекламы.

Формат ‘BTHome версии 1’ не сильно оптимизирован по размеру пакета передачи, но позволяет разбирать данные в пакете рекламы любому простому чипу с BLE. ‘BTHome версии 2’ не позволяет произвести такие действия, т.к. устройству требуются постоянные обновления таблиц описателей дескрипторов данных, которые часто добавляются в интеграции ‘BTHome’.

Дальнейшее развитие BLE в IoT завязано на поддержке со стороны Linux. На сегодня новые ядра и драйверы, включая ‘Bluez’, частично исправлены и позволяют работать с форматами Bluetooth версии 4.0. 
Однако полной поддержки Bluetooth 5.0, выпущенной ещё в 2016 году и реализованной во многих BT адаптерах в Linux не наблюдается. В основном, развитие BLE тормозят разнообразные утилиты и зоопарк нагромождений API. При этом имеющиеся в продаже адаптеры BT давно позволяют полноценно работать с Bluetooth 5.0+.

Для уменьшения потребления и дальности связи в датчиках BLE желательно использовать введенные после выхода стандарта Bluetooth 5.0 форматы. Но они не поддерживаются в текущих реализациях ПО на Linux. К примеру, использование Coded PHY (Long Range) и периодической рекламы позволяет увеличить дальность связи в 4 раза и значительно уменьшить потребление от батареи...

В итоге, в последующих версиях, использование вариантов рекламных форматов будет сведено к   ‘BTHome версии 1’ и, или, возможно к ‘BTHome версии 3’, если она будет выпущена и не иметь структурных сложностей для обработки в сторонних приложениях.

tido-, 2023-03-26:
**Bluetooth Advertising Formats** (https://github.com/pvvx/ATC_MiThermometer#bluetooth-advertising-formats)

You mention one of four can be configured: Xiaomi, ATC, Custom, **BTHome** and encrypted options (5).

What I am not sure and I don't want to brick the device - which code do I have to put into: **Send Config** box, to get the bare minimum over BThome (Battery Charge, Temperature & Humidity)?

Thank you.

pvvx, 2023-03-27:
The greatest dependence is associated with the scanning mode of the external adapter. If the active scanning mode is set, the consumption is maximum.

tido-, 2023-03-27:
Thank you for your prompt answer.

My bad, I used Vivaldi Webbrowser which is also based on Chrome (flashing was fine), but all the Options you are mention were not visible, on re-connect.
I tried MS Edge (on another PC) and now I understand what you meant by point 3: Several configuration options appear  :)