pvvx/ATC_MiThermometer #299: Activating 'long range' is dangerous?!?!

buserror, 2023-02-20:
In installed 4.2, and clicked the 'long range' thinking it was just a change in power transmission. I was wrong, it seems it goes into another mode that makes it unusable. The ESP32 no longer receive beacons, and even my phone fails to recognize it when scanning, so I can't revert the setting!

I know, it's not a 'stable' firmware, but a warning on that checkbox might be useful, like "if your phone/receiver can't do that particular mode as well, you're screwed" :-)

pvvx, 2023-02-20:
Remove and insert the battery.

yinzhefuyun, 2023-02-22:
> > I know, it's not a 'stable' firmware, but a warning on that checkbox might be useful, like "if your phone/receiver can't do that particular mode as well, you're screwed" :-)
> 
> [image]
> 
> Remove and insert the battery - option will be disabled.
> 
> This is the inclusion of the Bluetooth LE standard option from **2016**. 90% of smartphones have an adapter with Bluetooth 5.0 or higher. ESP32 not supported.
> 
> BLUETOOTH SPECIFICATION Version 5.0 | Vol 1, Part C page 291 _Core Specification Change History_
> 
> _9 CHANGES FROM v4.2 TO 5.0 9.1 NEW FEATURES Several new features are introduced in the Bluetooth Core Specification 5.0 Release. The major areas of improvement are: • Slot Availability Mask (SAM) • 2 Msym/s PHY for LE • **LE Long Range** • High Duty Cycle Non-Connectable Advertising • LE Advertising Extensions • LE Channel Selection Algorithm 2_

Xiaomi 12Tpro Bluetooth 5.3 also can not discover the LYWSD03MMC thermometer with 4.2 firmware

pvvx, 2023-02-22:
Для проверки адаптера смартфона используйте nRFConnect.

Web Bluetooth API пока не работает с BT5.0 адаптерами.

Home Assistant работает с LE Long Range. 
[Работа термометров в Home Assistant c Bluetooth LE Long Range.](https://github.com/pvvx/ATC_MiThermometer/issues/297)

Сброс опции возможен и в nRFConnect, как показано в [README.md](https://github.com/pvvx/ATC_MiThermometer)

yinzhefuyun, 2023-02-22:
> LYWSD03MMC

Sending 0x56 via nRFConnect, Chrome can now connect to the LYWSD03MMC thermometer

pvvx, 2023-02-22:
> Sending 0x56 via nRFConnect, Chrome can now connect to the LYWSD03MMC thermometer

This action sets the default configuration. And in the default configuration, "LE Long Range" is disabled (BLE advertising thermometer works in BT4.2 mode).

closed 2023-02-22 by pvvx