pvvx/ATC_MiThermometer #300: Flasher web page doesn't detect any of my newly bought LYWSD03MMC devices.

AdrianGarside, 2023-02-21:
I'm trying to set this up with some devices I just bought off ali express. However they don't show up in the flasher page. I can see them on my phone via a bluetooth le scanning app, and I can connect them to the Mi Home app where they appear pretty much instantly when I go to add. But I've tried to find them via flasher (both the original and the new v5.2 here) in multiple browsers on a couple of PCs w/ bluetooth, my iPhone and my rasperry Pi and they simply never show up. Did the latest versions block the flashing tool detecting them somehow?

AdrianGarside, 2023-02-21:
token_extractor got these from xiaomi cloud in case that helps?
  ---------
   NAME:     Temp1
   ID:       blt.2.1d5ps....
   BLE KEY:  58b8c6edf...
   MAC:      A4:C1:38:...
   TOKEN:    637550...
   MODEL:    miaomiaoce.sensor_ht.t2
   ---------

AdrianGarside, 2023-02-21:
I went away, got takeout, came back and tried again. This time I tried a different laptop. And that one just worked straight off the bat.

pvvx, 2023-02-21:
Problems with Bluetooth adapters and installed versions of operating systems?

closed 2023-02-22 by pvvx