pvvx/ATC_MiThermometer #302: Bluetooth PHY switching using nRF Connect. After disconnecting there is a problem with finding the thermometer. Bug?

DienoX, 2023-02-28:
Hi,
I don't know if this is a bug or it was intended for this thread.
0. Set in the LE CODED thermometer settings.
1. Connect via nRF Connect to the thermometer with your firmware.
2. In the nRF Connect app, switch the PHY mode to 1M.
3. Disconnect the connection
4. The nRF Connect application and Google Chrome will not search for your ATC thermometer

The way to fix it is to remove the battery from the thermometer.
I wonder if a malicious neighbor can spoil thermometers for less advanced users in this way.

pvvx, 2023-03-03:
1. Set in the LE CODED thermometer settings.
2. Connect via nRF Connect to the thermometer with your firmware.
3. In the nRF Connect app, switch the PHY mode to 1M.
4. Disconnect the connection
5. Connect via nRF Connect to the thermometer with your firmware - ok
6. Disconnect the connection

> I wonder if a malicious neighbor can spoil thermometers for less advanced users in this way. ?

You are not describing something.
Everything works in nRFConnect. Any switching to any PHY:

[image]

```
nRF Connect, 2023-03-04
ATC_0B5EED (A4:C1:38:0B:5E:ED)
V	01:57:30.042	Connecting to A4:C1:38:0B:5E:ED...
D	01:57:30.042	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	01:57:31.045	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	01:57:31.045	Connected to A4:C1:38:0B:5E:ED
V	01:57:31.057	Discovering services...
D	01:57:31.057	gatt.discoverServices()
D	01:57:31.096	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	01:57:31.322	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	01:57:32.427	[Callback] Services discovered with status: 0
I	01:57:32.427	Services discovered
V	01:57:32.444	Generic Access (0x1800)
- Device Name [N R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R W] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Model Number String [R] (0x2A24)
- Serial Number String [R] (0x2A25)
- Firmware Revision String [R] (0x2A26)
- Hardware Revision String [R] (0x2A27)
- Software Revision String [R] (0x2A28)
- Manufacturer Name String [R] (0x2A29)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Environmental Sensing (0x181A)
- Temperature Celsius [N R] (0x2A1F)
   Client Characteristic Configuration (0x2902)
- Temperature [N R] (0x2A6E)
   Client Characteristic Configuration (0x2902)
- Humidity [N R] (0x2A6F)
   Client Characteristic Configuration (0x2902)
Unknown Service (00010203-0405-0607-0809-0a0b0c0d1912)
- Unknown Characteristic [R WNR] (00010203-0405-0607-0809-0a0b0c0d2b12)
   Characteristic User Description (0x2901)
Unknown Service (00001f10-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R WNR] (00001f1f-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000fe95-0000-1000-8000-00805f9b34fb)
D	01:57:32.445	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	01:57:32.451	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
D	01:57:32.454	gatt.setCharacteristicNotification(00002a1f-0000-1000-8000-00805f9b34fb, true)
D	01:57:32.459	gatt.setCharacteristicNotification(00002a6e-0000-1000-8000-00805f9b34fb, true)
D	01:57:32.462	gatt.setCharacteristicNotification(00002a6f-0000-1000-8000-00805f9b34fb, true)
D	01:57:32.465	gatt.setCharacteristicNotification(00001f1f-0000-1000-8000-00805f9b34fb, true)
I	01:57:32.499	Connection parameters updated (interval: 20.0ms, latency: 49, timeout: 4000ms)
V	01:57:38.912	Reading PHY...
D	01:57:38.912	gatt.readPhy()
I	01:57:38.932	PHY read (TX: LE Coded, RX: LE Coded)
V	01:57:45.327	Requesting preferred PHYs...
D	01:57:45.327	gatt.setPreferredPhy(LE 1M, LE 1M, coding option = No preferred)
[... 89 lines cut ...]
Environmental Sensing (0x181A)
- Temperature Celsius [N R] (0x2A1F)
   Client Characteristic Configuration (0x2902)
- Temperature [N R] (0x2A6E)
   Client Characteristic Configuration (0x2902)
- Humidity [N R] (0x2A6F)
   Client Characteristic Configuration (0x2902)
Unknown Service (00010203-0405-0607-0809-0a0b0c0d1912)
- Unknown Characteristic [R WNR] (00010203-0405-0607-0809-0a0b0c0d2b12)
   Characteristic User Description (0x2901)
Unknown Service (00001f10-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N R WNR] (00001f1f-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000fe95-0000-1000-8000-00805f9b34fb)
D	01:58:16.964	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	01:58:16.968	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
D	01:58:16.971	gatt.setCharacteristicNotification(00002a1f-0000-1000-8000-00805f9b34fb, true)
D	01:58:16.975	gatt.setCharacteristicNotification(00002a6e-0000-1000-8000-00805f9b34fb, true)
D	01:58:16.978	gatt.setCharacteristicNotification(00002a6f-0000-1000-8000-00805f9b34fb, true)
D	01:58:16.981	gatt.setCharacteristicNotification(00001f1f-0000-1000-8000-00805f9b34fb, true)
I	01:58:17.041	Connection parameters updated (interval: 20.0ms, latency: 49, timeout: 4000ms)
V	01:58:22.187	Reading PHY...
D	01:58:22.188	gatt.readPhy()
I	01:58:22.201	PHY read (TX: LE Coded, RX: LE Coded)
```

DienoX, 2023-03-04:
Hi,
I will try to repeat the process again. I've managed to do it twice already, but I think the problem may be due to my damaged hardware.

When I test it, I'll let you know what my results were.

pvvx, 2023-03-04:
Specify the sequence of actions.

closed 2023-03-14 by pvvx