pvvx/ATC_MiThermometer #304: Major battery drain (faulty unit?)

mlichvar, 2023-03-05:
I have four LYWSD03MMC B2.0, bought at the same time from the same source couple months ago. I flashed the pvvx firmware v3.9. Three are still running fine with the original battery and report about 90% battery left. The fourth one just died with second battery. I thought the original battery was bad, but now I suspect something is wrong with the HW. I measured the consumption and it seems to be around 0.2 mA.

I tried upgrading the firmware to 4.1, but it's still draining. IIRC the only changes I originally made in the configuration was set a PIN, disabled the "comfort" reporting, and set the measurement recording to 0 (disabled). Now I tried "set default" and also switch "lowpower mode", still no luck.

Is there a factory reset or anything else I could try to fix it? Has anyone else seen this issue?

erapade, 2023-03-15:
I have one out of 10 (with the same configuration and from the same batch) that drains battery.

erapade, 2023-03-16:
Re-flashing to 4.1 seams to partly have solved my problems. And it might be that I was running on 4.2 beta before, but when checked it was advertising itself as 4.1 but I re-flashed (around 7 PM) anyway and that had a big change

[image]
[image]

pvvx, 2023-03-28:
Check what is displayed in the [Management GPIO_RS (Reed Switch)](https://github.com/pvvx/ATC_MiThermometer#reed-switch-or-button-functions) for the "Switch" and "Counter" options:

[image]

And what is the specified interval:

[image]

ens4dz, 2023-04-11:
I have a question. What is the name of the device used to measure uA in a chart in this great repository? I searched for power, uA, and measure but didn't find any clues.

pvvx, 2023-04-11:
From ready-made that it is possible to buy - [Power Profiler Kit II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2)
If DIY assemble and calibrate yourself  - [PowerProfiler](https://github.com/pvvx/UBIA/tree/master/PowerProfiler)
Or https://github.com/pvvx/PowerProfiler_pcm1802 + INA199An type chips, or or other operational amplifier chips...
And options using an oscilloscope with INA199An type chips..

closed 2023-10-30 by pvvx