pvvx/ATC_MiThermometer #305: Can't activate "AdFlags"

marph91, 2023-03-05:
I would like to activate "AdFlags" in order to add my devices to Home Assistant (as mentioned [here](https://github.com/Bluetooth-Devices/bthome-ble/issues/26#issuecomment-1332196521)). However, always when I check the checkbox, it gets unchecked immediately after sending the config. Any idea why it isn't working?

My config:

_Status: Hardware Version: LYWSD03MMC B1.4, Software Version: 3.8_

[image]

pvvx, 2023-03-06:
There was an insert of new models of thermometers, and errors arose ...
"AdFlags" fixed.

marph91, 2023-03-06:
Works now. Thanks for the fast fix!

closed 2023-03-06 by marph91