pvvx/ATC_MiThermometer #307: MJWSD05MMC

pvvx, 2023-03-06:
_Сброс привязки к Mi-Home для новой активации._
 
1.	Удерживать две кнопки пока дисплей не моргнет (10 cек)
2.	Коротко нажать верхнюю кнопку (0.5 сек)
3.	Коротко нажать нижнюю кнопку (0.5 сек)
4.	Подключиться и Активировать MJWSD05MMC в [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) 

Далее возможно записать альтернативную прошивку с именем файла типа ‘BTH_vNN.bin’.
После прошивки устройство изменит имя bluetooth на что-то вроде "BTH_XXXX".

_В альтернативной прошивке MJWSD05MMC_

Верхняя боковая кнопка:
1.	Нажатие переключает отображение на главной строке: время, температура, влажность, заряд батареи в %, напряжение батареи, внешнее число с атрибутами. 
2.	Нажатие на эту кнопку от 0.2 сек включает быстрый “рекламный интервал” на 180 сек для улучшения соединения, если установлены длинные интервалы. 
3.	Нажатие и удержание более 15 секунд перезагружает устройство с параметрами по умолчанию.

Нижняя боковая кнопка по умолчанию работает как ‘Switch’ в Home Assistant. К примеру для включения/выключения света и т.д..

Возврат к оригинальной версии сопровождается стиранием всей памяти замеров и установок. Оригинальная версия имеет большой размер и использует другую структуру разметки OTA в Flash. По этим причинам требуется очистить и переразметить всю память Flash для обратного перехода к оригинальной версии. Аналогично и при переходе на альтернативную прошивку, но это происходит без уведомлений. 
Если большая часть Flash уже заполнена или размечена, тогда при перовой попытке выполнить Big.OTA может возникнуть тайм-аут соединения и потеря связи, или рассинхронизация связи с внешним BT адаптером. 
Если соединение в первого разорвется или заливка будет происходить очень медленно - тогда требуется повторный перезапуск  Big.OTA.  Не бойтесь разорвать соединение при OTA, если оно идет медленно. Повторное соединение может обеспечить нормальную скорость - от 40 до 60 сек на всю OTA.  
Часто помогает сброс адаптера USB-BT путем переустановки в USB разъеме или в системных меню. Это баг у Low-cost адаптеров типа Realtek RTL8761 и исправление невозможно - перед Активацией и заливкой альтернативной прошивки  желательно такой адаптер пересбросить. Иначе получение кодов и OTA на оригинальной прошивке будет происходить минутами и посадит батарею в термометре...

weichenw, 2023-11-02:
Hi,

Are there any instructions on how to flash/unbrick https://gthub.com/pvvx/ATC_MiThermometer/issues/316 using USB-TTY?

pvvx, 2023-11-02:
[After the Zigbee firmware](https://github.com/devbis/z03mmc/issues/41), it is required to use the RESET output when starting the "activation" in any programmer.

weichenw, 2023-11-03:
Thanks, I will give that ago.