pvvx/ATC_MiThermometer #309: Can't connect to LYWSD03MMC after fkashing

Thomas1457, 2023-03-07:
Hi, 
I flashed my LYWSD03MMC nd since i can't connect it first to acess settings. 
here is the log of my connection : 
`19:22:09: Reconnect 5 from 5
19:22:10: Hardware Revision String: B1.4
19:22:10: Software Revision String: V4.1
19:22:10: Detected Mi device
19:22:10: NotSupportedError: GATT operation failed for unknown reason.
19:22:10: Something went wrong, too many reconnects`

pvvx, 2023-03-07:
Read [Read.me](https://github.com/pvvx/ATC_MiThermometer#readme): 

_Warnings_

_For reliable connection and refirmware on LYWSD03MMC, the battery level must be more than 40%. This is a design feature - a [cheaper version of the hardware from Xiaomi](https://user-images.githubusercontent.com/12629515/105704705-423fc600-5f20-11eb-88ca-a2db712d0222.png)._

After connecting, set "Connect latency" to less than or equal to 1000 ms:
[image]
Or download ver4.2 -> automatically set to 1000 ms.

Thomas1457, 2023-03-08:
Thanks for your answer. I changed battery and was able to connect once after flash, but since i got the same issue. Not a big issue but still wondering what happened.

pvvx, 2023-03-08:
On some BT adapters, after flashing to an alternative firmware, they should be rebooted. The MAC does not change and the adapter can remember the old attributes from the last connection and does not perceive that they have been changed.

closed 2023-03-14 by pvvx