pvvx/ATC_MiThermometer #315: MJWSD05MMC time sync and send data from esphome

WildeRNS, 2023-03-14:
hello. as i understant, for now, isn't aviliable to sync time with MJWSD05MMC and send data to display from esphome?

WildeRNS, 2023-03-14:
the problem is, that i have not bluetooth on server - only by esphome (esphome perfectly works with my few LYWSD03MMC sensors: send temperature from other sensor by platform: pvvx_mithermometer).

i alredy on stable 4.2 firmaware.

pvvx, 2023-03-14:
ESPHome with BLE has many problems.
I do not support him, because. this is the most undesirable of all options.
ESPhome with ESP32 does not work with BT5.0, resets constantly, has a lot of receive loss in BLE, and drains the batteries of all BLE devices very badly when connected.
ESPhome with ESP32-C3 is even more unstable and doesn't support BT5.0.
Also, ESPHome does not support the official BTHome format for Home Assistant.

closed 2023-03-14 by pvvx