pvvx/ATC_MiThermometer #316: MJWSD05MMC never activate

tremebundo, 2023-03-15:
I am able to connect to it with Telink flasher but always activating  takes for ever. Never gets activated.
I have tested with a PC with windows,a MAC, a laptop and an Adroid tablet leaving in the "Activating now, please wait..." state for hours. And the activation never ends. I have also reset the device several times with no luck.
When connected, it says that it has Hardware Revision V2.3 and Software version 0026.

tremebundo, 2023-03-16:
If this helps, this is what I could see at the console of the DevTools when trying to activate:

Searching for devices
commons.js:2 Channel: Error in handleResponse UNK/SW_UNREACHABLE contextMenu createContextMenu
_handleResponsePromise @ commons.js:2
(anonymous) @ commons.js:2
Promise.catch (async)
_sendMessageToExtension @ commons.js:2
sendToExtension @ commons.js:2
send @ commons.js:2
listener @ commons.js:2
commons.js:2 Uncaught (in promise) Error: Could not establish connection. Receiving end does not exist.
fromGeneric @ commons.js:2
fromAny @ commons.js:2
_handleResponsePromise @ commons.js:2
(anonymous) @ commons.js:2
Promise.catch (async)
_sendMessageToExtension @ commons.js:2
sendToExtension @ commons.js:2
send @ commons.js:2
listener @ commons.js:2
TelinkMiFlasher.html:1634 Get Advertising MAC (Web Experimental Features!)
TelinkMiFlasher.html:1630 Catching advertisement from "MJWSD05MMC"...
TelinkMiFlasher.html:1630 Status: Connecting to: MJWSD05MMC ...
TelinkMiFlasher.html:1630 Received advertisement from "MJWSD05MMC"...
TelinkMiFlasher.html:1630 Uint8Array(12) [48, 88, 50, 40, 1, 110, 116, 251, 56, 193, 164, 8, buffer: ArrayBuffer(12), byteLength: 12, byteOffset: 0, length: 12, Symbol(Symbol.toStringTag): 'Uint8Array']
TelinkMiFlasher.html:1634 MiVer5, DevID: 0x2832-MJWSD05MMC, FnCnt: 1, CtrID: 0x5830 Not bound, Standard certification, MAC: A4C138FB746E, Capability: 0x08 - front binding
TelinkMiFlasher.html:1630 Found GATT server
TelinkMiFlasher.html:1630 Services: 0000fe95-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:1630 Services: 00010203-0405-0607-0809-0a0b0c0d1912
TelinkMiFlasher.html:1630 Services: ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 Services: 0000180a-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:1630 Found Telink OTA service
TelinkMiFlasher.html:1630 Found Telink OTA write characteristic
TelinkMiFlasher.html:1630 Found Device Information Service
TelinkMiFlasher.html:1630 Get SoftWare Revision String
TelinkMiFlasher.html:1630 Status: Get SoftWare Revision String...
TelinkMiFlasher.html:1630 SoftWare Revision String: 0026
TelinkMiFlasher.html:1630 Get HardWare Revision String
TelinkMiFlasher.html:1630 Status: Get Hardware Revision String...
TelinkMiFlasher.html:1630 Hardware Revision String: V2.3
TelinkMiFlasher.html:1630 Status: Detected Mi device
TelinkMiFlasher.html:1630 Found MiMain service
TelinkMiFlasher.html:1630 Check micharacteristics:
TelinkMiFlasher.html:1630 ebe0ccb7-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccb9-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccba-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccbb-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccbc-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccbe-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccc2-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccc4-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccc8-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0ccd4-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 ebe0cce4-7a0a-4b0c-8a1a-6ff2997da3a6
TelinkMiFlasher.html:1630 Found MiTemp characteristic
TelinkMiFlasher.html:1630 Found Main service
TelinkMiFlasher.html:1630 Found enc_10 char
TelinkMiFlasher.html:1630 Found enc_19 char
TelinkMiFlasher.html:1630 Status: MJWSD05MMC HW:V2.3 Connected, you can now Do the Activation to either get the Token or flash a new Firmware
TelinkMiFlasher.html:1630 Send: a2000000
TelinkMiFlasher.html:1630 Enc_19: 000000000100
TelinkMiFlasher.html:1630 Send: 00000101
TelinkMiFlasher.html:1630 Enc_19: 010001000000
TelinkMiFlasher.html:1630 Send: 00000100
TelinkMiFlasher.html:1630 Send: 15000000
TelinkMiFlasher.html:1630 Send: 000000030400
commons.js:2 Channel: Error in handleResponse UNK/SW_UNREACHABLE contextMenu createContextMenu
_handleResponsePromise @ commons.js:2
(anonymous) @ commons.js:2
Promise.catch (async)
_sendMessageToExtension @ commons.js:2
sendToExtension @ commons.js:2
send @ commons.js:2
listener @ commons.js:2
commons.js:2 Uncaught (in promise) Error: Could not establish connection. Receiving end does not exist.
fromGeneric @ commons.js:2
fromAny @ commons.js:2
_handleResponsePromise @ commons.js:2
(anonymous) @ commons.js:2
Promise.catch (async)
_sendMessageToExtension @ commons.js:2
sendToExtension @ commons.js:2
send @ commons.js:2
listener @ commons.js:2
TelinkMiFlasher.html:1630 Temp: 21.8°C, Humi: 38%, 3099 mV
TelinkMiFlasher.html:1630 Temp: 21.8°C, Humi: 37%, 3099 mV
TelinkMiFlasher.html:1630 Temp: 21.7°C, Humi: 37%, 3099 mV
TelinkMiFlasher.html:1630 Temp: 21.8°C, Humi: 37%, 3099 mV

pvvx, 2023-03-16:
[README.md](https://github.com/pvvx/ATC_MiThermometer#flashing-or-updating-the-firmware-ota) _Addition for MJWSD05MMC: ..._

tremebundo, 2023-03-16:
I have done that many  times.
As yu can see in the pictue, Telink connects to the thermometer and I can the "Do activation" button appears, but it will not never finish.
[image]

tremebundo, 2023-03-16:
May be this device has a new sw version that behaves differently?
I have just bought two and they are pretty new

pvvx, 2023-03-16:
I have HW: V2.3, SW: 0026, [Production date 2022.12](https://pvvx.github.io/MJWSD05MMC/) A few pieces are fine.
In tests, this was done more than 20 times and with alternating several registrations in MiHome.

pvvx, 2023-03-16:
Especially for you, Activation and OTA on a new LUM out of the box:
```
02:01:19: Searching for devices
02:01:21: Connecting to: MJWSD05MMC
02:01:30: Hardware Revision String: V2.3
02:01:30: Software Revision String: 0026
02:01:30: Detected Mi device
02:01:30: Connected
02:01:31: Activating now, please wait...
02:01:34: Registration successfull
02:01:34: Send Login, please wait...
02:01:35: Received device infos are correct
02:01:35: Login successfull
02:01:38: Load firmware file 'BTH_v42.bin'...
02:01:38: File: BTH_v42.bin
02:01:38: File size: 85332 bytes
02:01:38: Count: 5334
02:01:39: Start DFU
02:02:23: Update done after 43.634 seconds
02:02:24: Disconnected.
```

tremebundo, 2023-03-17:
I finally got it.
What I did, was perfoming the reset just AFTER hitting the connect button and rapidly clicking the activate once connected. Otherwise, I cannot make it work. I understood it with the Mi Home app, which did not connect too.
I have now my two units flashed.
Sorry for my misunderstandig and thanks a lot for your patience.

closed 2023-03-17 by pvvx