pvvx/ATC_MiThermometer #317: Spikes in temperature reporting

erapade, 2023-03-15:
I can see some spikes in the temperature measure. They are not frequent and I have no issues with them, I just want to report it in case it's a FW issue. I don't see the same issue in the other metrics and so far this is just on two out or 10 devices:

[image]

pvvx, 2023-03-15:
https://github.com/pvvx/ATC_MiThermometer/issues/271  ?

erapade, 2023-03-16:
"Specify platform used: , integration installed, BT adapter used, thermometer type, and BLE advertising format type."
Are there anything missing below

Hardware Version: LYWSD03MMC B1.6, Software Version: 4.1, Sensor: SHTC3 (SHTV3)
Custom config HEX string: 55032000002804a9043184b400
Advertising type: BTHome v1
I'm using a Raspberry PI 3B with Home Assistant, no extra BT adapter (to my knowledge)

One thing though. If Beta FW 4.2 is advertised as 4.1, I might have been running on that (at least on one of the sensors. The reason me thinking so is that one of the units also drained a lot of battery and when re-flashed (about 19:00 yesterday) with 4.1 the battery was not that heavy drained anymore, see pictures below:
[image]
[image]

Why the cyan one started to drop in voltage in the middle of the night I have no clue, that one is definitely not running on Beta

erapade, 2023-03-16:
As I'm using the RPI 3B internal BT connection I guess that is the root cause for the spikes and in that case I guess this can be closed. 
Question though. 

1. Is this something that can be fixed on the HA side?
2. Will encrypting fix the issue

pvvx, 2023-03-16:
In the release of version 4.2, the calculation of battery per transmission is now averaged over 16 measurements. Otherwise, there was a lot of noise in the charts.

> 1. Is this something that can be fixed on the HA side?

I do not think that's possible. I don't have RPi3/4 to find out possible errors.

> 2. Will encrypting fix the issue

Perhaps, but there are nuances. If the error does not create an analogue of a close format for another type of device.

erapade, 2023-03-16:
Thanks for answering so quick. I will monitor it for some days, maybe switch to beta for half of the fleet

Concerning the voltage measure I switched batteries between the two devices with the lowest voltage and this was the result

[image]

Surprisingly (at least for me) both got a higher (and normal) measured voltage. Maybe as you said it could have been a bad connection to the batteries. I guess the alternative would be if it was the power-on reboot that fixed the voltage issue