pvvx/ATC_MiThermometer #318: Question: config Sensor in "LowPower mode"

glsf91, 2023-03-18:
I see now an option "Sensor in "LowPower mode" in the config. I have a LYWSD03MMC.
What is this doing when on or off?

pvvx, 2023-03-19:
For the SHTV3 (SHTC3) sensor, switches the measurement time to a shorter and less accurate one. But it allows you to slightly reduce consumption.

For the SHT4x sensor, it switches the measurement diagram. If this option is enabled, the transmission of measurement will be delayed by the period of transmission of BLE advertising. But this greatly reduces the consumption of the system. If the option is disabled, the transmitted data will be updated immediately, and transmitted on the next advertising transmission. But this requires more power - the CPU will wait while the probe is measuring.

glsf91, 2023-03-19:
Thanks.
Maybe an idea to put this information in de readme.cmd or other file.

pvvx, 2023-03-19:
I don't speak English and don't want to learn it. The automatic translator makes many mistakes and is difficult to work with.

closed 2023-03-19 by glsf91

vanyasem, 2023-12-21:
@pvvx можно например сделать README на русском, а коммьюнити будет потихоньку переводить его на английский - как вариант

pvvx, 2023-12-21:
Создание технической документации требует значительно больше времени, чем создать новую программу.
Посмотрите нормы для работы квалифицированного инженера - 1 печатный лист в рабочий день.