pvvx/ATC_MiThermometer #319: Feature: No Weekdays on MJWSD05MMC

benep, 2023-03-23:
I still find it very practical that your software allows displaying the time on the small LYWSD03MMC. Thank you very much for that. Unfortunately, even with the correction factor, it's not perfect. (Btw. INT is not big enough for the correction factor with one of my LYWSD03MMCs).
That's why I couldn't resist buying the new MJWSD05MMC with a clock when I saw that you also support it.
And I'm thrilled with how well it works. However, one small thing bothers me: the Chinese weekdays.
Would it be possible to turn them off via the flasher like the smiley? That would be a dream. Thank you for your effort.

pvvx, 2023-03-24:
Smiley off:
[image]

benep, 2023-03-25:
I could imagine it like this.
[image]

pvvx, 2023-03-25:
And what are the problems with showing the days of the week?

ariekraakjr, 2023-04-18:
It is in chinese. A lot of us can't read chinese.

max-koptev, 2023-11-03:
Полностью поддерживаю! Если скрыть китайские дни недели не сложно, я был бы очень Вам благодарен за такую функцию!

martini1992, 2023-12-19:
Also being able to swap the date from MM/DD to DD/MM would be excellent too.

pvvx, 2023-12-19:
> Also being able to swap the date from MM/DD to DD/MM would be excellent too.

https://github.com/pvvx/ATC_MiThermometer/issues/401

martini1992, 2023-12-19:
Ah ok, thanks :)

closed 2025-04-23 by pvvx