pvvx/ATC_MiThermometer #321: Updated from 2.4. to 4.2 and now Home Assistant no longer receiving updates

dalanmiller, 2023-03-26:
I have upgraded my `LYWSD03MMC` to the latest version of `pvvx` firmware and have stopped seeing temperature and humidity updates. 

Looking at a few other issues, I've checked on `AdFlags` and left the advertising type to the new `BTHome v1`. This differs from my older setup which was using the `All` advertising type. But with these settings, I still am not receiving any updates on my Home Assistant instance. 

The BLE portion of my `configuration.yaml` looks like this:

```
ble_monitor:
```

I've just left everything default and it was working to pick up the updates before. 

Is there anything else I should be doing? I wasn't able to find anything in the docs but I may have missed a setup guide somewhere.

pvvx, 2023-03-27:
`AdFlags` are required for `Bluetooth` integration to work in `passive scanning` mode.

[ble_monitor](https://github.com/custom-components/ble_monitor) integration works with any settings in the thermometer.

To work at long distances in the `LE long range` mode, an adapter with support for `Bluetooth 5.0` and higher is required. And `Bluetooth` integration.

Guirigall, 2023-03-29:
Is there any way to get an older firmware? The last one has broken all the BLE communication and as the OP, I get no data from the ATC. I have other ATCs at 3.7 firmware that are working flawlessly.

TIA

pvvx, 2023-03-30:
https://github.com/pvvx/ATC_MiThermometer/releases

pvvx, 2023-03-30:
Support for older versions in TelinkMiFlasher is supported as far as possible...

Zebble, 2023-04-10:
4.2/4.3 broke HA/ESPHome integration for me as well.  I tried enabling Adflags, but that doesn't seem to make a difference.  I'm using BTHome v1 packet types.

Is there a solution/setting I'm missing?

pvvx, 2023-04-12:
I haven't seen BTHome support in ESPHome.
And in ESPHome on ESP32 only BT4.0.
ESP32-C3 works extremely unstable in BTHome.

ESP Home previously supported the ATC1441 format.
I don't use ESPHome and I don't recommend it.

Zebble, 2023-04-12:
Thanks @pvvx,

You are corrent.  There isn't direct BTHome support in ESPHome.  It acts as a bluetooth proxy for HomeAssistant.

3.9 has been working perfectly in this configuration.  Only 4.x seems to break things...

pvvx, 2023-04-12:
Then why does everything else work, but ESP Home doesn't? :)

pvvx, 2023-04-12:
Switching formats:
[image]

Home Assistant:

ATC1441
[image]

PVVX(custom)
[image]

MIJIA(MiHome)
[image]

BTHome:
[image]

pvvx, 2023-04-12:
Windows, [python-interface](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface)

[image]

pvvx, 2023-04-12:
>3.9 has been working perfectly in this configuration. Only 4.x seems to break things...

>[ESPHome Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy.html)

This means that ESP Home does not work with BLE. The BT Home format has not changed from fw version 3.7 to 4.3.

closed 2023-04-12 by pvvx

50494554524F, 2023-04-14:
> 4.2/4.3 broke HA/ESPHome integration for me as well. I tried enabling Adflags, but that doesn't seem to make a difference. I'm using BTHome v1 packet types.
> 
> Is there a solution/setting I'm missing?
> 
> FYI, it worked with 3.9 and have gone back to 3.9.

i tested v4.2 + bthome + esp32 bluetooth proxy and it works...

pvvx, 2023-04-14:
> i tested v4.2 + bthome + esp32 bluetooth proxy and it works...

ESPHome does not work with Bluetooth 5.0 from 2016.