pvvx/ATC_MiThermometer #322: MJWSD05MMC features questions

erich74, 2023-03-26:
First off I would really like to thank you for your work.
I have flashed my MJWSD05MMC with your latest 4.3 beta fw

I have few questions though:

1. In you readme page it is stated that it is possible to cycle the screen with different values like display time, then temperature etc... I do not see this option for the MJWSD05MMC , is it possible, or not (yet) implemented?
2. Is there a way to permanently display the battery icon (and/or percent) at the top of the display)?  I do not see any such option, looks like the icon is displayed only when the main screen is displaying battery (% or v)
3. after flashing the top button is disabled, do you think it could be possible to re enable it in order for instance to scroll between the various screens (main display being time, temp, hum...)

Thanks a lot!
Eric

pvvx, 2023-03-26:
1. Auto-switching of screens is not provided for MJWSD05MMC.
2. The battery icon does not contain gradations of the charge level.
3. The upper button enables connection by short pressing, and switching screen types by long pressing.

erich74, 2023-03-27:
Thank you very much for your answer:

1. Ok no problem
2. Ok I did not notice that, then I agree there's no point displaying it
3. Thanks! I completely missed the long press and sorry I meant to say main value displayed, not auto-switching  (well I meant to say auto switching the main value displayed but now I understand I can do it by long pressing the top button ;) )

Would you please mind elaborating a little bit how the button button can be used as a switch in HA? FYI I am integrating the MJWSD05MMC with OpenMQTTGateway BLE gateway based on ESP32. I am retrieving the data from MQTT to HA.
Here is a typical advertisement I am receiving, how would the buttom button should be advertised?:

{"id":"A4:C1:38:DF:DE:2F","mac_type":0,"adv_type":0,"name":"BTH_DFDE2F","rssi":-46,"servicedata":"2fdedf38c1a4f4088711410c64bc05","servicedatauuid":"0x181a","brand":"Xiaomi","model":"LYWSD03MMC","model_id":"LYWSD03MMC_PVVX","type":"THB","tempc":22.92,"tempf":73.256,"hum":44.87,"batt":100,"volt":3.137}

erich74, 2023-03-27:
Thanks for the detailed explanation.
Please note that I am not using ESP Home but the openmqttgateway project (https://docs.openmqttgateway.com/).

pvvx, 2023-03-27:
> Thanks for the detailed explanation. Please note that I am not using ESP Home but the openmqttgateway project (https://docs.openmqttgateway.com/).

I didn't find BLE version 5.0+ there. 
For ESP 32 chip - Bluetooth 4.2 max
ECP32-C3 - Bluetooth 5.0 is specified, but there are no implementations.

erich74, 2023-03-27:
Yes this is BLE 4.2 (I am using a regular ESP32 chip). This is not an issue for reading temp & hum as I don't need to have a high refresh rate.

For the switch this might be a different story but in any case, I changed RS Mode to "switch" but it is not advertised for what I see in my MQTT client:

{"id":"A4:C1:38:DF:DE:2F","mac_type":0,"adv_type":0,"rssi":-45,"servicedata":"2fdedf38c1a4d009b510430c648805","servicedatauuid":"0x181a","brand":"Xiaomi","model":"LYWSD03MMC","model_id":"LYWSD03MMC_PVVX","type":"THB","tempc":25.12,"tempf":77.216,"hum":42.77,"batt":100,"volt":3.139}

pvvx, 2023-03-27:
The "[BTHome](https://bthome.io/v1/) v1" protocol is used to transmit events
0x181C - 16-bit UUID Service 0x181C BTHome v1, no security
0x181E - 16-bit UUID Service 0x181E BTHome v1, security enable
Opening/Closed, Opening counter

erich74, 2023-03-27:
Ok I am using your custom advertisement method and not BTHome so this should be the reason?

pvvx, 2023-03-27:
Pressing and holding two buttons for a long time (more than 20 seconds) on the MJWSD05MMC resets to the default settings.

erich74, 2023-03-27:
ok good to know, thanks a lot for your great work and support!

pvvx, 2023-03-27:
On all thermometers with a button, you can assign different events. On LYWSD03MMC, a revision is required - the installation of a reed switch. Then it turns into a door or window sensor. As well as the counter.
There is no "Connect" assignment option for MJWSD05MMC, since it was originally assigned to another button.