pvvx/ATC_MiThermometer #324: LYWSD03MMC и python

JurySpb, 2023-03-28:
Пытаюсь подключить с помощью Python термо-гигрометр (прошит v42). Использую библиотеку bluepy (bluepy3) под Linux.
В режиме сканирования читаю строку данных, как пример "1a185fed3138c1a4f807640d010c64bd0f".
Как понял :
первых два байта - номер сервиса (байты переставлены) "181a"
Потом мак адрес - тоже преставлены байты.
Потом два байта - температура, еще два - влажность, далее два байта - батарея в мВ, байт батарея в %.
Остали непонятны еще три байта.
Кто может помочь?

pvvx, 2023-03-29:
Термометры используют 4 формата и каждый ещё в вариантах с шифрацией. 

У вас включен вариант формата - "[custom](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/custom_beacon.h#L20)"
```
// GATT Service 0x181A Environmental Sensing
// All data little-endian
typedef struct __attribute__((packed)) _adv_custom_t {
	uint8_t		size;	// = 18
	uint8_t		uid;	// = 0x16, 16-bit UUID
	uint16_t	UUID;	// = 0x181A, GATT Service 0x181A Environmental Sensing
	uint8_t		MAC[6]; // [0] - lo, .. [6] - hi digits
	int16_t		temperature; // x 0.01 degree
	uint16_t	humidity; // x 0.01 %
	uint16_t	battery_mv; // mV
	uint8_t		battery_level; // 0..100 %
	uint8_t		counter; // measurement count
	uint8_t		flags;
} adv_custom_t, * padv_custom_t;
```
Примеры для Python:

https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface

https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/ble_parser/atc.py

JurySpb, 2023-03-29:
Да, спасибо это я первым делом посмотрел. Меня смутило то, что есть отличия структуры "adv_custom" и то что я получаю.  Приходит 17 байт, т.е. какие то поля у меня отсутствуют. Это или первые или последние?

pvvx, 2023-03-29:
У вас нет заголовка:
```
1216+
"1a185fed3138c1a4f807640d010c64bd0f"
"0102030405060708091011121314151617"
```

JurySpb, 2023-03-29:
Понятно. Спасибо.
Вообще я получаю список, например такой:
[(1, 'Flags', '06'), (22, '16b Service Data', '1a185fed3138c1a4f807640d010c64bd0f'), (9, 'Complete Local Name', 'ATC_3AE39B')]
Может тут есть недостающая информация, хотя это уже и не принципиально.
Интересный факт, у меня есть usb модуль wifi/bluetooth, версия bluetooth 4.0. (Realtek RTL8723BU). Так он принимает список без последнего элемента (9, 'Complete Local Name', 'ATC_3AE39B')

JurySpb, 2023-03-30:
Может кому-то пригодится, простая программа python. Сканирует BLE устройства и читанет данные с устройства по MAC адресу. Результат выводит в строку.
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

# bluepy3 from https://github.com/Mausy5043/bluepy3
from bluepy3.btle import Scanner, DefaultDelegate

import time
import struct

# data parsing
def pars_data(d):
    p = d[8:]
    temperature = round(int(struct.unpack("<H",p[0:2])[0])*0.01,1)
    humidity = round(int(struct.unpack("<H",p[2:4])[0])*0.01,1)
    battery_v = int(struct.unpack("<H",p[4:6])[0])
    battery_l = int(p[6])
    count = int(p[7])
    flag = int(p[8])    
    print(f"\t{temperature=}\u00b0C \t{humidity=}% \t{battery_v=}mV \t{battery_l=}%  \t{count=} \t{flag=}")

    # for scan
class ScanDelegate(DefaultDelegate):
    def __init__(self):
        DefaultDelegate.__init__(self)

    def handleDiscovery(self, dev, isNewDev, isNewData):
        global bo
        if isNewDev:
            print (f"Discovered device adr:{dev.addr} RSSI:{dev.rssi}dB time:{round(time.monotonic()-t_start,2)}s")
        elif isNewData:
            # if adress
            if dev.addr == adr:
                print (f"\nReceived data from {dev.addr} Packets received={dev.updateCount}")
                pars_data(bytearray.fromhex(dev.getScanData()[1][2]))
                bo = True
            
# mac address of device
adr = "xx:xx:xx:xx:xx:xx"

t_start = time.monotonic()

bo = False
print("Scan...")
scanner = Scanner().withDelegate(ScanDelegate())
# start the scanner and keep the process running
scanner.start()
while True:    
    scanner.process()
    if bo:
        scanner.stop()
        break
    print ("\tScan running...")

print("\nFull time s, ",time.monotonic() - t_start)    

```

pvvx, 2023-03-30:
Это не работает с "LE Long Range" и прочими фишками Bluetooth 5.0.

pvvx, 2023-03-30:
Дык Linux умирает, особенно с BLE :)
В API Bluez ныне есть затычки в виде типа "новых" функций для поддержки BT5.0. Но все остальные функции не имеют опций для поддержки 5.0+ и требуют полного пересмотра. В них есть только установка secondary PHY, а primary PHY нет, хотя это описано в стандартах Bluetooth 5.0 от 2016 года. Так же и в ядре нет этой поддержки. Приняты только частичные исправления старых вредоносных ведений нацеленных против развития BLE в виде включения патчей от кентов из Intel в период с 2016 года, убивающих BLE на корню. BLE - это Low Energy, а из него Linux делает Big Energy :)
Так что Linix мертв для BLE. Поддержка есть только в Andoid, а это Google.

pvvx, 2023-03-30:
> Может и не надо дальних расcтояний? Мой адаптер v4.0 видит просто толпу всяких устройств от соседей вокруг.

Это только малая часть. Где в Linux поддержка 2M PHY в приложениях? 
Где в Linux поддержка интервалов рекламы в 10 сек, описанных в стандарте ещё BT4.0? 
Где в Linux поддержка соединения с устройствами имеющими "расширенную рекламу"? 
Где в Linux поддержка "периодической рекламы"? 
И т.д. 
Linux просто мертв из-за некомпетентности принимающих решения по ядру и развитию...

JurySpb, 2023-03-30:
> Linux делает Big Energy :)

Я думаю, что если это так важно, то народ продавит...

Из любопытства поставил птичку на [BT5+ PHY:] Настольный комп. с v5.0 видит, а 4.0 нет.

JurySpb, 2023-03-30:
> Где в Linux поддержка интервалов рекламы в 10 сек, описанных в стандарте ещё BT4.0?

Често говоря, для меня это пока темный лес :) Рекламу я только в одном смысле понимаю.

JurySpb, 2023-03-30:
> Каким образом, если в Linux нет требуемого API?

Я по своей наивности полагаю, что желающих до всех этих нововведений не много. Где там этот bluetooth, да в основнов в наушниках. И зачем там расстояние если все в пределах 1м. Вот поддержка какого нибудь AptX - это да, так ее сделали, насколько я знаю.

pvvx, 2023-03-30:
> Я по своей наивности полагаю, что желающих до всех этих нововведений не много. 

У LE Long Range расстояния связи больше чем у WiFi. А потребление в сотни раз меньше, как и излучение.
Для большинства IoT пропускной способности WiFi не требуется. И тем более у WiFi есть "замирания" - выпадения на секунды, что требует буферизации. Плюс поддержка TCP/IP и прочее в IP требует больших ресурсов у чипов, которые в свою очередь требуют ещё и энергии. Для типичного IP устройства необходимо поддерживать множество портов с разными форматами и функциями. Только реализация TCP стека с соблюдением стандартов требует от контроллера наличия от 200 килобайт памяти. И это минимум для пару сокетов.

Зачем датчику или выключателю это всё?
 
---

scanning-api

https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#scanning-api

Feature/Platform | Chrome OS | Android | Mac | Linux | Windows
-- | -- | -- | -- | -- | --
Advertisements Scanning |   | 🚩 | 🚩 |   |  

Никаких Linux нет :)

JurySpb, 2023-03-30:
> Зачем датчику или выключателю это всё?

Тогда вопрос, зачем им bluetooth с гигагерцами и затуханием? Есть, например, 433Мгц и дальность до километра.

> scanning-api

Это же для WEB? Там и Win нет.  Но я не программист, это у меня хобби (электроникой с детства увлекаюсь).

pvvx, 2023-03-30:
> Тогда вопрос, зачем им bluetooth с гигагерцами и затуханием? Есть, например, 433Мгц и дальность до километра.

А для LE Long Range это одинаково - при +0дБм TX и приеме на -108..112 дБ (типичное потребление при передаче 5..7 мА для чипов "бытового" техпроцесса) - 1 км. При малом увеличении отдачи и приемной антенны уже к 1.6 км.
И чипы с полной поддержкой уже есть у всех в смартфонах.

pvvx, 2023-03-30:
> Это же для WEB? Там и Win нет. Но я не программист, это у меня хобби (электроникой с детства увлекаюсь).

Bleak претендует на кроссплатформенность. Но они так-же кивают в сторону отсутствия API у Linux.

pvvx, 2023-03-30:
> Никакой bluetooth не сможет обмануть закон сохранения энергии. :) Смотрел тут недавно... платка 433Мгц с контроллером на борту в зависимости от режима, потребление (из описания) -16мА, 3,6мА, 80мкА.

"Обман" у вас в руках. И имеет множество поддерживаемых стандартов и проверенную систему приоритетов, каналов и прочего необходимого для функционирования сотен устройств в доме.

JurySpb, 2023-03-30:
> "Обман" у вас в руках. И имеет множество поддерживаемых стандартов и проверенную систему приоритетов, каналов и прочего необходимого для функционирования сотен устройств в доме.

Ну и "кушает" соответственно :), вы правы не так как wifi, Но все же не святым духом питается :)

> А описанное вами приложение пожирает батарейки у всех окружающих устройств своими запросами в режиме "активного сканирования".

Само собой, без батарейкм никак, но не надо сканировать непрерывно. У меня для двух устройств это занимает 20-30 сек. Тем более это простой пример. Можно непосредственно соединятся с нужным MAC адресом, но энергия все равно нужна.

> При 433МГц с возможной модуляцией и требуется во много раз больше энергии.
> LYWSD03MMC при 0дБм и передачи по 3-м каналам с общим временем передачи в пару мс потребляет в пике 7мА.

на 433 и полоса 400кГц, а для датчика или выключателя время, что 2мс, что 200 без разницы.

pvvx, 2023-03-30:
> на 433 и полоса 400кГц, а для датчика или выключателя время, что 2мс, что 200 без разницы.

При одинаковом периоде между передачей информации на 433МГц вы говорите надо 16 мА на 200 мс, а для LE Long Range это не более 5 мс на 7 мс.  Т.е. разница в 91.5 раза.
CR2032 * 91 = АКБ на 18200 мА.ч

JurySpb, 2023-03-30:
> на 433МГц вы говорите надо 16 мА на 200 мс

Если верить описанию, то 16мА - (4-80)мс, при 80мкА - 500мс
Чтоб сравнивать нужны реальные измерения, у меня их нет. А так это просто обмен мнением...

pvvx, 2023-03-30:
Вы приводите усредненные значения для бесконечных периодов. Типа потребление в режиме сна микроконтроллера.

У SoC в LYWSD03MMC с текущей прошивкой  это 1.8 мкА. Плюс чип дисплея и датчика. Итого к 6 мкА при опции включенного LCD. Если дисплей отключен и LYWSD03MMC B1.9 - это 2 с копейкой мкА.

> Можно непосредственно соединятся с нужным MAC адресом, но энергия все равно нужна.

А зачем вообще "соединяться"? Датчик передает данные и исполнительное устройство их принимает. За счет дублирования обходятся неизбежные выпадения при приеме от помех и коллизий в радиоэфире.
В диапазоне 433МГц значительно больше и требуется повышение мощности передатчика и ограничения чувствительности приемника...

pvvx, 2023-03-30:
> Какие там периоды, я не разбирался. Если дойдут руки может проведу измерения.

С какой целью? В документации уже всё описано и обычно приводят 3 параметра: ток при TX,  при RX и когда чип "спит".

> Спрос - рождает предложение.

Спрос есть давно и он реализуется в Android и подобных. Но не в Linux.
Разрабатываются стандарты и ПО. BT5.0 - это 2016 год, а вы предлагаете что-то из 1800-х.

JurySpb, 2023-03-30:
> В документации уже всё описано и обычно приводят 3 параметра: ток при TX, при RX и когда чип "спит".

Обычно приводят для идеальных условий.

> Спрос есть давно и он реализуется в Android и подобных. Но не в Linux.
> Разрабатываются стандарты и ПО. BT5.0 - это 2016 год, а вы предлагаете что-то из 1800-х.

Реализован там где есть спрос. В Linux выходит не нужен.
Вот често, я не знаю чего меня с Linux лишили? Если датчик будет, так как есть работать на 20м, то и хватит.

pvvx, 2023-03-30:
В использованном API [в примере](https://github.com/pvvx/ATC_MiThermometer/issues/324#issuecomment-1489827690) есть функции и опции чтобы уменьшить потребление и работать с BT5.0 и выше.
Проблемы возникают в других случаях, когда требуется кроссплатформенность.

JurySpb, 2023-03-30:
> Возможности обычных пользователей использовать современное оборудование, экономить на батарейках, развивать "Умный дом".

Самостоятельно, как я, этим занимаются единицы. Кому мы интересны? 

> Вы сами указали что надо 433МГц - т.е. некие специфические чипы, которых нет у обычных пользователей.
> Так наверно и считают в Linux. Есть ещё ZigBee - это альтернатива, но не полная - его нет у пользователей в типовых устройствах и ПО на него частично закрыто. Т.е. не для всех.

Зачем все это обычным пользователям? Проще пойти и купить готовое решение. Андроид - это деньги, там и делают для обычных пользователей.

pvvx, 2023-03-30:
> Я не понял, это ж мой пример.

А я про то и пишу. 
Текущая прошивка термометра и так уже содержит более 80% хлама, чтобы обойти ограничения созданные в Linux.
А пример показывает как не надо делать.

pvvx, 2023-03-30:
Во первых, это устаревший формат, принятый для возможности передать данные на имеющемся тогда оборудовании и ПО. Он устарел. Далее был создан специальный формат, ныне названный “BTHome” и под него выделили специальный официальный UUID. Чтобы всё было в соответствии со стандартами Bluetooth ассоциации. Но версию 2 разработали программисты, не знакомые и не учитывающие аппаратные возможности малых  чипов...
Во вторых в Linux ранее было принято, что по умолчанию всякие опции отключены. И в представленном коде не наблюдается включения доп. опций. Работает типа наобум, как получится. А если до этого адаптер переключен на другой формат PHY и прочего?
Активное сканирование не выбирает кого теребить дополнительными запросами, сажая батарею. Тут даже писатели Bluez  ещё не определились как и что там происходит,  стандарте от Bluetooth SIG. А каждое устройство BLE передает специальные флаги…

JurySpb, 2023-03-30:
> В итоге этот код где-то когда-то может работать, а где-то нет. Но все претензии придется разгребать другим :)

У вас это наверно любимая мозоль... Я же рад тому, что что то работает. Сделают что то лучше - хорошо, а нет ... Что других проблем нет?

pvvx, 2023-03-30:
Этот и формат ‘ATC’ пока оставлен ради “совместимости” с приложениями ESPHome. Там всё очень новое очень долго развивается…
Когда появится возможность перейти на новые форматы в Linux, этот формат будет удален из поддержки. Спрос пользователей и уменьшение потребления требует перехода на “периодическую BLE рекламу” или другие фичи от BT5.+. Оборудование уже давно доступно всем.

JurySpb, 2023-03-30:
> или другие фичи от BT5.+. Оборудование уже давно доступно всем.

Я например, не готов менять компьютер ради BT5.+.

> Это не мозоль - кто-то должен пинать лежачих для развития :) У них пока стадия “круговая порука” и “моя хата с краю”.

Меня пинать бесполезно...

pvvx, 2023-03-30:
А я меня нет никаких личных претензий. Есть цели и они достигаются...

JurySpb, 2023-03-30:
> А у меня нет никаких личных претензий. Есть цели и они достигаются...

А у меня тем более :)

pvvx, 2023-03-30:
Год назад на малых системах с Linux было невозможно принять данные в другом доме на моем участке от термометра в бане. А ныне работает, путем пары доп. команд.  И так далее.

JurySpb, 2023-03-30:
> А ныне работает,

Это ж хорошо, попробую и я на даче использовать.

pvvx, 2023-03-30:
> Спрос - рождает предложение.

Курица или яйцо? Сложный вопрос... :)

pvvx, 2023-03-30:
> Картошку что то не рекламируют, а спрос есть 

Говорят, что картошку кто-то завез и навязал... А в техническом развитии и других темах почему-то стартует “прецедент”, а не спрос. Странно. По вашему должен быть "спрос"?

---

Я там случайно не ту кнопку нажал... и попортил ваше сообщение. Посмотрите, правильно ли восстановил?

JurySpb, 2023-03-30:
> Говорят, что картошку кто-то завез и навязал...

Грешат на Петра, но распространение получила только лет через сто.

> Я там случайно не ту кнопку нажал...

Ничего не заметил.

> Если бы не навязали и не привезли, иначе одну свеклу так и ели, как и дублировали бы устаревшие примеры на Python :)

Свекла - хороший продукт. Устаревшее не значит плохое. Примеры если работают чем плохи? Альтернативы же нет.

> И “прецеденту” не нужно качество или что-то ещё, достаточно просто информации и наличия.

Народ разный кому то и этого достаточно. Вот кефир - известный, хороший продукт, а мне не надо - не люблю с детства. Или вот вышел супер-пупер новый процессор всем хорош, но я не готов покупать. Цена тоже конечно имеет значение, но не только.

pvvx, 2023-03-30:
> Или вот вышел супер-пупер новый процессор всем хорош, но я не готов покупать. Цена тоже конечно имеет значение, но не только.

Тут вроде прослеживается это – не бери больше чем можешь унести. Т.е. готовность и возможности. А если переложить на технические вещи, то это: Зачем нужно то, что не можешь освоить. (Как и в случае со скриптом на Питоне :) )

JurySpb, 2023-03-30:
> Тут вроде прослеживается это – не бери больше чем можешь унести.

Есть еще такое умное выражение - Разумная достаточность.

> Как и в случае со скриптом на Питоне :)

Там ничего сложного, даже для меня - дилетанта.

pvvx, 2023-03-30:
> Там ничего сложного, даже для меня - дилетанта.

Как раз куча сложностей и требует инструкции для пользователей на десятки страниц, да ещё с областью применения и конфигурации системы.
Запустите перед ним `hcitool -i hci1 cmd 08 31 03 04 04`
И кто дал root доступ вашему приложению к адаптеру?
Несколько других вопросов уже указано ранее.

JurySpb, 2023-03-30:
> Как раз куча сложностей и требует инструкции для пользователей на десятки страниц, да ещё с областью применения и конфигурации системы.

Не надо преувеличивать. Все это для так называемых DIY проектов, т.е. Какой то минимум у человека должен быть.

> Запустите перед ним `hcitool -i hci1 cmd 08 31 03 04 04`
> Переключите на формат 'ATC1441' и наслаждайтесь :)

Это сложно... ;) и вроде не hci1, а hci0 у Linux.

> И кто дал root доступ вашему приложению к адаптеру?

Я и дал, для части bluepy3 - это все есть на сайте библиотеки. Это если не возится с рут правами и дальше.

> Тем более это какой-то огрызок-обрывок, а не законченная вещь/программа и использовать такое простому пользователю невозможно :)

 Тут есть все, что нужно в начале. Вы не поверите, но на самом деле, вот таких простых вещей и не хватает.  То что предлагается как пример на python - это какой то монстр на все случаи жизни. По своему опыту скажу, что по быстрому там разобраться невозможно.

> И не забудьте организовать поддержку - ответы на вопросы пользователей этого куска скрипта :)
Нет проблем. Как говорится - тот кто ничего не умеет - тот учит. :)

JurySpb, 2023-03-30:
Еще вопрос, что считается максимальным и минимальным для RSSI?

pvvx, 2023-03-30:
> Лучше бы, чтоб не возникли. :)

Тогда такое безобразие случится в самый неожиданный момент.

`p = d[8:]` и далее... А если там нету этих 8 ми и далее байт? :)
Термометр  кроме этой структуры передает ещё и другие. Но ваш код ничего не разбирает.
Это такая специально введенная уязвимость?

> Еще вопрос, что считается максимальным и минимальным для RSSI?

Это смотря для какой версии Bluetooth. Что по уровню максимума в TX, что по минимуму для RX.
В этом одно из серьезных отличий для увеличения дальности связи от привычного вам Bluetooth, ранее навязанного как средство для ближней связи, почти дословно - личного пространства. А теперь его раздвинули до 1.6 км :)

JurySpb, 2023-03-30:
> `p = d[8:]` и далее... А если там нету этих 8 ми и далее байт? :)
> Термометр кроме этой структуры передает ещё и другие. Но ваш код ничего не разбирает.
> Это такая специально введенная уязвимость?

Когда он передает другие? Все что я видел - только это и другого не было.

> Это смотря для какой версии Bluetooth.

В тех рамках, что сейчас v4.x

pvvx, 2023-03-30:
Сейчас Bloetooth 5.3 как минимум.
И по ссылке описаны некоторые разновидности, т.к. есть отличия при разных PHY.

JurySpb, 2023-03-30:
> Когда включены опции для геркона или на термометрах с кнопкой.

Стандартно, ничего этого  в датчике нет. Здесь только один вариант датчика. Желающим копать глубже придется делать за свой счет.

> Вы хотите чтобы я написал инструкцию к вашему куску кода?
> Типа так не включай, так не делай, найди старый адаптер, поставь древнее ПО, ... :)

Любой путь начинается с первого шага и он обычно самый трудный. Я его сделал и делюсь своим опытом.
На многое не претендую от объяснений не отказываюсь. Вобщем  AS IS ...

pvvx, 2023-03-30:
Какие RSSI интересуют?
Для дешевого BT адаптера?

К примеру такого:
[image]

Когда у него прием нормален и пропусков мало в LE Long Range:
[image]

pvvx, 2023-03-30:
> Стандартно, ничего этого в датчике нет. Здесь только один вариант датчика. Желающим копать глубже придется делать за свой счет.

Всё есть. в некоторых аж две кнопки. А если уж писать программы стали, то DIY с паяльником тут-же и припасть геркон к самому дешевому термометру дело пяти минут как макс. Не покупать же дорогой датчик двери или счетчик воды с контроллером?

JurySpb, 2023-03-31:
> Какие RSSI интересуют?
> Для дешевого BT адаптера?

Да, все просто. Есть USB свисток с ВТ4 и этот датчик. Хочу положить датчик в баню на даче.
В доме увижу какой то  RSSI и хотелось бы понять плохо это или нет.

> Всё есть. в некоторых аж две кнопки.

Не надо некоторых - вот есть конкретный. Зачем путать людей. Монстр для всего уже есть, только там разбираться - замучаешься.

> `<H` - это [unsigned short](https://docs.python.org/3/library/struct.html)

Хорошо, можно исправлять. Температуры бывают отрицательные, правда в доме такого обычно не бывает. ;) 

> MAC адрес внесен в тело структуры не для того, чтобы его отрезали.

Конкретно в этом примере он зачем? Не надо плодить лишних сложностей. Для новичков, вроде меня это не нужно.

JurySpb, 2023-03-31:
> Сам разбор структуры пакета BLE-рекламы показан в [многочисленных картинках в поисковиках](https://www.google.com/search?q=ble+advertising+packet+format)...

Это познавательно, я даже чуть чуть стал больше понимать, спасибо. Применить правда не смогу, т.к. ВТ4.0, на настольном ПК только 4.2.

closed 2023-03-31 by pvvx

JurySpb, 2023-04-02:
Вас понял, вопросов нет.