pvvx/ATC_MiThermometer #325: Looking for bluetooth 5.0+ usb adapter recommendations

delet-this, 2023-04-01:
Are there any specific, preferably cheap bt 5.0+ usb dongles you would recommend to use with this firmware + LYWSD03MMC? Or should I just choose something from here https://www.home-assistant.io/integrations/bluetooth#realtek-rtl8761bu-adapters

Thanks!

pvvx, 2023-04-01:
There are many RTL8761BU adapters. The speaker in RTL8761BU is loaded by the system, and it makes no sense to look at the label on the adapter. The downloadable FW in RTL8761BU has long supported Bluetooth 5.1.

delet-this, 2023-04-01:
Thank you! I think any RTL8761BU adapter should fit my needs for now.

closed 2023-04-01 by delet-this

pvvx, 2023-04-01:
Orange Pi 3 TG, Pi 5, ... have already been adjusted and are working with the `LE Long Range`.