pvvx/ATC_MiThermometer #326: LCD boot info?

tablatronix, 2023-04-02:
How do I get the boot info back?
My older devices show ATC and the last mac octet ID on the lcd on power up

I am not seeing this now..

pvvx, 2023-04-02:
https://github.com/atc1441/ATC_MiThermometer

tablatronix, 2023-04-02:
Are you saying you removed this, or was this added after forking? I mean its a good feature, I have to write addresses on the back in sharpie...

pvvx, 2023-04-03:
For this function, it is necessary to pick out the battery and break the cover fastenings.
The device name and MAC are changed programmatically.
Identification is performed when connecting to the device and is displayed by the "connections" icon.

tablatronix, 2023-04-03:
failure of clocks, you mean timestamps? I dont use logging. But I do need to know the id of the device I just powered up, doesn't really harm anything.

Popping the battery on a mijia lyd is not that big a deal, maybe on others.

pvvx, 2023-04-03:
The [fw variant from ATC1441](https://github.com/atc1441/ATC_MiThermometer) has the necessary function.

tablatronix, 2023-04-04:
Yours is better 😄

closed 2023-10-04 by tablatronix