pvvx/ATC_MiThermometer #327: CGDK2 -> firmware v4.3 runs fine!

cultur98, 2023-04-03:
This is actually not an issue but I found out with my CGDK2 devices.
I bought 3 where the SHT4x was not detected in v4.2:
Zero values for temperature and humidity.

However the the v4.3 released last week works like a charm!
I tried to measure the current with my [tiny current](https://www.n-fuse.co/devices/tinyCurrent-precision-low-Current-Measurement-Shunt-and-Amplifier-Device.html) device and a scope. But the 300kHz seems to make it impossible to track the fast changing signal.
Perhaps someone has a hint what I could improve.

Other than that the voltage was stable for the last couple of days.
So there seems to be no bigger current leak here.

I keep you updated!
[image]

joe-sydney, 2023-06-13:
How is this going after a couple of months? Any noticeable battery drain?

cultur98, 2023-06-15:
Last week I measured a voltage of 2.705 Volts.
Everything looks quite fine so far.
I am curious how long a battery will last!

cultur98, 2023-07-08:
My CGDK2s seem to work fine.
No measurement spikes experienced so far.

joe-sydney, 2023-07-08:
ok,. one of mine also from time to time dips to about 38% humidity. In the screenshot below, you can see several 38% dips plus one 18% dip...

[image]

pvvx, 2023-12-13:
@joe-sydney - These are problems of adapters, Bluez and Home Assistant integration in Linux.

SouthButovo, 2025-01-14:
@pvvx I have a problem after flashing CGDK2 to bluetooth v49. After the firmware is installed, the display shows zero temperature and humidity. After returning the original firmware, the correct temperature and humidity values ​​are displayed

closed 2025-04-23 by pvvx