pvvx/ATC_MiThermometer #330: MCCGQ02HL support?

miksumin, 2023-04-08:
Добрый день,
Подскажите, возможно ли добавить поддержку MCCGQ02HL (Xiaomi Mijia Window Door Sensor 2)?

pvvx, 2023-04-09:
Я не знаю какой там чип.
И каков в этом смысл? Датчик MCCGQ02HL достаточно дорогой и нераспространённый. Он вроде и так работает нормально?

miksumin, 2023-04-09:
В России датчик стоит 700 руб, что на мой взгляд совсем недорого по сравнению с другими датчиками.
Обратился к Вам, потому что не нашел как его можно прочитать без использования Mi Home и Bind Key.

pvvx, 2023-04-12:
А что пишет [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) на MCCGQ02HL?

miksumin, 2023-04-12:
Пишет "Not found Telink OTA service!" Хотя обновление прошивки по воздуху в Mi Home работает.
Описания и фото внутренностей в Инете не нашел, но могу свой датчик вскрыть и сфоткать, если это поможет.

miksumin, 2023-04-12:
После выполнения Pairing через Mi Home он теперь шлет вот такие данные, судя по всему зашифрованные:
02 01 06
19 16 [95FE] 58588B0986 [4B2563ECAAE4] 83A0F1FB000000D0FF22E7
Bind Key для него я вытащил, осталось сделать дешифровку

miksumin, 2023-04-12:
[image]
[image]

miksumin, 2023-04-12:
Теперь у вас есть информация о чипе. Подскажите, есть возможность сделать поддержку такого типа датчиков?

pvvx, 2023-04-12:
> Bind Key для него я вытащил, осталось сделать дешифровку

Тут есть дешифровка для данного датчика
[ble_monitor](https://github.com/custom-components/ble_monitor/blob/master/custom_components/ble_monitor/ble_parser/xiaomi.py)

> Подскажите, есть возможность сделать поддержку такого типа датчиков?

Данный репозиторий связан только с датчиками температуры и влажности на Telink чипах типа TLSR825x.
А у вас чип от Silabs [EFR32BG21](https://www.silabs.com/wireless/bluetooth/efr32bg21-series-2-socs).
Создавать свою прошивку для данного чипа я точно не буду, т.к. не вижу в этом смысла и у меня нет, и не предвидится, таких датчиков. Для прошивки EFR32BG21 по OTA требуется крипто подпись и никто данными хаками не занимается.

miksumin, 2023-04-13:
Ясно, спасибо.

closed 2023-04-28 by pvvx