pvvx/ATC_MiThermometer #331: Unable to connect to device with atc1441 fw

molnarti, 2023-04-10:
All my Mi Thermometer devices are flashed with the original firmware from 
When i try to update them to this version, I am not able to connect to the devices, getting the following errors:

```
10:52:06 PM: Connecting to: ATC_D9C2EC
10:53:31 PM: MAC: A4C138D9C2EC, Bat: 100%, Vbat: 3155 mV, Temp: 26.4°C, Humi: 43%, Count: 4
10:53:35 PM: NetworkError: Connection Error: Connection attempt failed.
10:53:35 PM: Reconnect 1 from 5
10:53:49 PM: NetworkError: Connection Error: Connection attempt failed.
10:53:49 PM: Reconnect 2 from 5
10:53:54 PM: TypeError: getDevVersion(...).then is not a function
10:53:54 PM: Reconnect 3 from 5
10:53:54 PM: TypeError: getDevVersion(...).then is not a function
10:53:54 PM: Reconnect 4 from 5
10:53:54 PM: TypeError: getDevVersion(...).then is not a function
10:53:54 PM: Reconnect 5 from 5
10:53:54 PM: TypeError: getDevVersion(...).then is not a function
10:53:54 PM: Something went wrong, too many reconnects
10:55:17 PM: Disconnected.
```

I have tried to connect to multiple devices from desktop PC and Android phone as well, but always the same result. What am I doing wrong?

molnarti, 2023-04-12:
how do i flash your fw then over the original atc1441 one? via the old telink flasher?

pvvx, 2023-04-13:
> how do i flash your fw then over the original atc1441 one? via the old telink flasher?

Yes, https://atc1441.github.io/TelinkFlasher.html

closed 2023-04-28 by pvvx