pvvx/ATC_MiThermometer #333: Configuration is not accessible in `TelinkMiFlasher.html`

michaelkleinhenz, 2023-04-14:
I successfully flashed a LYWSD03MMC, but when I connect to it again using `TelinkMiFlasher.html`, I don't get the configuration options though it seems to be connected successfully.

[image]

Any ideas what I am doing wrong there?

michaelkleinhenz, 2023-04-14:
Hitting `Get Config` does nothing, it just displays `Settings 55 was send successful`.

pvvx, 2023-04-15:
So far, this has not been observed. 
Clarifications are needed.
This can happen if the BT adapter remembers the previous version of the device.
Rebooting the BT adapter or removing a previously binded BLE device in Windows may help.

wjcarpenter, 2023-04-24:
I was having the same problem, but the suggestions from @pvvx worked for me. Until this time, I had never seen the "friendly" configuration stuff in Telink Flasher.

I am using a no-name BT adapter on an x86 Linux system (Ubuntu 22.04). Here's what I did. I'm pretty sure not all of these steps are necessary.

- In Telink OTA, connect to the device and flash original firmware. My hardware is version 1.4, so I used downloaded Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0109.bin file.
- Pull the battery from the device and waited more than 30 seconds.
- In my Linux BT "devices" menu, select the device, right click, "untrust" and then "remove".
- Turn my Linux BT off, then back on.
- In Telink Flasher, connect to the device, do the activation step, select the 4.2 custom firmware, and flash it.
- Atain, In my Linux BT "devices" menu, select the device, right click, "untrust" and then "remove".
- Again, Turn my Linux BT off, then back on.
- In Telink Flasher, connect to the device. Configuration options suddenly appear. Yay!

As I said, some of those steps might not be necessary, but I was kind of frustrated so I finally did everything I could think of all at once. (I repeated the above steps just now to make sure I was remembering correctly.)

wjcarpenter, 2023-04-24:
BTW, after going back to the original firmware and then re-flashing the 4.2 custom firmware, the device had somehow remembered the custom firmware options I had set the first time around. The only thing I had to do with set the time again. I don't know where this was remember (browser, device memory, device flash, magic electric spirits in the room, etc).

pvvx, 2023-05-23:
There is a problem with Bluez. When a BLE device is found, its characteristics are recorded in `/var/lib/bluetooth/` with the MAC index.
When scanning is active, files for each device are constantly overwritten. If you have a lot of devices, then the recording stream is almost continuous. This can be seen with the `fatrace -t -f W`.
```
10:13:38.739291 bluetoothd(1152): CW /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:0B:5E:ED
10:13:39.149561 bluetoothd(1152): CWO /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:19:F0:74
10:13:39.151285 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:19:F0:74.WPR6U1
10:13:39.152665 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:19:F0:74.WPR6U1
10:13:39.157835 bluetoothd(1152): CW /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:19:F0:74
10:13:39.669563 bluetoothd(1152): CWO /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:05:F8:FA
10:13:39.671082 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:05:F8:FA.0BEHV1
10:13:39.671640 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:05:F8:FA.0BEHV1
10:13:39.676441 bluetoothd(1152): CW /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/A4:C1:38:05:F8:FA
10:13:40.040981 bluetoothd(1152): CWO /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/E7:2E:01:83:48:6E
10:13:40.042466 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/E7:2E:01:83:48:6E.G0R2U1
10:13:40.043057 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/E7:2E:01:83:48:6E.G0R2U1
10:13:40.047803 bluetoothd(1152): CW /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/E7:2E:01:83:48:6E
10:13:41.769233 bluetoothd(1152): CW /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/58:2D:34:14:50:87
10:13:41.770424 bluetoothd(1152): W /var/lib/bluetooth/8C:88:2B:00:E7:EA/cache/58:2D:34:14:50:87.W63HV1
```

---

It is possible that clearing `/var/lib/bluetooth/` can help with the memory problem of the characteristics of the previous device.
The current solution in "Bluez" with recording characteristics is not yet a correct and proven solution, since it increases the TBW of disks very quickly and there is no way to redirect to "temp". It stores the characteristics and the BT adapter.

misdoro, 2023-08-21:
Strangely, for me configuration tab works straight away if I flash the firmware ATC42.bin, and keeps working if I go to ATC43.bin.

Doesn't work if I flash from stock to ATC43 directly.

closed 2023-12-23 by pvvx