pvvx/ATC_MiThermometer #336: cgdk2-2

FreelandA, 2023-05-07:
I bought what I thought was a new batch of CGDK2s, but turned out to be a new revision, the CGDK2-2. I can't get them to connect to the Telink web app. Are they compatable?

pvvx, 2023-05-09:
You have provided very little information to answer. Several new CGDK2 have appeared for a long time. All of them are supported.

FreelandA, 2023-05-10:
I just checked the listing and they added a not that it doesn't support bluetooth. 
https://www.aliexpress.us/item/3256801989523547.html?spm=a2g0o.order_list.order_list_main.22.d6d518027goeyi&gatewayAdapt=glo2usa

closed 2023-05-10 by FreelandA

pvvx, 2023-05-12:
This is "Qingping Temp & RH Monitor Lite E".
And this is the old version.
[CGDK2-2](https://pvvx.github.io/CGDK2/CGDK2-2)