pvvx/ATC_MiThermometer #338: I don't see a HW: B1.5 of LYWSD03MMC mentioned, which I have. Do you know if it is also supported or not?

dzidza, 2023-05-10:
These are relevant info from the logs:

Hardware Revision String: B1.5
Software Revision String: 0130
Firmware Revision String: 1.0.0_0130

Thanks

pvvx, 2023-05-11:
Supported. 
But they are rare and I did not make a complete test.

dzidza, 2023-05-12:
It works great so far.
Using BTHome format, adjusting a different name, advertising interval, setting a bind key, and a few more tweaks, it all worked first try, and is logging as expected in HA.
The only strange thing I noticed is that after the flashing, it is now recognized as "Hardware Revision String: **B1.6**" instead of 1.5. not sure if that can cause a problem if I try to flash it again next time.

Anyway, thanks for the awesome job you did ;)

jddonovan, 2023-05-17:
Photos of the 1.5 version since I couldn't find any. Manufactured December 2022. Bought few months ago.
[image]

pvvx, 2023-05-18:
Could you clarify what the "02" command will show on version B1.5?

Example (CGDK2):

[image]

[image]

jddonovan, 2023-05-22:
Old FW didn't show anything on "02", but after upgrading to 4.3 beta:

09:06:03: Settings 02 was send successful
09:06:04: Sensor: I2C addres 0x70, LCD driver: SPI or UART

09:06:53: Settings 05 was send successful
09:06:54: Sensor ID: 00000887

dzidza, 2023-05-22:
Yes, it's the same with my three devices.

`8:52:18 AM: Settings 02 was send successful
8:52:19 AM: Sensor: I2C addres 0x70, LCD driver: SPI or UART`

Ircama, 2023-05-22:
Alternative query from [the *atc_mi_interface* package](https://github.com/pvvx/ATC_MiThermometer/tree/master/python-interface#general-usage):

```
python3 -m atc_mi_interface -c -icm A4:C1:38:AA:BB:CC
```

Check numbers:
```
...
19 Hardware Revision
...
21 Software Revision
...
2: Internal devices
...
3: I2C scan
...
85: Internal configuration
    firmware_version:
...
```

pvvx, 2023-05-23:
Version 4.3 has built-in recognition of HW: B1.5.
There is no way to fully check - out of stock LYWSD03MMC with HW: B1.5

jddonovan, 2023-05-24:
4.3 software now detects the version correctly:

14:38:24: Hardware Revision String: B1.5
14:38:24: Software Revision String: V4.3
14:38:24: Firmware Revision String: [github.com/pvvx](http://github.com/pvvx)
14:38:24: Detected custom Firmware
14:38:29: Hardware Version: LYWSD03MMC B1.5, Software Version: 4.3, Sensor: SHTC3 (SHTV3)
14:38:29: Custom config HEX string: 55012000009004a97cb38a3c

Thank you!

yousaf465, 2023-05-28:
How to check the revision from the packing or the thermometer itself?

jddonovan, 2023-05-28:
> How to check the revision from the packing or the thermometer itself?

You need to open the inside back panel with torx screws. Then you can see the revision number on the pcb board. The packaging will tell you the manufacturing time, but that's only loosely correlated with the revision version.

yousaf465, 2023-06-05:
> > How to check the revision from the packing or the thermometer itself?
> 
> You need to open the inside back panel with torx screws. Then you can see the revision number on the pcb board. The packaging will tell you the manufacturing time, but that's only loosely correlated with the revision version.

[image]

My new ones are HW1.5 manufactured 2022