pvvx/ATC_MiThermometer #339: [WorkInProgress] Meawow MHO-C122 (2AWMOMHOC122) support

FaBjE, 2023-05-13:
## Introduction
Recently I acquired a few of the Meawow MHO-C122 (2AWMOMHOC122) sensors for a bargain price.
Hoping I was able to scrape the sensor data from the advertisement or basic readout of some characteristics.
Long story short, with the default firmware: Data is not advertised, Readout is only on a proprietary protocol, you can only connect over Bluetooth when you just inserted the battery.

I tried if the OTA update page of this project had any luck but unfortunately not. The published characteristics are nowhere near what the default Xiaomi units publish.

```
4:11:00 PM: Searching for devices
4:11:02 PM: Connecting to: C121
4:11:13 PM: Not found Telink OTA service!
4:11:13 PM: Disconnected.
```

Opening up the unit I saw the PCB with almost identical setup/components as the Xiaomi units.
No wonder as these are produced by the same company (I believe)
Giving me the Idea if with maybe some minor tweaks the software of this project can be run on those  units as well.
This issue will be my little blog/progress report. Feel free to help out if you have anything to add.

## Meawow MHO-C122 (2AWMOMHOC122) 
https://meawow.com/product/thermo-hygrometer/mho-c122/

### Exteriour
The unit LCD/casing is identical to the MHO-C101

### Interiour
I took the following pictures of the PCB
Front:
[image]
Back:
[image]

On that we can see the (familiar)
MCU: TLSR8251F512ET24
LCD: IST3055NA0
Sensor: (Probably) SHTV3

### Software
The default software gives the following log when connecting with the nRF Connect app.
[meawow connection log.txt](https://github.com/pvvx/ATC_MiThermometer/files/11469623/meawow.connection.log.txt)

### ToDo List
- [ ] Establish UART connection
- [ ] Retrieve original firmware
- [ ] Program with custom firmware + test

AvA1970, 2023-05-13:
I have purged also 5 of these. Would be nice to flash firmware or te advertise ble so HA could pick these up.

I will follow this topic with interest.

killabeenl, 2023-05-17:
I have also a couple of these and it would be nice if we can flash them to use with HA

NH-Networks, 2023-05-19:
same here! hope someone can figure this out..

ZekiZico, 2023-05-20:
I'm interested in this too, can't get them connected to HA

FaBjE, 2023-05-20:
I've made some progress. I have made a little programmer jig with a CHD340 I'm able to readout the flash contents of the MCU.
I should be able to program the unit with new firmware, But I haven't tried (yet).
You only need the three connection pads on the right of the PCB.

[Original_Firmware_MHO_C122.zip](https://github.com/pvvx/ATC_MiThermometer/files/11522404/Original_Firmware_MHO_C122.zip)

Using this schematic/instruction:
https://github.com/pvvx/TlsrComSwireWriter#telink-swire-simulation-on-a-com-port

- VCC is connected to the RTS to be able reset the MCU
- The PCB is clamped between a dual-row header (and some cardboard stuff) This provides exact spacing and does not require soldering to the PCB.

[image]

#### Things I have learned
- The silkscreen labels on the PCB can be deceiving. Reset is not the MCU reset, but just an IO pin.
- You cant use the programmer scripts from a Virtual machine, WSL or a docker container. The timing becomes too unstable.
- A full readout on the flash takes about 15 minutes. (for 512KB) But hey, I'm glad it works without buying an expensive programmer.
- This command worked best for me: `\python.exe TLSR825xComFlasher.py -p COM23 -t 70 rf 0 0x80000 firmware_download.bin`

ZekiZico, 2023-05-20:
Good job!

NH-Networks, 2023-05-20:
I will use your findings and will try out to replicate it, have all tools
needed.

Do you think we can flash the same firmware without any modifications?

FaBjE, 2023-05-21:
Some progress was made. I managed to flash the ATC_v42.bin. and the good news is that bluetooth works.
The sensor and battery readout seems to work fine.
But the LCD is messed up. It seems to work, but the segment layout is different, so you get really weird data displayed.

I tried some other firmwares and these are the results:
- ATC_v42.bin: LCD Seems to work messed up segments, BT Temp/Humidity OK 
- MHO_C401_v42: No LCD, BT Temp/Humidity OK
- MHO_C401N_v42.bin: No LCD, BT Temp/Humidity OK

#### Things I Learned
- My programming jig is really temperamental. So don't recommend copying it. I suspect the power from the RTS pin is not sufficient.

@pvvx Is there a way to easily identify the LCD segment layout? Like flash each segment one by one?

pvvx, 2023-05-21:
[Control function ID when connected](https://github.com/pvvx/ATC_MiThermometer#control-function-id-when-connected)
|  ID  | Command                                       |
| :--: | --------------------------------------------- |
|0x60 | Get/Set LCD buffer|

[image]

[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html).

6001020304...
60 - command
01020304... - lcd buffer

Creates a bit-map similar to the one shown here - [epd_mho_c401n.c](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/epd_mho_c401n.c#L17)

pvvx, 2023-05-22:
> [Original_Firmware_MHO_C122.zip](https://github.com/pvvx/ATC_MiThermometer/files/11522404/Original_Firmware_MHO_C122.zip)

OTA File -> https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_MHO_C122_tuya.bin

pvvx, 2023-05-23:
The latest versions of Tuya BLE transmission of advertising packages with sensor data. The transfer occurs when the change is made.
Also in Tuya BLE there is a download OTA with a key signature. Some of the codes can be found on the Tuya Cloud.

"LocalTuya integration"  in the "HA" misleads users by not indicating that BLE is not supported.

FaBjE, 2023-05-26:
I've made an LCD segment map. I'll try to cook up a lcd driver for it.
[image]

pvvx, 2023-05-27:
> ATC_v42.bin: LCD Seems to work messed up segments, BT Temp/Humidity OK

What is the [address of the LCD controller](https://github.com/pvvx/ATC_MiThermometer/issues/338#issuecomment-1553563595)?

pvvx, 2023-05-28:
I recorded the firmware MHO-C122 in Xiaomi LYWSD03MMC.

Registered in "Smart Life":

[image]

[image]

Average Power:

[image]

nRFConnect (advertising):
[image]
Data is transmitted from the sensor.
[image]

https://iot.tuya.com/  Device Management -> Get device details:
```
{
  "result": {
    "active_time": 1685239585,
    "biz_type": 18,
    "category": "wsdcg",
    "create_time": 1685239585,
    "icon": "smart/icon/bay1607936793441YbTx/03baef663c3c1a2149c7fbf98110e134.png",
    "id": "bf7xxxxxxxxhrn",
    "ip": "",
    "lat": "xx.9400",
    "local_key": "d+xxxxxxxxxxxxxx9-^",
    "lon": "xx.1000",
    "model": "MHO-C121",
    "name": "Temperature Humidity Sensor",
    "online": false,
    "owner_id": "47xxxx61",
    "product_id": "3nxxxxw6",
    "product_name": "Temperature Humidity Sensor",
    "status": [
      {
        "code": "va_temperature",
        "value": 223
      },
      {
        "code": "va_humidity",
        "value": 39
      },
      {
        "code": "battery_percentage",
        "value": 100
      },
      {
        "code": "temp_sampling",
        "value": 6
      },
      {
        "code": "temp_unit_convert",
        "value": "c"
      }
    ],
    "sub": false,
    "time_zone": "+03:00",
    "uid": "eu164xxxxxxxxxxxx5HE",
    "update_time": 1685239587,
    "uuid": "15xxxxxxxxxxx9c9"
  },
  "success": true,
  "t": 1685xxxxx7614,
  "tid": "78xxxxxxxxxxxxxxxxxxxxxxxxxxxx2a"
}
```
* The data-keys has been changed (replaced with "xxxxx").

pvvx, 2023-05-28:
> Apologies for the mix between 0 indexed (bytes) and 1 indexed (bits), I guess I wasn't fully awake yet....

[image]

pvvx, 2023-05-28:
Added a test version of MH OC 122.
So far, flash programming is only via hardware - SWS.
[TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) does not work with the original MHO-C122. With firmware MHO_C122_v43.bin something will work...
Check if everything is correct with the display.

FaBjE, 2023-05-28:
> What is the [address of the LCD controller](https://github.com/pvvx/ATC_MiThermometer/issues/338#issuecomment-1553563595)?
```
12:30:06 PM: Settings 02 was send successful
12:30:07 PM: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3C
```

> And how are things with Tuya OTA?
Could not tell you. I tried an OTA in the app, but there was "no update available".

> [image]

Whoops. I guess i was really still sleeping...
I checked and yes your correction is right.

> Check if everything is correct with the display.

Thanks so much! I just can't find the time...
I will test this right now and report back.

FaBjE, 2023-05-28:
@pvvx 
I tested the LCD driver, it had some issues unfortunately.
But I manged to find them and submit a pull request with fixes: https://github.com/pvvx/ATC_MiThermometer/pull/344
Afaik it works good now, It's on my desk and I will monitor it.

pvvx, 2023-05-28:
To complete work with MHO-C122(C121), an OTA is required.
This is not the first thermometer [with TUYA BLE PROTOCOL VERSION 4](https://pvvx.github.io/TH-05/).
I haven't seen any solutions for the Tuya BLE (new version) OTA.
Also, there is no integration local [BLE Tuya in "HA"](https://github.com/custom-components/ble_monitor/issues/1063).

pvvx, 2023-05-29:
The description of the OTA protocol is in the cloud.

But you need to register.

https://developer.tuya.com/en/docs/iot-device-dev/OTA_BLE
---
[image]
---

It seems to give out all the keys.

---
[image]
---

It remains to understand the encryption method.

FaBjE, 2023-05-29:
https://developer.tuya.com/en/docs/iot-device-dev/OTA_BLE?id=Kawer1ehuv8si seems freely available to me.
The example document at the bottom: https://images.tuyacn.com/smart/docs/OTA-update-sample.docx seems to list an example how to implement the tuya way in the MCU. 
Maybe this helps you to implement it in the TelinkFlasher page?

pvvx, 2023-05-30:
[image]

Full sample of Telink SDK Tuya.
https://github.com/pvvx/pvvx.github.io/blob/master/MHO_C122/bin/ble_sdk_multimode_tuya.zip

[image]

[image]

pvvx, 2023-05-30:
Maybe this helps you to implement it in the TelinkFlasher page?

There is still a lot to figure out. 
1. Add to the firmware the initial correction of the OTA download to another area. 
2. Restoring Tuya firmware.
3. Request keys from the user or from the cloud
But the main thing is how to establish a connection with the cipher.

I didn't find anyone doing this. Disassembled the old version and with Wi-Fi. I haven't seen anyone who could handle the new version of Tuya.

pvvx, 2023-05-30:
I found the source package [tuya_ble_sdk_Demo_Project_tlsr8253](https://github.com/pvvx/pvvx.github.io/blob/master/MHO_C122/bin/tuya_ble_sdk_Demo_Project_tlsr8253.zip) in the archive.

FaBjE, 2023-05-30:
> There is still a lot to figure out.
>
>     1. Add to the firmware the initial correction of the OTA download to another area.
>
>     2. Restoring Tuya firmware.
>
>     3. Request keys from the user or [...]

I think the most beneficial one is to "update" from stock tuya firmware to "atc firmware"
Going back to Tuya firmware would be a "nice to have" imho. But I don't think it will be a much used feature.

> PS: And for the whole summer I usually take a break from working at computers...
You should definitely do that too :)

FaBjE, 2023-06-10:
I've been running this firmware: https://github.com/pvvx/ATC_MiThermometer/raw/20a087dcc2c234f48790ee618d7273444288a2b9/MHO_C122_v43.bin on three units now for two weeks.
Without any problems. So I'm declaring this "finished/stable"

For the initial firmware update a programmer is needed.
After the initial 'convert' the updates can be done over BLE using the telink flasher page.

AvA1970, 2023-06-12:
Hey, i do not have a programmer and also no skills to make the programmer. Is it possible to send my 5 units so someone can program the firmware for me?

TrevorVr, 2023-06-13:
Awesome work! Is there any way in which the firmware can be flashed for mortals like me without having additional hardware to do so?

pvvx, 2023-06-13:
At the moment, it is possible to flash using a USB-COM adapter.
Connect only TX-SWS and GND wires:
https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version

It's boring to disassemble the BLE-Tuya firmware protocol alone... Analysis of BLE-Tuya is possible only in autumn... In the summer, I limit my work at the computer.

cardshare007, 2023-06-17:
Hi Guys, have been following this post and reading a lot. I am actually quite stuck now...I took the following steps:
1) solder 2 wires to the GND and SWS
2) connect to USB serial flasher;
3) opened the USB com flasher html
4) tried two different files: MHO_C122_v43.bin and ATC_v43.bin
Now two C122 don't seem to start when I assemble them again. The screen stays blank.
I tried 2 different usb serial flashers. What am I doing wrong? Should it first be put in flash mode or what am I missing?
Much appreciated all your hard work and thanks in advance for your help!

FaBjE, 2023-06-17:
> Hi Guys, have been following this post and reading a lot. I am actually quite stuck now...I took the following steps:
>
>     1. solder 2 wires to the GND and SWS
>
>     2. connect to USB serial [...]

If you insert the battery, can you see the device over bluetooth?
You can use https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US to check.

If so, the [zebra strip](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US) of the LCD might be mis-aligned.

If not, there might be an other issue, I have not used the flasher html page myself.
But if the device behavior changes there is at least something happening.

cardshare007, 2023-06-17:
> If you insert the battery, can you see the device over bluetooth? You can use https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US to check.
>
> If so, the [zebra [...]

Seems to be completely dead, I see nothing happening, also no Bluetooth signal. I have a third one, so there is another chance, but then I have to be sure I am doing the right thing...any comments on the steps I took? It feels somehow I miss anything

FaBjE, 2023-06-17:
> Seems to be completely dead, I see nothing happening, also no Bluetooth signal. I have a third one, so there is another chance, but then I have to be sure I am doing the right thing...any comments on the steps I took? It feels somehow I miss anything

I would not sacrifice your third one yet. Flashing should not be "destructive." If you flash wrong software, you should be able to reflash good one.
Your steps seem OK to me, can you show a picture of your setup? 
How did you connect to GND and SWS to your USB serial?

cardshare007, 2023-06-17:
[image]

cardshare007, 2023-06-17:
I think that follows the instruction properly right?

FaBjE, 2023-06-17:
Upper wire is GND, Lower wire is SWS.
What serial converter do you use? What voltage levels is it (set to)?
You should use 3V3 logic level voltages (maximum)
And you should place the battery when programming, as you are not powering the unit by wire.

NH-Networks, 2023-06-17:
Got these pins DTR, RXD, TX, VCC, CTS, GND.

On wich pin do i need to connect sws ?

FaBjE, 2023-06-17:
Serial Converter TX -> SWS 
Serial Converter GND -> GND

Make sure to use 3V3 logic levels!
FTDI Converters are reported **not** to work
I build mine using a [CH340E based](https://www.tinytronics.nl/shop/nl/communicatie-en-signalen/serieel/usb/ch340e-ttl-usb-naar-serieel-converter-3.3-5v) converter (should be available on aliexpress as well)

ZekiZico, 2023-06-17:
@FaBjE you didn't use the html page to flash, but how did you flash the thermometer? I saw in one of your first posts 
that you used python script ?

cardshare007, 2023-06-17:
> Upper wire is GND, Lower wire is SWS. What serial converter do you use? What voltage levels is it (set to)? You should use 3V3 logic level voltages (maximum) And you should place the battery when programming, as you are not powering the unit by wire.

Sure I use the 3.3, but I think the middle is the sws or am I missing something?

cardshare007, 2023-06-17:
> Upper wire is GND, Lower wire is SWS. What serial converter do you use? What voltage levels is it (set to)? You should use 3V3 logic level voltages (maximum) And you should place the battery when programming, as you are not powering the unit by wire.

What I didn't do is use the battery to power it. Is this necessary?

pvvx, 2023-06-17:
> FTDI based converters are reported not to work

If RX. if FTDI is used, it checks the bit frame, but it does not match the 1-wire Telink
Chinese chips do not check each bit for several samples and the total timing of the RS symbol.
When an error is detected, RX FTDI does not output the violated character to the buffer, but sets an error flag.

FaBjE, 2023-06-17:
> @FaBjE you didn't use the html page to flash, but how did you flash the thermometer? I saw in one of your first posts that you used python script ?

I use the https://github.com/pvvx/TlsrComSwireWriter from the repository. But I think you also need to connect the RX for that.
Using a resistor https://github.com/pvvx/TlsrComSwireWriter#telink-swire-simulation-on-a-com-port

> but I think the middle is the sws or am I missing something?

Yes middle connection pad, but his lower wire.

> What I didn't do is use the battery to power it. Is this necessary?

Yes, see pvvx his explanation.

> If RX. if FTDI is used, it checks the bit frame, but it does not match the 1-wire Telink

Thanks for explaining

cardshare007, 2023-06-17:
Thanks mate, will try with a battery and let you know the results

pvvx, 2023-06-17:
1. [USBCOMFlashTx.html](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) has no feedback. Perhaps the firmware will be executed unsuccessfully. In this case, repetition is required.
2. And the second problem is that the chip may be in sleep mode. For reliable programming, it is necessary to turn on the power during activation (ATime) or use a reset signal from the chip.
3. After such programming, it is necessary to make a new OTA, since there is no guarantee of correct firmware using 'USBCOMFlashTx'.

NH-Networks, 2023-06-18:
Tried on breadboard yesterday with ch340 of wittyboard gnd sws and rst connected, but cant seem to flash new firmware.. do i need to set it in flash mode somehow? Every try i did it still had original firmware afterwards

cardshare007, 2023-06-18:
[image]

cardshare007, 2023-06-18:
What I did:
1) tried with a battery and only connect SWS and GND: nothing happens
2) see the picture: I also tried to connect the VCC on the 3.3: no result
3) I don't use FTDI but a wchusbserial1420

Still nothing seems to happen - please let me know if there are any further ideas. Thanks a lot for your support!

cardshare007, 2023-06-18:
> I use the https://github.com/pvvx/TlsrComSwireWriter from the repository. But I think you also need to connect the RX for that. Using a resistor [...]

I was struggling a bit to get this to work since I think python on my Mac is giving some troubles. Also, I see a different way of connecting the devices, I havn't been using resistors and not sure how to do this. Let me know if this is absolutely a necessity and I will check this direction...

FaBjE, 2023-06-18:
> do i need to set it in flash mode somehow

Yes, programming mode is achieved by sending a "magic sequence" on the SWS line.
This must be done directly after power-up. Therefore I / the script uses the "RTS" line to switch power on the right moment.
You can either switch the power yourself and tune with the "activation time" or let the script do it for you using the RTS line.
My USB > UART converter was able to supply enough current over this line to power the programming circuit, but there is no guarantee.
I only used the script from the repository:
`\python.exe TLSR825xComFlasher.py -p COM24 -t 50 wf 0 .\ATC_Thermometer.bin`

> Let me know if this is absolutely a necessity and I will check this direction...

Well it is not an absolute necessity, but the "TX only" method, is a bit risky.
The resistor is used to make from a 1 signal wire "uart" (The SWS protocol), a two-wire signal for a normal USB->UART converter.
When only the TX is connected the webpage/script just sends the data hoping the microcontroller accepts it. (Having no clue if it does or not)

With the resistor (and the RX line connected) the Python script from this repository is able to receive the reply from the micro-controller and check if it has accepted the data or not.

To all of you, I had a lot of struggles getting this to work either. It is simulating a (expensive) programmer on cheap hardware. So it is all a bit "tongue at the right angle" and hoping it works stuff.
Sometimes I just have reconnect the USB device and re-run the script and it does work for some magic reason.

cardshare007, 2023-06-18:
Thanks mate, seems a bit more complex than I initially thought. I will give it another shot next weekend, but for now I'll leave it. I also hope the OTA will appear in a few months, seems to be my best bet ;-)

rpvandoorn, 2023-07-10:
Hi all,

Just for my understanding. Can I already use the BLE updater for the C122 ?

Thanks all for you efforts on this project.

pvvx, 2023-07-11:
I will start considering the OTA option (firmware update via Bluetooth) only after the end of summer, when we run out of warm weather.
And it is desirable that by that time someone has support for receiving advertising the "Tuya" format. Otherwise, support for BLE advertising in the "Tuya" format will never appear in anyone's open-source.

pvvx, 2023-11-28:
You forgot about the second point of the conditions.
Nobody wants to deal with Tuya.
Closed https://github.com/rospogrigio/localtuya/issues/1166 as completed.
https://github.com/custom-components/ble_monitor/issues/1063
...

christiaandouma, 2023-12-01:
Nice work so far. I got two of c122 flashed. They work and can connect with smart life from tuya! After a minute or two they go offline and don't report their status anymore untill i remove the battery and readd them in the app. 

Just wondering if we can expect zigbee because that is something thats not dependend on the smart life app and online access. Next to that it would be easy to integrate into home assistant. 

Anyway thanks so far!

pvvx, 2023-12-01:
> Was looking at https://github.com/pvvx/ZigbeeTLc but don't seem to get it to work

What didn't work?
I don’t have an MHO-С122 and have nothing to test it on.
And there is an even more exotic option - https://github.com/pvvx/BZdevice 
https://github.com/pvvx/BZdevice/raw/master/bin/1141-020b-01113001-ZMHOC122.zigbee
But there are still problems with BLE OTA...

christiaandouma, 2023-12-01:
@pvvx I was able to flash with https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html and the tuya binary. I couldnt flash the zigbee binary using the same url. How should I flash with the zigbee version ?

pvvx, 2023-12-04:
> This is for TH05?

MHO-С122

almirus, 2023-12-15:
@pvvx это не оно? https://github.com/SoCXin/PHY6222

pvvx, 2023-12-15:
No. This is the SDK from the PHY website shown earlier in the link.
But http://wiki.phyplusinc.com/ is not available.

[10 more replies]