pvvx/ATC_MiThermometer #340: B1.5/v4.2, cannot update/wrong advertising packets. Other versions work fine.

jddonovan, 2023-05-24:
I have two B1.5 devices that did not deliver advertising data to HA. Both devices worked perfectly fine at first. 40 minutes after replacing the battery, the issue started. Re-replacing the battery did not fix the issue. Dropping the transmit power from 0db to -5db did not help. I have more B1.5, but these seem to be working fine still.

I can connect with the flasher-page and I can change settings and see current temperature. I tried updating to v4.3 to see if it helps, but update always fails:

<details><summary>Log when updating:</summary>
- 13:05:18: Searching for devices
- 13:05:37: Connecting to: ATC_EC91F8
- 13:05:45: Hardware Revision String: B1.6
- 13:05:45: Software Revision String: V4.2
- 13:05:45: Firmware Revision String: [github.com/pvvx](http://github.com/pvvx)
- 13:05:45: Detected custom Firmware
- 13:05:45: Hardware Version: LYWSD03MMC B1.6, Software Version: 4.2, Sensor: SHTC3 (SHTV3)
- 13:05:45: Custom config HEX string: 55011000009004a931b384b4
- 13:05:48: Load firmware file 'ATC_v43.bin'...
- 13:05:48: File: ATC_v43.bin
- 13:05:48: File size: 81284 bytes
- 13:05:48: Count: 5081
- 13:05:49: Start DFU
- 13:05:55: Update error: NotSupportedError: GATT Error Unknown.
- 13:05:55: Disconnected.
</details>

I then scanned the advertising data, and it seems that the raw data is quite different from what it's supposed to be.
<details><summary>Advertisement data:</summary>
- Device Address : A4:C1:38:EC:91:F8
- Device Name : ATC_EC91F8
- Rssi : -25 dBm
- Raw Data : 02 01 06 12 16 1A 18 F8 91 EC 38 C1 A4 EA 0A 00 0C 9B 0C 64 19 04 0B 09 41 54 43 5F 45 43 39 31 46 38
</details>

I believe have sent this configuration string to all B1.5 and other versions:
550110000090049A31B304B4
Re-setting to defaults settings did not fix the issue.

What works:
- changing settings

What doesn't work:
- advertising data corrupted(?)
- updating to new firmware OTA

jddonovan, 2023-05-24:
[image]
Both devices are B1.5 with same firmware. Batteries were changed around 13:30.

The other failed some 40 minutes later. It's possible there was problems with power contants, but it should not cause permanent changes to FW. I cannot remember if I changed the settings when it stopped working. But it shouldn't be bricked because of it.

pvvx, 2023-05-27:
> Advertisement data:
> Device Address : A4:C1:38:EC:91:F8 Device Name : ATC_EC91F8 Rssi : -25 dBm Raw Data : 02 01 06 12 16 1A 18 F8 91 EC 38 C1 A4 EA 0A 00 0C 9B 0C 64 19 04 0B 09 41 54 43 5F 45 43 39 31 46 38

```
02 01 06 12 - BLE Flags
16 1A 18 - UUID:181A
	F8 91 EC 38 C1 A4 
	EA 0A - 2794 - 27.94C
	00 0C - 3072 - 30.72%
	9B 0C - 3227 - 3.227V
	64  - 100 - 100%
	19 - 25 - id-count:25
	04 - flags
0B 09 - name
	41 54 43 5F 45 43 39 31 46 38 - ATC_xxxx
```
All Ok

Ircama, 2023-05-28:
In addition to pertinent @pvvx's note:

> ...
> I believe have sent this configuration string to all B1.5 and other versions: 550110000090049A31B304B4 Re-setting to defaults settings did not fix the issue.
> ...

```python
from atc_mi_interface import cfg, normalize_report

STR = "550110000090049A31B304B4"

print(normalize_report(str(cfg.parse(bytes.fromhex(STR)))))
```

We lazily use the command ("55") just in place of the initial version byte, needed for decoding with the "*cfg*" `construct` model.

``` Result
Container:
    firmware_version: <<< DISREGARD THIS
        major = 5
        minor = 5
    flg:
        lp_measures = False
        tx_measures = False
        show_batt_enabled = False
        temp_F_or_C = (enum) temp_C 0
        blinking_time_smile = (enum) blinking_smile 0
        comfort_smiley = False
        advertising_type = (enum) adv_type_custom 1
    flg2:
        screen_off = False
        longrange = False
        bt5phy = False
        adv_flags = True
        adv_crypto = False
        smiley = (enum) smiley_off 0
    temp_offset = 0.0
    temperature_unit = '°C' (total 2)
    humi_offset = 0.0
    humidity_unit = '%' (total 1)
    advertising_interval = 9.0
    adv_int_unit = 'sec.' (total 4)
    measure_interval = 4
    rf_tx_power = (enum) RF_POWER_Negative_3p61_dBm 154
    connect_latency = 1.0
    connect_latency_unit = 'sec.' (total 4)
    min_step_time_update_lcd = 8.950000000000001
    min_s_t_upd_lcd_unit = 'sec.' (total 4)
    hw_cfg:
        sensor = (enum) sensor_SHT4x 0
        reserved = 0
        hwver = (enum) hwver_LYWSD03MMC_B1_6 4
    averaging_measurements = 180
```

As mentioned, the first byte ("55", which is actually a command and not a version ID) and its decoded data (`firmware_version=Container(major=5, minor=5)`) need to be disregarded. All other data are pertinent.

The default values are preconfigured in the tool and should reflect the firmware default settings (provided that I was not wrong [when developing it](https://github.com/pvvx/ATC_MiThermometer/blob/ce91d1c89bba3fca92340e3ada61bd9d12b9f91f/python-interface/atc_mi_interface/atc_mi_config.py#L207)...):

```
python3 -m atc_mi_interface.atc_mi_config -t -m 11:22:33:44:55:66
```

Output:

```
...

85: Internal configuration
    firmware_version:
        major = 4
        minor = 3
    flg:
        lp_measures = True
        tx_measures = False
        show_batt_enabled = False
        temp_F_or_C = (enum) temp_C 0
        blinking_time_smile = (enum) blinking_smile 0
        comfort_smiley = True
        advertising_type = (enum) adv_type_custom 1
    flg2:
        screen_off = False
        longrange = False
        bt5phy = False
        adv_flags = True
        adv_crypto = False
        smiley = (enum) smiley_off 0
    temp_offset = 0.0
    temperature_unit = '°C' (total 2)
    humi_offset = 0.0
    humidity_unit = '%' (total 1)
    advertising_interval = 2.5
    adv_int_unit = 'sec.' (total 4)
    measure_interval = 4
    rf_tx_power = (enum) RF_POWER_Positive_0p04_dBm 169
    connect_latency = 1.0
    connect_latency_unit = 'sec.' (total 4)
    min_step_time_update_lcd = 2.45
    min_s_t_upd_lcd_unit = 'sec.' (total 4)
    hw_cfg:
        sensor = (enum) sensor_SHT4x 0
        reserved = 0
        hwver = (enum) hwver_LYWSD03MMC_B1_6 4
    averaging_measurements = 180
...

```

Notice that `hex(85)` is 0x55.

I generally use [WinMerge](https://winmerge.org) to quickly compare outputs, running the tool itself, creating a new text (Control N), then pasting the two outputs (left and right parts respectively) and pressing F5 (Refresh). The following values look different:

```
lp_measures
comfort_smiley
advertising_interval
rf_tx_power
min_step_time_update_lcd
```

Isn't for instance the `advertising_interval` (9.0 secs.) too slow?

pvvx, 2023-05-28:
> Isn't for instance the advertising_interval (9.0 secs.) too slow?

On all recent (two years) smartphones with Android, the Bluetooth SIG BLE standard is supported - the advertising interval is up to 10 seconds.
On Windows, there are exceptions related to the BT adapter. But most support an interval of up to 10 seconds.

pvvx, 2023-05-28:
Corrected code. Initial source: https://gist .github.com/mironovdm/cb7f47e8d898e9a3977fc888d990e8a9
```
/* https://gist.github.com/mironovdm/cb7f47e8d898e9a3977fc888d990e8a9 
* gcc main.c -o leactimeout 
* $ sudo leactimeout --read    # Show current HCI_LE_AUTOCONN_TIMEOUT
* $ sudo leactimeout           # Set new value for le autoconnect timeout
*/
#include <endian.h>
#include <stdbool.h>
#include <stdlib.h>
#include <stdio.h>
#include <sys/socket.h>
#include <unistd.h>

#include "bluetooth/bluetooth.h"
#include "bluetooth/hci.h"

#define NEW_LE_AUTOCONN_TIMEOUT 22000

struct hci_event {
	uint16_t code;
	uint16_t controller_id;
	uint16_t size;
	uint16_t cmd_opcode;
	uint8_t cmd_status;
};

#define EVT_CODE_CMD_COMPLETE 0x0001
#define EVT_CMD_STATUS_SUCCESS 0x00

#define PARAM_TYPE_LE_AUTOCONN_TIMEOUT 0x001b

unsigned char read_buf[1024];
const unsigned char read_conf_cmd[6] = {
	/* cmd opcode */ 0x4b, 0x00, 
	/* controller id */ 0x00, 0x00,
	/* data len */ 0x00, 0x00};
unsigned char set_le_autoconn_timeout_cmd[] = {
	/* cmd opcode */ 0x4c, 0x00, 
	/* controller id */ 0x00, 0x00, 
	/* data len */ 0x05, 0x00, 
	/* data */ 0x1b, 0x00, 0x02, 0x00, 0x00
};
const int set_timeout_cmd_value_offset = 9;

int get_autoconn_timeout_from_result(const unsigned char *buf, size_t size)
{
	/* Skip event header */
	buf += 6;

	const unsigned char * const buf_end = buf + size;

	while (buf < buf_end) {
		uint16_t param_type = le16toh(*(uint16_t *)buf);
		uint8_t param_data_size = *(uint8_t *)(buf + 2);
		
		if (param_type == PARAM_TYPE_LE_AUTOCONN_TIMEOUT) {
			uint16_t value;

			switch (param_data_size) {
				case 1:
					value = *(uint8_t *)(buf+3);
					break;
				case 2:
					value = le16toh(*(uint16_t *)(buf+3));
					break;
				default:
					fputs("Unexpected value length", stderr);
					return -1;
			}

			printf("LE Autoconnect Timeout=%d \n", value);

			return value;
		}

		buf += 2/*type*/ + 1/*len*/ + param_data_size;

	}
	puts("Param type LE Autoconnect Timeout not found in result");

	return -1;
}

void print_buf(const unsigned char *buf, size_t len)
{
	for (int i = 0; i < len; i++) {
		printf("%02X ", (unsigned)buf[i]);
	}
	puts("");
}

bool is_read_only_param(char *param)
{
	static const char read_param[]= "--read";
	
	return strncmp(param, read_param, sizeof(read_param)) == 0;
}

struct hci_event *get_event_from_buf(void *buf)
{
	struct hci_event *evt = malloc(sizeof(struct hci_event));
	if (!evt)
		exit(EXIT_FAILURE);

	memcpy(evt, buf, sizeof(*evt));

	evt->code = le16toh(evt->code);
	evt->controller_id = le16toh(evt->controller_id);
	evt->size = le16toh(evt->size);
	evt->cmd_opcode = le16toh(evt->cmd_opcode);

	return evt;
}

int set_le_autoconn_timeout(int sockfd, uint16_t new_val)
{
	int status, result;
	struct hci_event *event;

	new_val = htole16(new_val);

	memcpy(
		&set_le_autoconn_timeout_cmd[set_timeout_cmd_value_offset], &new_val, sizeof(new_val)
	);

	status = send(sockfd, set_le_autoconn_timeout_cmd, sizeof(set_le_autoconn_timeout_cmd), 0);
	if (status < 0) {
		perror("Failed to set params");
		return -1;
	}

	memset(read_buf, 0, sizeof(read_buf));

	status  = recv(sockfd, read_buf, sizeof(read_buf), 0);
	if (status < 0) {
		perror("Failed to set params");
		return -1;
	}

	event = get_event_from_buf(read_buf);
	result = (event->code == EVT_CODE_CMD_COMPLETE 
		&& event->size >= 3 
		&& event->cmd_status == EVT_CMD_STATUS_SUCCESS
	) ? 0 : -1;
	free(event);
	
	if (result)
		print_buf(read_buf, status);

	return result;
}

int main(int argc, char *argv[static 1])
{
	int status;
	struct sockaddr_hci addr = {0};
	int sockfd = socket(AF_BLUETOOTH, SOCK_RAW | SOCK_CLOEXEC, BTPROTO_HCI);
	
	if (sockfd < 0) {
		perror("socket create error");
		return EXIT_FAILURE;
	}

	addr.hci_family = AF_BLUETOOTH;
	addr.hci_dev = HCI_DEV_NONE;
	addr.hci_channel = HCI_CHANNEL_CONTROL;

	status = bind(sockfd, (struct sockaddr *)&addr, sizeof(addr));
	if (status < 0) {
		perror("socket bind failed");
		close(sockfd);
		return EXIT_FAILURE;
	}

	status = send(sockfd, read_conf_cmd, sizeof(read_conf_cmd), 0);
	if (status < 0) {
		perror("socket write failed");
		close(sockfd);
		return EXIT_FAILURE;
	}

	status = recv(sockfd, read_buf, 1024, 0);
	if (status <= 0) {
		perror("Didn't receive response");
		close(sockfd);
		return EXIT_FAILURE;
	}

	struct hci_event *event = get_event_from_buf(read_buf);

	if (event->code == EVT_CODE_CMD_COMPLETE && event->cmd_status == EVT_CMD_STATUS_SUCCESS) {
		int autocon_timeout;

		printf("Command complete, result size: %hu.\nBinary response: \n", event->size);
		print_buf(read_buf, status);

		autocon_timeout = get_autoconn_timeout_from_result(read_buf, event->size);
		if (autocon_timeout < 0) {
			return EXIT_FAILURE;
		}

		if (argc < 2 || !is_read_only_param(argv[1])) {
			printf("Set new LE Autoconnect Timeout value to %d... ", NEW_LE_AUTOCONN_TIMEOUT);
			if (set_le_autoconn_timeout(sockfd, NEW_LE_AUTOCONN_TIMEOUT) == 0) {
				puts("OK");
			} else {
				puts("ERROR");
			}
		}
	} else {
		fputs("Could not read current settings\n", stderr);
	}

	free(event);
	close(sockfd);

	return EXIT_SUCCESS;
}
```

jddonovan, 2023-05-28:
I have written my own BLE parser for multiple BLE devices, this is for PVVX/ATC support:

`supported = ((len(data) == ((16-2)*2) or len(data) == ((19-2)*2)) and (data[0:4] == "1a18") )`
All my 100+ devices have worked 100% with this.

After installing 4.2 to 1.5HW: With 2 out of 20 devices the advertising packet is prepended with "02 01 06".

The ones with "02 01 06" are the same ones that cannot be updated to newer software. Why is the "02 01 06" included on some devices randomly? Are they in some different broadcast mode where updating is not possible (tried with Android & Windows)?

All the 20 devices have the same settings (batch flashed). Resetting to defaults (2.5s advertising interval etc.) doesn't remove the broadcast flag "02 01 06" and still doesn't allow updating. Is there some other hard reset possible?

This is very perplexing.

pvvx, 2023-05-28:
"02 01 06" = "Ad Flags" - ble advertising flags.
Without "AdFlags" Bluez does not accept BLE advertising packages in passive scanning mode.
The absence of "AdFlags" means zero value (Bluetooth SIG specification).
[image]
Enabling or disabling the "AdFlags" broadcast is selected in the [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) settings.

closed 2023-06-01 by pvvx