pvvx/ATC_MiThermometer #348: 'hci_rx_fifo', 'hci_tx_fifo', 'puts' undefined symbols

Ircama, 2023-05-30:
I performed the following commands to compile the firmware using Ubuntu or [WSL](https://learn.microsoft.com/en-us/windows/wsl/install):

```bash
git clone https://github.com/pvvx/ATC_MiThermometer.git
cd ATC_MiThermometer
sudo apt-get update
sudo apt install make
make
```

The output includes three undefined symbols.

In fact, when running `python3 ./src/TlsrMemInfo.py -t ./SDK/tools/linux/tc32/bin/tc32-elf-nm ./out/ATC_T
hermometer.elf` I get:

```
TLSR825x MemInfo version 12.11.20
Warning: Undefined symbol 'hci_rx_fifo'!
Warning: Undefined symbol 'hci_tx_fifo'!
Warning: Undefined symbol 'puts'!
===================================================================
 Section|          Description| Start (hex)|   End (hex)|Used space
-------------------------------------------------------------------
 ramcode|   Resident Code SRAM|           0|        59D0|   22992
    text|           Code Flash|        7200|       13780|   50560
  rodata| Read Only Data Flash|       13780|       13D68|    1512
  rtdata|       Retention SRAM|      8459D0|      8471F0|    6176
      nc|   Wasteful Area SRAM|      8471F0|      847200|      16
   ictag|     Cache Table SRAM|      847200|      847300|     256
  icdata|      Cache Data SRAM|      847300|      847B00|    2048
    data|       Init Data SRAM|      847B00|      847B0C|      12
     bss|        BSS Data SRAM|      847B10|      847D51|     577
 irq_stk|        BSS Data SRAM|      847B10|      847C90|     384
   stack|       CPU Stack SRAM|      847D51|      850000|   33455
   flash|       Bin Size Flash|           0|       13D74|   81268
-------------------------------------------------------------------
Start Load SRAM : 29168 (ICtag: 0x7200)
Total Used SRAM : 32081 from 65536
Total Free SRAM : 16 + stack[33455] = 33471
```

When performing the *nm* on *ATC_Thermometer.elf* I get the same three undefined symbols:

```bash
./SDK/tools/linux/tc32/bin/tc32-elf-nm ./out/ATC_Thermometer.elf | grep '         U'
```

Output:

```
         U hci_rx_fifo
         U hci_tx_fifo
         U puts
```

The first two look external references required by [liblt_8258.a](https://github.com/pvvx/ATC_MiThermometer/blob/master/SDK/components/proj_lib/liblt_8258.a) and [liblt_8258_rtos.a](https://github.com/pvvx/ATC_MiThermometer/blob/master/SDK/components/proj_lib/liblt_8258_rtos.a), while *puts* might be needed by [print](https://github.com/pvvx/ATC_MiThermometer/tree/master/SDK/components/application/print).

Why does this happen?

Should we ignore these warnings?

Ircama, 2023-05-30:
I think that the *puts* undefined symbol is because of how [`u_printf()`](https://github.com/pvvx/ATC_MiThermometer/blob/20a087dcc2c234f48790ee618d7273444288a2b9/SDK/components/application/print/u_printf.c#L212-L217) is implemented.

In fact, editing [`UART_PRINT_DEBUG_ENABLE`](https://github.com/pvvx/ATC_MiThermometer/blob/20a087dcc2c234f48790ee618d7273444288a2b9/src/app_config.h#L50) to `1` removes the warning because [a UART debug function](https://github.com/pvvx/ATC_MiThermometer/blob/20a087dcc2c234f48790ee618d7273444288a2b9/SDK/components/application/print/putchar.c#L141-L146) becomes enabled.

```
./SDK/tools/linux/tc32/bin/tc32-elf-nm -u ./out/ATC_Thermometer.elf
         U hci_rx_fifo
         U hci_tx_fifo
```

I don't think that enabling this is appropriate (maybe useless) and anyway `u_printf()` might not be used at all, so the `puts` undefined symbol could be possibly ignored (or `u_printf()` can be modified not to call `puts()` if `UART_PRINT_DEBUG_ENABLE` is disabled).

pvvx, 2023-05-31:
> Output:
```
         U hci_rx_fifo
         U hci_tx_fifo
         U puts
```
Because they are so declared in blob libraries, but are not used in the current program.

Ircama, 2023-06-01:
> Because they are so declared in blob libraries, but are not used in the current program.

Thanks, good to know they are unused and that the warnings can be ignored.

The [Telink SIG Mesh SDK Developer Handbook](http://wiki.telink-semi.cn/doc/an/AN_17120400-E_Telink%20SIG%20Mesh%20SDK%20Developer%20Handbook.pdf) mentions that "hci_tx_fifo and hci_rx_fifo are fifos to define and transmit data by peripherals, e.g., gateway nodes and gateway firmware USB communication".

There is an example of the definition of *hci_rx_fifo* and *hci_tx_fifo* in the [Ai-Thinker-Open/Telink_825X_SDK repository](https://github.com/Ai-Thinker-Open/Telink_825X_SDK/blob/a56437852a59a5d7a4ef0f872e4543e80975e0d6/example/8258_hci/app.c#L66-L82).

The warnings disappear if we put that code in [app.c](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/app.c), or alternatively if we just put
```c
my_fifo_t hci_rx_fifo;
my_fifo_t hci_tx_fifo;
```

But this is at the cost of slightly increasing the size of the executable and slightly reducing the free RAM.

pvvx, 2023-06-01:
The program that outputs these "warnings" was written by me, purely to increase information.
In reality, with a normal build, you will never see these "warnings".
Moreover, it is not necessary to designate the type of these variables. The linker is looking for just a name. You can point to an empty space, to an address to nowhere.

Enter in boot.link:
```
    PROVIDE(hci_rx_fifo = .);
    PROVIDE(hci_tx_fifo = .);
```

closed 2023-06-08 by pvvx