pvvx/ATC_MiThermometer #352: Esphome proxy

yousaf465, 2023-06-06:
Which protocol and integration should be used for compatibility with the esphome proxy? To extend the range of thermometers. I have an esphome Bluetooth proxy  as follows:

`esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
bluetooth_proxy:
  active: true      `

pvvx, 2023-06-07:
I do not recommend using Esphome for any BLE. There are a lot of reception misses and there is no support for Bluetooth 5.0 released back in 2016.

closed 2023-06-07 by pvvx

yousaf465, 2023-06-07:
Any other way for proxying the connection to increase the range? tasmota maybe?

pvvx, 2023-06-07:
"LE Long Range" provides communication at a distance of up to 1 km in a straight line. More than WiFi.
My last checks of various software with ESP32-C3 Bluetooth 5.0 showed terrible instability.

yousaf465, 2023-06-08:
> ["LE Long Range"](https://github.com/pvvx/ATC_MiThermometer/issues/297) provides communication at a distance of up to 1 km in a straight line. More than WiFi. My last checks of various software with ESP32-C3 Bluetooth 5.0 showed terrible instability.
> 
> Reception from a thermometer in a neighboring house on my plot, located at a distance of 200 meters: [image] No additional or extended antennas are used on the adapter.

Which adapter was used for this test? I need to deploy these senors in a large building, so I need some kind of mesh.

pvvx, 2023-06-08:
1. Low-cost Realtek RTL8761 BT adapter
[image]

2. https://github.com/pvvx/ATC_MiThermometer/issues/297#issuecomment-1546475502

yousaf465, 2023-06-08:
> 1. Low-cost Realtek RTL8761 BT adapter
>    [image]
> 2. https://aliexpress.com/item/1005004929786108.html //
>    [Работа термометров в Home Assistant c Bluetooth LE Long Range. #297 (comment)](https://github.com/pvvx/ATC_MiThermometer/issues/297#issuecomment-1546475502)
>    [image]
> 
> > I need to deploy these senors in a large building, so I need some kind of mesh.
> 
> Use normal SoСs with antenna separation (WiFi/BT or LAN) and write your own code on them.

Antenna separation? need specific aliexpress links.

pvvx, 2023-06-08:
OpenWRT + 3G/4G Wi-Fi + USB-BT
[image]

Consumption is less than that of ESP 32.
5V 500mA мах.

Connect the USB-BT adapter to the USB cord at a distance of 50 cm...

yousaf465, 2023-06-08:
> OpenWRT + [3G/4G Wi-Fi](https://aliexpress.com/item/1005005501593909.html) + [USB-BT](https://aliexpress.com/item/1005004929786108.html) [image] [image]
> 
> Consumption is less than that of ESP 32. 5V 500mA мах. [image]
> 
> Connect the USB-BT adapter to the USB cord at a distance of 50 cm...

OpenWrt link for this router? This looks interesting. I don't think anything after openwrt 19 will be supported on this.

pvvx, 2023-06-08:
On small printed circuit boards, the RT5350 has an even lower price...  Less than $1.5

> I don't think anything after openwrt 19 will be supported on this.

Does the BLE<-> MAT task require more?

yousaf465, 2023-06-09:
I have an OpenWRT router running 22.03; how can I set up BLE<--->MQTT or BLE <-->Wifi?

pvvx, 2023-06-09:
It is advisable to write your application in C. I have met ready-made ones only for Python

DjordjeMandic, 2024-04-12:
So you do not suggest esphome btproxy because of:
  - shared antenna
  - bt version 4.2

1) Would esphome btproxy be acceptable option if it would have:
  - W5500 eth
  - bluetooth le v5.3 (esp32-c6, 1x 160mhz)
  - everything compiled on just esp-idf sdk (no arduino stack)
  - can do both TX/RX and active connections

2) Another future option would be esp32-h4:
  - W5500 eth
  - bluetooth le v5.4 (esp32-h4, 2x 96mhz)
  - esp-idf sdk (no arduino stack)
  - can do both TX/RX and active connections
  
 3) CH32V208W ble v5.3 seems perfect for this, you implemented it already but there is no duplex proxy option, just ble adv -> tcp/ip which is not natevly supported by home assistant as i understood it. Can only do RX. No active connections
 
 4) Home assistant supports multiple bluetooth adapters(esphome btproxyies and usb ones, idk about usb/ip because its not documented and tehre is no example how to use it), which solution would you recommend to cover 2-3 buildings with 2 floors?

What do you think about option 1 with esp32-c6 and eth?

DjordjeMandic, 2024-04-12:
> > So you do not suggest esphome btproxy because of:
> > 
> > * shared antenna
> > * bt version 4.2
> > 
> > 
> > 1. Would esphome btproxy be acceptable option if it would have:
> > 
> > 
> > * W5500 eth
> > * bluetooth le v5.3 (esp32-c6, 1x 160mhz)
> > * everything compiled on just esp-idf sdk (no arduino stack)
> > * can do both TX/RX and active connections
> > 
> > 
> > 2. Another future option would be esp32-h4:
> > 
> > 
> > * W5500 eth
> > * bluetooth le v5.4 (esp32-h4, 2x 96mhz)
> > * esp-idf sdk (no arduino stack)
> > * can do both TX/RX and active connections
> > 
> > 
> > 3. CH32V208W ble v5.3 seems perfect for this, you implemented it already but there is no duplex proxy option, just ble adv -> tcp/ip which is not natevly supported by home assistant as i understood it. Can only do RX. No active connections
> > 4. Home assistant supports multiple bluetooth adapters(esphome btproxyies and usb ones, idk about usb/ip because its not documented and tehre is no example how to use it), which solution would you recommend to cover 2-3 buildings with 2 floors?
> > 
> > What do you think about option 1 with esp32-c6 and eth?
> 
> I used Olimax POE as proxies in my project.

Thats BLE 4.2 without coded phy. Hows the latency? Does it miss some adverts?

pvvx, 2024-04-13:
https://github.com/pvvx/ADV_BLE2UART/tree/master/source/esp32-c3/ble50_scan

pvvx, 2024-04-13:
> * CH32V208W ble v5.3 seems perfect for this, you implemented it already but there is no duplex proxy option, just ble adv -> tcp/ip which is not natevly supported by home assistant as i understood it. Can only do RX. No active connections

This is just one example of a use case created for the purpose of obtaining commercial implementations for users from third parties. No other goals were pursued.

> * Home assistant supports multiple bluetooth adapters(esphome btproxyies and usb ones, idk about usb/ip because its not documented and tehre is no example how to use it), which solution would you recommend to cover 2-3 buildings with 2 floors?

If system autonomy is required, then the best option is a BLE-Zigbee repeater. 
Other options involve additional wiring throughout the building, which is not efficient.