pvvx/ATC_MiThermometer #353: Different temperature graph on HA with the same thermometer type

gminadak, 2023-06-08:
After my last issue reported here https://github.com/pvvx/ATC_MiThermometer/issues/286#issue-1535871659.
I have now configured an ESP32 Bluetooth hub with the following configuration:
```
esphome:
  name: ble-hub
  friendly_name: ble-hub

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

esp32_ble_tracker:
  scan_parameters:
    interval: 1100ms
    window: 1100ms
    active: False

# Enable logging
logger:
#  level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "*********************************************************************"

ota:
#  password: "************************************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: .lan
# Manual IP
  manual_ip:
      static_ip: 192.***.***.***
      gateway: 192.***.***.***
      subnet: 255.255.255.0
      dns1: 192.***.***.***

# Sensor Component
sensor:
#------------------Thermometers Configuration----------------------#

# Configuration example for PVVX MiThermometer firmware set to “Custom” advertisement
# For livingroom
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:27:CD:7E"
    temperature:
      name: "livingroom temperature"
    humidity:
      name: "livingroom humidity"
    battery_level:
      name: "livingroom Battery-Level"
    battery_voltage:
      name: "livingroom Battery-Voltage"
    signal_strength:
      name: "livingroom Signal"

# Configuration example for PVVX MiThermometer firmware set to “Custom” advertisement
# For bedroom
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:EC:06:CC"
    temperature:
      name: "bedroom temperature"
    humidity:
      name: "bedroom humidity"
    battery_level:
      name: "bedroom Battery-Level"
    battery_voltage:
      name: "bedroom Battery-Voltage"
    signal_strength:
      name: "bedroom Signal"

# Configuration example for PVVX MiThermometer firmware set to “Custom” advertisement
# For office
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:08:A4:59"
    temperature:
      name: "office temperature"
    humidity:
      name: "office humidity"
    battery_level:
      name: "office Battery-Level"
    battery_voltage:
      name: "office Battery-Voltage"
    signal_strength:
      name: "office Signal"

# Configuration example for PVVX MiThermometer firmware set to “Custom” advertisement
# For kidroom
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:58:D4:97"
    temperature:
      name: "kidroom temperature"
    humidity:
      name: "kidroom humidity"
    battery_level:
      name: "kidroom Battery-Level"
    battery_voltage:
      name: "kidroom Battery-Voltage"
    signal_strength:
      name: "kidroom Signal"

# Configuration example for PVVX MiThermometer firmware set to “Custom” advertisement
# For bathroom
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:E4:73:BB"
    temperature:
      name: "bathroom temperature"
    humidity:
      name: "bathroom humidity"
    battery_level:
      name: "bathroom Battery-Level"
    battery_voltage:
      name: "bathroom Battery-Voltage"
    signal_strength:
      name: "bathroom Signal"
#--------------------------END Thermometers Configuration-------------------------------#

text_sensor:
  - platform: template
    name: "Last Restart Time"
    id: last_restart_time
    icon: mdi:clock
    entity_category: "diagnostic"

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time_sync:
      then:
        # Update last restart time, but only once.
        - if:
            condition:
              lambda: 'return id(last_restart_time).state == "";'
            then:
              - text_sensor.template.publish:
                  id: last_restart_time
                #  state: !lambda 'return id(homeassistant_time).utcnow().strftime("%Y-%m-%dT%H:%M:%S.000000+00:00");'
                  state: !lambda 'return id(homeassistant_time).now().strftime("%d-%m-%Y, %H:%M");'
```

The problem I have is that the new bought Xiaomi Mijia (LYWSD03MMC) thermometers have very different graph on HA with many spikes as shown bellow:
[image]

All the settings are the same on the configuration as can been seen here:

https://github.com/pvvx/ATC_MiThermometer/assets/45330594/26c6474f-124c-45b2-a82f-eb123a7422af

https://github.com/pvvx/ATC_MiThermometer/assets/45330594/6632ea72-9475-48b2-9be9-12ee8650c4a9

Only difference is that the new bought thermometers have different sensor/hardware:
Livingroom: LYWSD03MMC B1.7 (SHT4x)
Office: LYWSD03MMC B1.6 (SHTV3)
Both bought from the same store.

pvvx, 2023-06-08:
> Office: LYWSD03MMC B1.6 (SHTV3)
[image]

For the SHTv3 sensor, turning on "low power" gives a reduced measurement accuracy.

[image]

Enabling "Bluetooth v5.0+" support is unnecessary if not used.

gminadak, 2023-06-09:
Hello @pvvx

Thank you so much for your prompt response and assistance. 
I followed the steps you provided, and I'm happy to report that everything is working perfectly now.

[image]

Once again, thank you for your time and valuable assistance. 
I'm grateful for your help, and I couldn't have resolved this issue without your guidance.

closed 2023-06-10 by pvvx